tzapu/WiFiManager #807: No auto redirect to captive portal on Android and Windows with release 0.14

domingosl, 2019-01-15:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [X] Master
- [ ] Development

**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [X] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

On Windows 10 and Android devices after a successful connection to the WiFi network, there is no redirect to the captive portal. On IOS and Mac, it works. From the serial debug I can see the ESP trying to lunch the captive portal redirect but nothing happens on the devices mentioned before. Either way, with auto-redirect or not, the portal always works by manually going to 192.168.244.1.

tablatronix, 2019-01-15:
I had some issues with windows 10, I think there was a security or firewall blocking it

domingosl, 2019-01-16:
> I had some issues with windows 10, I think there was a security av or firewall blocking it

I have the firewall turned off on Windows and no firewall on my Android phone. Are you able to test the 0.14 on Windows or Android? can you confirm is not a problem only on my hardware?

draconusdesigns, 2019-02-17:
Android version 7 likes to ignore captive portals. I ran into this as a network engineer developing smb routers for an ISP.

workonwomen, 2020-02-28:
Have the same issue that my Android S8 is not redirecting at all... Wondering if an DNS service may not help  here, so user is typing in an random page and then he is redirected...
Pls help.

tablatronix, 2020-02-28:
There is a huge thread on esp about this, dns issue some people had luck my using google dns ips etc instead of class c