tzapu/WiFiManager #809: Spurious IP Address

W4KRL, 2019-01-16:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X] Development

**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X] ESP12 E/F/S (wemos D1 Mini)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [X] 2.4.2
- [ ] staging (master/dev)

### Description
When running the AutoConnectWithFeedbackLED example, the device spins up 192.168.244.1 instead of 192.168.4.1. This has happened with every example I tried. Also used some fresh D1 Minis with same result. Also used release version 0.14 with same results.

### Settings in IDE (Arduino 1.8.8)

Module:  			LOLIN(WEMOS) D1 R2 & mini

Additional libraries: 
#include <Ticker.h>
### Sketch

// LED will blink when in config mode

#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

//for LED status
#include <Ticker.h>
Ticker ticker;
int LED = BUILTIN_LED;

void tick()
{
  //toggle state
  digitalWrite(LED, !digitalRead(LED));     // set pin to the opposite state
}

//gets called when WiFiManager enters configuration mode
void configModeCallback (WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode");
  Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  Serial.println(myWiFiManager->getConfigPortalSSID());
  //entered config mode, make led toggle faster
  ticker.attach(0.2, tick);
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  
  //set led pin as output
  pinMode(LED, OUTPUT);
  // start ticker with 0.5 because we start in AP mode and try to connect
  ticker.attach(0.6, tick);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wm;
  //reset settings - for testing
  // wm.resetSettings();

  //set callback that gets called when connecting to previous WiFi fails, and enters Access Point mode
  wm.setAPCallback(configModeCallback);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if (!wm.autoConnect()) {
    Serial.println("failed to connect and hit timeout");
    //reset and try again, or maybe put it to deep sleep
    ESP.restart();
    delay(1000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");
  ticker.detach();
  //keep LED on
  digitalWrite(LED, LOW);
}

void loop() {
  // put your main code here, to run repeatedly:

}

### Debug Messages
*WM: Connection result: 
*WM: 4
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: password
*WM: Custom AP IP/GW/Subnet
*WM: AP IP address: 
*WM: 192.168.244.1
*WM: HTTP server started
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Info
*WM: Sent info page

tablatronix, 2019-01-16:
I saw this in another issue hmmm

Joeboyc2, 2019-01-17:
> Mine is also 192.168.244.1

I've seen this behaviour also with a sketch that im currently working on,
but it is not consistent, 9 times out of 10 I will see 192.168.4.1

I'm not aware of a way to force this other ip so cant say why it happens

tablatronix, 2019-01-17:
It should always be `192.168.4.1` that is the default ip

tablatronix, 2019-01-17:
erase flash , this is probably corrupt flash problem

pieman64, 2019-01-17:
I have previously erased the flash and it still comes back at 192.168.244.1.

AFAIK when I send the bin file out the world they also use 244.1.

domingosl, 2019-01-17:
I can also confirm, all my ESPs on the latest version of WiFiManager (Master) runs on 244.1

tablatronix, 2019-01-17:
are you confirming this by actually accessing it , or serial output ?

tablatronix, 2019-01-17:
hmm
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp#L173

tablatronix, 2019-01-17:
I cannot make sense of this, some docs on the net clearly say that 244.1 is the def ip in softap, I have never ever seen this ip. Other things say 192.168.4.1

[image]

tablatronix, 2019-01-17:
I think 244.1 is the ip if the custom ip struct from flash or softapconfig is invalid, so I still think this sounds like some kind of flash corruption or esp bug.

another possibility is that master branch is setting ipconfig when it was not set or supposed to and corrupting it.

can you reproduce on development branch ?

W4KRL, 2019-01-18:
I have cleared the SPIFFS several times and tried using earlier versions of WiFi Manager back to 0.9 all with the result that the AP comes up as 192.168.244.1.

Two screen captures are attached showing the captive screens. BTW, is there any way to suppress the IP address forms?
[image]
[image]

Here is serial output from the development version running the AutoConnectWithFeedbackLED example:

*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [2] Custom STA IP/GW/Subnet/DNS 
*WM: [1] STA IP set: (IP unset)
*WM: [3] WIFI station disconnect 
*WM: [1] No saved credentials, skipping wifi 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  ESP_3400ac
*WM: [1] Custom AP IP/GW/Subnet: 
*WM: [0] [ERROR] softAPConfig failed! 
*WM: [2] AP has anonymous access! 
*WM: [1] AP IP address: 192.168.244.1
Entered config mode
192.168.244.1
ESP_3400ac
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.244.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] WiFi Scan ASYNC completed in 2205 ms
*WM: [2] WiFi Scan ASYNC found: 4
*WM: [2] NUM CLIENTS: 0

tablatronix, 2019-01-18:
Clearing spiffs is not a full flash erase.

W4KRL, 2019-01-18:
I ran EraseEsp8266Flash by Ken Taylor https://github.com/kentaylor/EraseEsp8266Flash. It seemed to work. Then immediately ran AutoConnectWithFeedbackLED. Surprisingly, it happily connected to my router! Obviously, Ken's sketch did not erase everything. Can you recommend a way to erase flash without resorting to esptool?

tablatronix, 2019-01-18:
you need to full erase from esptool or arduino ide

W4KRL, 2019-01-18:
Please tell me how to do it using the Arduino IDE.

tablatronix, 2019-01-18:
google it, its in the board menu, I do not use arduino

domingosl, 2019-01-19:
I also see the IP extra fields (Static ip, Static DNS, ...) in the config page as mentioned by @W4KRL and there is no auto-redirect to the captive portal on Windows and Android (Works on Mac and IOS) as I mentioned in this other issue: https://github.com/tzapu/WiFiManager/issues/807 that might be related with this.

tablatronix, 2019-01-20:
I am fairly certain it is corrupt flash or a bug, as this ip comes from esp lib when ip is invalid.

https://github.com/esp8266/Arduino/issues/5627#issuecomment-455700748

tablatronix, 2019-01-20:
You see the ip config form? That is odd as it should only show if you are setting config, so maybe it is a I said above that stable version is setting apconfig mistakinly, but I wont fix or investigate it until it is confirmed in development branch

domingosl, 2019-01-20:
I'll try the dev branch ASAP. BTW, how stable is that branch?

tablatronix, 2019-01-20:
soso, I would say still alpha as I do not get much feedback about usage that is not bugs, but it is tested and used

https://github.com/tzapu/WiFiManager/issues?q=is%3Aissue+milestone%3Adev+is%3Aopen

pfeerick, 2019-01-25:
I'm still on the master branch... i.e. v0.14.0, but this may explain why serial output on the clock project I'm working on changed when I moved from v2.4.2 of the ESP8266 core to the 2.5.0 betas... With no changes to the code, 2.5.0 added the `*WM: Custom STA IP/GW/Subnet` and `IP unset` notation. 

**From 2.4.2:**
```
Build date/time:         Jan 25 2019 / 20:21:50
Core version:            2_4_2
SDK version:             2.2.1(cfd48f3)

*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 192.168.0.131
*WM: freeing allocated params!
```

**From 2.5.0 beta 2:**
```
Build date/time:         Jan 25 2019 / 21:01:06
Core version:            2_5_0_BETA2
SDK version:             3.0.0-dev(c0f7b44)

*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: (IP unset)
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.0.141
*WM: freeing allocated params!
```

tablatronix, 2019-01-25:
https://github.com/esp8266/Arduino/blob/e3bc3c226b4789f30e6f6a77a5522776b01f83bc/cores/esp8266/IPAddress.cpp#L133

tablatronix, 2019-01-27:
It appears that evaluating an ipaddress no longer returns false is not set or null due to changes in ipaddress for ipv6 support. There is now an isset method but it is new so no idea what to do here

damianargento, 2019-02-14:
Hi! Did you found any solution for this? I have exactly the same issue, I've already erased flash, and even reinstaled all libraries and arduino ide but is still connecting to 192.168.244.1 and showing "IP unset" fields.

tablatronix, 2019-02-14:
How did you erase?