tzapu/WiFiManager #810: Need Saved WIFI Credentials to use WiFi.begin(ssid,password)

SummerSoldierGG, 2019-01-18:
### Basic Infos
I am using the OndemandConfigPortal to configure the SSID and PASSWORD for my ESP32 and for the connection I am using "autoconnectap" but the problem is if wifi with saved credentials is not available the system keeps restarting as per autoconnects working which I don't want. I want it to check the wifi for like 20-30 sec and if no wifi found it should move forward and do other operations and go back to sleep. Which is can easily do if I use WiFi.begin but unable to do with "autoconnectap" because it keeps restarting the ESP32 if wifi not found. 
So I want to use the  WIFI SSID and PASS saved by wifimanger to be passed in WiFi.begin.
Please help me how can I get those credentials.

#### Hardware
ESP32 (Adafruit Huzzah)

Thanks in advance and Sorry if I did any mistakes in posting I am Kinda new here.

closed 2019-01-29 by SummerSoldierGG

tablatronix, 2019-01-29:
I guess you figured out wifi.begin is saved

Also there should be no reatarts by wm

SummerSoldierGG, 2019-01-29:
I used the WiFi.SSID() and WiFi.psk() just after when onDemandAP gets connected and then there I saved them in EEPROM and used created a different function which takes ssid and password from EEPROM at beginning of the program to connect with wifi using wifi.begin().

And it was getting restarted if autoconnectap wasn't able to connect to saved WIFI that's why I had to go another way around.

Here are the functions I have created in case someone else is also looking for the same stuff.

//**********************************************************************************************************************
// Function to call ondemand AP in esp for wifi configuration. and save them as string in eeprom
void wifi_config()
{
  digitalWrite(13, LOW);
  digitalWrite(27, HIGH);
  AsyncWiFiManager wifiManager(&server, &dns);

  if (!wifiManager.startConfigPortal("OnDemandAP1")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    ESP.restart();
    delay(5000);
  }
  String wificonf = String(WiFi.SSID() + "#" + WiFi.psk() + ",");
  Serial.print(wificonf);
  Serial.println("eeprom :");
  EEPROM.writeString(address, wificonf);
  Serial.println(EEPROM.readString(address));
  delay(100);
  EEPROM.commit();
  wifi_conf_eeprom();
  Serial.println("connected...yeey :)");
  digitalWrite(13, HIGH);
  digitalWrite(27, LOW);
}
//**********************************************************************************************************************
// fuction to read string from eeprom and make ssid and pass
void wifi_conf_eeprom()
{
  Serial.println("Reading from eeprom :");
  Serial.println(EEPROM.readString(address));
  eprom_wificonf = EEPROM.readString(address);
  int split1 = eprom_wificonf.indexOf("#");
  int split2 = eprom_wificonf.indexOf(",");
  ssID = eprom_wificonf.substring(0, split1);
  pass = eprom_wificonf.substring(split1 + 1, split2);
  Serial.println(ssID);
  Serial.println(pass);
}
//**********************************************************************************************************************
//function to connect on wifi
void wifi_connect()
{
  // Start WIFI
  delay(4000);   //Delay needed before calling the WiFi.begin
  ssid = ssID.c_str();
  password = pass.c_str();
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) { //Check for the connection
    delay(500);
    Serial.println("Connecting to WiFi..");
    count++;
    if (count == 10)
    {
      Serial.println("WiFi Connection error");
      break;
    }
  }
  Serial.println();
  if (WiFi.status() == WL_CONNECTED)
  {
    Serial.println("Connected to the WiFi network");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP()); //Print IP
    Serial.println();
    macaddr(); //Print MAC
  }
}
//**********************************************************************************************************************

tablatronix, 2019-01-29:
The esp already saves credentials to flash itself. There is no need for this whatsoever.

SummerSoldierGG, 2019-01-29:
Then please tell me how i can directly read those saved credentials

And 

Also the restart i am talking about is not the one which i written here , thats i did on purpose in case esp is unable to connect while its getting the configuration.

The restart i am talking about happens when i only use this 
wifiManager.autoConnect("AutoConnectAP")
Instead of wifi.begin().
In the setup() to connect on saved credentials.

But if you can tell me how i can read the credentials saved by wm directly them indeed i don't have to save them again.
Bcoz i have seen many issues here and didn't find any way to directly access those saved credentials.
Or maybe i didn't looked on right place.

Any kind of help is appreciated.

tablatronix, 2019-01-29:
Are you getting a crash in serial monitor? There should be no restarts sounds like a bug yo should open with serial logs.

You can just call wifi.begin()

SummerSoldierGG, 2019-01-29:
U mean i can directly call wifi.begin() without passing ssid and pass. Why no one mentioned it before.

And about the logs , i will try to reproduce it and get you the serial log tomorrow.

Summer-16, 2019-01-30:
I think you are right it does getting crash there see.

this is my setup()
void setup() {
  Serial.begin(115200);
  Serial.println("\n Starting");
  pinMode(TRIGGER_PIN, INPUT_PULLUP);
  pinMode(27, OUTPUT);
  digitalWrite(27, LOW);

  AsyncWiFiManager wifiManager(&server, &dns);
  wifiManager.autoConnect("AutoConnectAP");
  Serial.println("connected...yeey :)");
}

and here are the serial monitor logs
 Starting
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Try to connect with saved credentials
*WM: Connection result: 
*WM: 6
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: About to scan()
*WM: About to scan()
*WM: Scan done
abort() was called at PC 0x4012c743 on core 1

Backtrace: 0x4008c530:0x3ffb1c80 0x4008c761:0x3ffb1ca0 0x4012c743:0x3ffb1cc0 0x4012c78a:0x3ffb1ce0 0x4012c2a3:0x3ffb1d00 0x4012c392:0x3ffb1d20 0x4012c349:0x3ffb1d40 0x400da91b:0x3ffb1d60 0x400dab5d:0x3ffb1dc0 0x400dad91:0x3ffb1e50 0x400d1e0f:0x3ffb1e90 0x400dc983:0x3ffb1fb0 0x4008e7b1:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

it keeps doing this whereas I want him to just move forward in case wifi is not available, perform other operations and go back to sleep. But if wifi is not available it keeps restarting like this.
I just called auto-connect so why does he even trying to start the ap if wifi not available. I wanted him to just move forward.