tzapu/WiFiManager #811: Question: After the ESP is connected to the local network, run a web portal with only has /param

WilkoV, 2019-01-20:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X ] Development

**Esp8266/Esp32:**
- [X ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [X] 2_5_0_BETA2
- [ ] staging (master/dev)

### Description

My questions is: Is it possible to run the web portal after the the ESP is connected to the local network and just provide the /params page. I want to do this, to enable the user to change some configurations but not the WIFI connection

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch
Not done yet

### Debug Messages

-

tablatronix, 2019-01-21:
You can modify the menu but the pages will not be locked out. See dev/ example

froggerXR7, 2019-01-22:
That sounds good. I Was thanking Like that yesterday. But Open Any Web Page On ESP after connection.
 Just jumped in to give another thought.  If it can Be done or not I don't Know.
This might be the same type of question asked.

 Have A Good Programmers Day. If You All Know What i mean. 
  Darn Bugs. Some times my bugs have wings.

dddhhhrrr, 2019-01-31:
is there a way of Not showing the label that appears below the different options in the menu that tells you which network you are connected to?

tablatronix, 2019-01-31:
Only removing it from the strings file atm