tzapu/WiFiManager #814: Stale Bot

tzapu, 2019-01-23:
is it worth adding something like [this](https://github.com/apps/stale) to help cleanup the issues?

cheers

https://github.com/apps/stale

tablatronix, 2019-01-23:
I closed all issues sometime ago and stay on top of them

tablatronix, 2019-01-23:
Ill have to check it out, last I checked you could not filter or exclude tags

tablatronix, 2019-01-23:
Ah it does exclude hmm it cant help, A bot that auto closes incomplete issues would be awesome also lol

tablatronix, 2019-01-23:
@tzapu install it I will configure it tomorrow

tzapu, 2019-01-30:
sorry about the delay, seems i m no longer getting notifications 🤷‍♂️ 

i have installed it, great job with everything

tzapu, 2019-01-30:
"A bot that auto closes incomplete issues" like this one ? https://probot.github.io/apps/no-response/

tablatronix, 2019-01-30:
woooo nice!

tablatronix, 2019-01-30:
sure

tzapu, 2019-01-30:
done, have fun :D