tzapu/WiFiManager #817: Configuring Parameter without SSID

dddhhhrrr, 2019-01-29:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x] Master
- [ ] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [x] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x] 2.4.0
- [ ] staging (master/dev)

### Description

Hi. I'm working on a project in which I will have two variants: one that works via WiFi and other that works over Zigbee. Being that the ESP12 is really cheap, it makes sense for me to use the same PCB for both versions and just add or not the Zigbee module.

I already have the ESP12 version working fine, and I want to use the WiFi capabilities for configuring the parameters needed in the Zigbee version (but turn the WiFi off after that).

Is there any way to use WifiManager to configure ONLY parameters (without needing to configure an SSID to connect to)? I know that WifiManager is meant precisely for the SSID configuration part, but since it's working perfect for the parameter configuration part, I would like to use it for the Zigbee version that will not need to connect to Wifi. Any help?

### Settings in IDE

Module:  			NodeMcu

tablatronix, 2019-01-30:
you can do this in development branch, using custom menu and only showing params/setup menu item. see DEV/ondemandconifg example

closed 2019-01-30 by tablatronix

Emmersonwonka, 2021-02-02:
i need to configurate parameters but i dont wanna conect to a ssid, but if I left ssid = "" wifimanager enter in some kind of bucle

*WM: No ssid, skipping wifi
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root