tzapu/WiFiManager #822: Configuration Portal does not load

DeltaVetal26, 2019-02-07:
#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP32

**Hardware:
ESP32 Dev Module

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

I'm trying to check the WifiManager on my ESP32 card. I loaded an example, but the configuration portal does not want to run. In the list of Wifi networks, I also do not see my fee.

### Sketch

```cpp
#include "soc/soc.h" // Lib For Brownout Detector (Low Power Mode Fix)
#include "soc/rtc_cntl_reg.h" // Lib For Brownout Detector (Low Power Mode Fix)
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

void setup() {
    // put your setup code here, to run once:
    WRITE_PERI_REG (RTC_CNTL_BROWN_OUT_REG, 0);
    Serial.begin(115200);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();

    
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
}

void loop() {}
```

### Debug Messages

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
*WM: [1] AutoConnect 

```

DeltaVetal26, 2019-02-08:
Does anyone have any idea?

tablatronix, 2019-02-08:
use an example and post logs

codebeat-nl, 2019-02-08:
Found this: https://www.esp8266.com/viewtopic.php?f=6&t=15993

This post of eduperez:

----------------

*"Android devices have the Google DNSs hard-coded, they will **always use 8.8.8.8** and 8.8.4.4, despite what the DHPC server might tell them to use. You need to configure your gateway to redirect all outgoing traffic to port 53 to your DNS."*

---------------

That gives the idea, use **8.8.8.8 as IP-address and as DNS-server address**. 
Tada! Works like a charm!

So this sketch will work:

    #include <ESP8266WiFi.h>
    #include <WiFiClient.h>
    #include <ESP8266WebServer.h>
    #include <DNSServer.h>
    #include <ESP8266mDNS.h>
    #include <EEPROM.h>
    
    /*
       This example serves a "hello world" on a WLAN and a SoftAP at the same time.
       The SoftAP allow you to configure WLAN parameters at run time. They are not setup in the sketch but saved on EEPROM.
       Connect your computer or cell phone to wifi network ESP_ap with password 12345678. A popup may appear and it allow you to go to WLAN config. If it does not then navigate to http://192.168.4.1/wifi and config it there.
       Then wait for the module to connect to your wifi and take note of the WLAN IP it got. Then you can disconnect from ESP_ap and return to your regular WLAN.
       Now the ESP8266 is in your network. You can reach it through http://192.168.x.x/ (the IP you took note of) or maybe at http://esp8266.local too.
       This is a captive portal because through the softAP it will redirect any http request to http://192.168.4.1/
    */
    
    /* Set these to your desired softAP credentials. They are not configurable at runtime */
    #ifndef APSSID
    #define APSSID "TheGeekMan"
    #define APPSK  "12345678"
    #endif
    
    const char *softAP_ssid = APSSID;
    const char *softAP_password = APPSK;
    
    /* hostname for mDNS. Should work at least on windows. Try http://esp8266.local */
    const char *myHostname = "thegeekman";
    
    /* Don't set this wifi credentials. They are configurated at runtime and stored on EEPROM */
    char ssid[32] = "";
    char password[32] = "";
    
    // DNS server
    const byte DNS_PORT = 53;
    DNSServer dnsServer;
    
    // Web server
    ESP8266WebServer server(80);
    
    /* Soft AP network parameters */
    //IPAddress apIP(192, 168, 4, 1);
    IPAddress apIP(8, 8, 8, 8);
    IPAddress netMsk(255, 255, 255, 0);
    
    
    /** Should I connect to WLAN asap? */
    boolean connect;
    
    /** Last time I tried to connect to WLAN */
    unsigned long lastConnectTry = 0;
    
    /** Current WLAN status */
    unsigned int status = WL_IDLE_STATUS;
    
    /** Is this an IP? */
[... 242 lines cut ...]
            Serial.print("IP address: ");
            Serial.println(WiFi.localIP());
    
            // Setup MDNS responder
            if (!MDNS.begin(myHostname)) {
              Serial.println("Error setting up MDNS responder!");
            } else {
              Serial.println("mDNS responder started");
              // Add service to MDNS-SD
              MDNS.addService("http", "tcp", 80);
            }
          } else if (s == WL_NO_SSID_AVAIL) {
            WiFi.disconnect();
          }
        }
        if (s == WL_CONNECTED) {
          MDNS.update();
        }
      }
      // Do work:
      //DNS
      dnsServer.processNextRequest();
      //HTTP
      server.handleClient();
    }

DeltaVetal26, 2019-02-09:
[image]

Now it is not compiled at all. He constantly asks for the ESP8266 library, although I have an ESP32 board

DeltaVetal26, 2019-02-09:
I found the Basic example, installed it. But nothing has changed. The access point is also not created.
```txt
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
*WM: [1] AutoConnect 

```

DeltaVetal26, 2019-02-09:
@codebeat-nl 

I'm trying to find an ESP network through an iPhone and a laptop (HP, Windows 10), but nowhere to see the ESP32 access point.

tablatronix, 2019-02-09:
you have not posted logs of the basic example serial output.

DeltaVetal26, 2019-02-09:
@tablatronix 

I am using ESP32 DEV Module V1.
In this example, I apply a setting that disables the low voltage detector. Without this, my board will reboot indefinitely if I use the USB port of my computer. But this is clearly not the case, since I connected the board to a good power source, but nothing works there either.
The serial port displays only this information, nothing else.

tablatronix, 2019-02-09:
well I was assuming it was your code that was breaking it, but no serial output ? That is strange.

I am at a loss, there should be something 
no reset?

DeltaVetal26, 2019-02-10:
@tablatronix 

Is it possible to somehow enable the expanded log?

tablatronix, 2019-02-10:
Its on full by default

DeltaVetal26, 2019-02-11:
@tablatronix 
Happened! The problem is solved by thoroughly cleaning the flash with esptool.py.

tablatronix, 2019-02-11:
very odd, I would have expected some kind of crash or reboot in serial log.
Glad you got it fixed, can it be closed?

closed 2019-02-11 by DeltaVetal26

giulicrenna, 2022-08-30:
I had the same problem, my solution was because I over included the **ESP8266WiFi.h**  in some part of the code.

tablatronix, 2022-08-31:
#1482

mheig, 2024-05-05:
> Happened! The problem is solved by thoroughly cleaning the flash with esptool.py.

Hi! How did you do that?