tzapu/WiFiManager #823: Hide Static IP Station Fields

SomeGuyNamedJay, 2019-02-09:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x] Master
- [ ] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)
- Not sure
### Description
How do I get the Static IP address fields to hide on the Configure WiFi page?
https://www.dropbox.com/s/9nrfkhrob9vp4o7/Screen%20Shot%202019-02-08%20at%209.07.37%20PM.png?dl=0

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {
  wifiManager.setCustomHeadElement("<div>Welcome to Wifi Setup</div>");

  String chipString = "nodemcu-" + chipId;
  const char* chipChar = chipString.c_str();
  //WiFiManagerParameter custom_text(chipChar);
  //wifiManager.addParameter(&custom_text);
    
  apName = "makeSmart";//-" + chipId; //mac_addr;
  apName.toCharArray(apNameCharBuf, 100);

  display.print("Connect your wifi to:\n" + String(apNameCharBuf));
  display.display();

  wifiManager.setAPStaticIPConfig(IPAddress(10, 10, 10, 10), IPAddress(), IPAddress(255, 255, 255, 0));
  //wifiManager.setSTAStaticIPConfig(IPAddress(10,10,10,10), IPAddress(), IPAddress(255,255,255,0));

  wifiManager.autoConnect(apNameCharBuf, apPassword);
}

SomeGuyNamedJay, 2019-02-09:
Looks like I am using Hardware: 2.5.0-beta2

tablatronix, 2019-02-09:
Update your esp lib, this was fixed upstream

SomeGuyNamedJay, 2019-02-09:
Wow, that was easy. Thanks for your help! Great project!!!

closed 2019-02-09 by SomeGuyNamedJay