tzapu/WiFiManager #826: Esp8266 12E connection issue

dprem06, 2019-02-15:
Hi,

My esp826612e is doit.am manufacturer, i tried 3 networks(2 mobile hitspot, 1 Dongle). 2 hotspots or connected AP to Station Mode, but Dongle is returning failed to connect.

chander-60905, 2019-03-23:
did you found any solution. i am also facing issues connecting to router wifi but mobile hotspot works prfectly.

tablatronix, 2019-03-24:
Does it work using begin()

chander-60905, 2019-03-24:
The thing is esp is connected to the router wifi and also to mobile wifi but on router wifi it is not accessing internet to use firebase. Router wifi is using with ssl port and internet is accessible to mobile and laptop but not to esp in lwip variant v2 lower memory and higher bandwidth but it sometimes access internet from router wifi using lwip variant v2 IPv6 lower memory. But after some time internet is again unaccessible.

tablatronix, 2019-03-24:
how is this a wm issue?