tzapu/WiFiManager #829: ESP32 and GoPro Hero 6

alinscodes, 2019-02-18:
Hi. 

I have established a connection between ESP32 and the GoPro but it loses connection after exactly 10 minutes and I don't know how to stop that.
I sent http command to record and after 10 minutes the connection fails and I can't stop the gopro from recording with another http command.
Why is this happening and how can I stop the connection from getting cut,
Thanks

tablatronix, 2019-02-19:
You are connecting to what to what exactly?

alinscodes, 2019-02-19:
your label is incorrect @tablatronix , I have clearly said that the connection is between ESP32 wifi controller and GoPro.

mindforger, 2019-02-19:
this may need more Details

who is the AP in this Scenario, how are you sending the requests?

There is a lot room for assumptions to make here :)

tablatronix, 2019-02-19:
Heh “between” means nothing here.

alinscodes, 2019-02-19:
> 
> 
> this may need more Details
> 
> who is the AP in this Scenario, how are you sending the requests?
> 
> There is a lot room for assumptions to make here :)
> 
> edited: the template when opening an issues is there for a reason

first time using this sorry, I added more detail

alinscodes, 2019-02-19:
> 
> 
> Heh “between” means nothing here.

is that better?

mindforger, 2019-02-19:
From the gopro forum

>You should also remember that the GoPro has battery conservation feature that turns off the WiFi signal after about 10 min. There is no way to disable this so what I do is touch the LCD screen every few minutes to let the camera know I'm still using it even if its not recording

Out of luck i guess, some people claim the 7 behaves better

alinscodes, 2019-02-19:
@mindforger it happens even when it is recording, to fix the problem it is wasn t recording I would wake it up and put it to sleep every 5 minutes, because it turns of wifi when it is sleeping too, but after 10 minutes exactly after recording starts it loses connection

mindforger, 2019-02-19:
@alinscodes this is a known issue and even happens when directly connected via the phone

but that issue with the 6 aside, is it a capacitive touch? can one stimulate the touch with pure electronics and without moving parts?! would be a funny idea to put a virtual finger on the display and fake a touch every 5 minutes or so XD

alinscodes, 2019-02-19:
@mindforger the GoPro will be put on commercial UAV, so that idea sounds funny but stupid for this application haha, so the HERO 6 doesn't have this problem ? 

I also fixed this using a stupid idea, saving the video after 9 minutes and starting it again after 1 second to keep the wifi connection alive

KonradIT, 2019-02-19:
Are you sending a Keep Alive packet every 3 seconds? This keeps the GoPro on and the WiFi active. Alternatively try connecting to the GoPro via BLE.

mindforger, 2019-02-19:
>so the HERO 7 doesn't have this problem ?

i have read people writing this but please don't go out and buy one on my comment here, check the gopro forums first for more insight, I just did a quick search for your issue there to get hints

KonradIT, 2019-02-19:
Literally all GoPros have this problem. It's probably that @alinscodes is not sending the keep alive message periodically.

tablatronix, 2019-02-20:
Do gopros have a io port that you could make a device that keeps it awake by sending harmless commands? Like fake button presses, shrug

alinscodes, 2019-02-20:
@KonradIT I leave it to sleep and then I send WoL packet when I wake it up in order to save battery power. But if I wait more than 10 minutes it loses connection so I try wake it up and put it to sleep every 5 minutes but the problem is when I record for more than 10 minutes it loses connection too, therefore every 9 minutes during recording I have to send command to stop recording and start again,

alinscodes, 2019-02-20:
@mindforger that was the information I needed, now I have to work around that gopro issue to make it do whatever I need. I think this issue is solved tbh

alinscodes, 2019-02-20:
@tablatronix you can t really send WoL packet when it is recording can you?

mindforger, 2019-02-20:
did you try to use a HTTP keepalive?

https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Keep-Alive

this would include to not drop the TCP from the request and requires to set a keep-alive flag in the request header as far as i remember (had this stuff in college last time, so can not give insight, but maybe that helps?)

tablatronix, 2019-02-20:
Sounds like this is a hardware level sleep, keep alive is probably not going to help

alinscodes, 2019-02-20:
yeah, not going to help, I fixed it by waking it and putting it back to sleep every 5 minutes, and then when it is recording for long duration I stop recording and start again every 9 minutes to keep wifi connection alive.

thanks guys for your replies. I appreciate it.

KonradIT, 2019-02-20:
Just in case someone stumbles across this issue and wants to solve this problem:

I'm the maintainer of multiple [GoPro API related repositories](http://github.com/konradit) so I've had this issue before and I know how to solve it. GoPro cameras, when the WiFi is turned on, can either be turned on or off. When the camera is turned off, the WiFi turns off after 10 minutes (in the case of HERO6 and onward, older cameras keep the WiFi on indefinitely). Bluetooth is kept on for roughly an hour in case you need to wake it up (via BT commands of course).

When the camera is turned on with wifi on, the client needs to send [a keep-alive message every 2.5 seconds](https://github.com/KonradIT/gopro-py-api/blob/master/goprocam/GoProCamera.py#L70) to avoid the camera from killing the wifi uplink after 10 minutes of inactivity. If that message is not sent the camera will turn off after a few minutes and then the wifi will be deactivated after 10 minutes. This is a measure to save power.

tablatronix, 2019-02-20:
Sounds easy enough, sending a keepalive using esp8266 is outside the scope of this library, BUT if anyone wants to follow up and post keep alive solutions here feel free for posterity.

mindforger, 2019-02-20:
Should be easy to combine, thank you @KonradIT

alinscodes, 2019-02-21:
@KonradIT is the keep alive message something different from sending WaL packet?
and where would I put this command and how would I send it to GoPro if I am using http commands to communicate with it ? I am a bigginer so I out appreciate your help.

KonradIT, 2019-02-21:
Yes it's different. This weekend I plan to add some more code to GoProControl and fnally get to work on gopro32 (a BLE client for ESP32 to control multiple gopro cameras).

alinscodes, 2019-02-21:
@KonradIT could you direct me to an example of sending this keep alive message from esp32 to gopro please

KonradIT, 2019-02-21:
```
void loop()
{
    WiFiClient client;
    if (!client.connect('10.5.5.9', 8554)) {
        Serial.println("Connection to host failed");
        delay(1000);
        return;
    }
    client.print("_GPHD_:0:0:2:0.000000" );
    client.stop()
}
```

alinscodes, 2019-02-21:
It doesn't look like you are using a keep alive message , I don t understand

KonradIT, 2019-02-21:
That is the keep alive message. Please continue this issue [on my ESP8266 library fork](https://github.com/konradit/GoProControl) or [my ESP32 BT library project](https://github.com/konradit/gopro32)

alinscodes, 2019-02-21:
what is the 8554? 
I am getting CONNECTION TO HOST FAILED

KonradIT, 2019-02-21:
8554 is the port where we send the keep alive messages and we receive the UDP live stream.

closed 2020-04-30 by alinscodes