tzapu/WiFiManager #830: The definition of the "init ()" method is missing

ferchinas, 2019-02-20:
Hello:
I am updating the version of WifiManager to use it with ESP32, and I find the following problems.

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [x] ESP32 D1 mini Mh-Et

**ESP32 Core Version:**
- [ ] 2.3.0
- [ ] 2.4.0
- [x] 1.0.0

### Description

There is a declaration of [init ()](https://github.com/tzapu/WiFiManager/blob/bbc98a7e74453999d3ad9e6f337e4eb2396209d9/WiFiManager.h#L128)  that does not have its definition.
The init () method, is defined as "protected", and I can not use it to initialize the object. (I do not initialize it in the constructor because the "wm" object I use it as a member of my own class)

I already solve these problems by modifying the library, and so I can compile my application.
It would be nice if the library includes the changes

Thank you

tablatronix, 2019-02-20:
init is a method of wifimanager parameter class...

what changes did you make and what are you doing ?

ferchinas, 2019-02-20:
I was using the library a year ago, now I wanted to update it and my code did not compile. That's why I found that:

The method  in WifiManager.h line 128, is not defined in WifiManager.cpp
```
 void init (const char * id, const char * label, const char * defaultValue, int length, const char * custom);

```

And this method in the current version of the library is protected.

My modification was:
1) Add the definition:

```
void WiFiManagerParameter :: init (const char * id, const char * label, const char * defaultValue, int length, const char * custom) {
init (id, label, defaultValue, length, custom, WFM_LABEL_BEFORE);
}
```

2) I made public the init () methods
```
//protected:			
    void init(const char *id, const char *label, const char *defaultValue, int length, const char *custom);
    void init(const char *id, const char *label, const char *defaultValue, int length, const char *custom, int labelPlacement);
```

tablatronix, 2019-02-20:
Oh I see there are 2 init, odd

closed 2019-03-06 by ferchinas