tzapu/WiFiManager #831: AutoConnectWithFSParametersAndCustomIP

eXadra, 2019-02-21:
[image]

Dear all,
Im trying to use this example of the WifiManager for my ESp8266 applications but I cannot use it with fixed ip. This example given does not work.

The code is just the code as is in the example.

"
#include <FS.h>                   //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include "WiFiManager.h"          //https://github.com/tzapu/WiFiManager

#include "ArduinoJson.h"          //https://github.com/bblanchon/ArduinoJson

//define your default values here, if there are different values in config.json, they are overwritten.
//length should be max size + 1 
char mqtt_server[40];
char mqtt_port[6] = "8080";
char blynk_token[33] = "YOUR_BLYNK_TOKEN";
//default custom static IP
char static_ip[16] = "192.168.1.180";
char static_gw[16] = "192.168.1.254";
char static_sn[16] = "255.255.255.0";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");

          strcpy(mqtt_server, json["mqtt_server"]);
          strcpy(mqtt_port, json["mqtt_port"]);
          strcpy(blynk_token, json["blynk_token"]);

          if(json["ip"]) {
            Serial.println("setting custom ip from config");
            //static_ip = json["ip"];
            strcpy(static_ip, json["ip"]);
            strcpy(static_gw, json["gateway"]);
            strcpy(static_sn, json["subnet"]);

            Serial.println(static_ip);
          } else {
            Serial.println("no custom ip in config");
          }
        } else {
          Serial.println("failed to load json config");
        }
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
  //end read
  Serial.println(static_ip);
  Serial.println(blynk_token);
  Serial.println(mqtt_server);

  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 5);
  WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 34);

  WiFiManager wifiManager;
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //set static ip
  IPAddress _ip,_gw,_sn;
  _ip.fromString(static_ip);
  _gw.fromString(static_gw);
  _sn.fromString(static_sn);

  wifiManager.setSTAStaticIPConfig(_ip, _gw, _sn);
  
  //add all your parameters here
  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_blynk_token);
  wifiManager.setMinimumSignalQuality();
  
  if (!wifiManager.autoConnect("AutoConnectAP", "password")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  //read updated parameters
  strcpy(mqtt_server, custom_mqtt_server.getValue());
  strcpy(mqtt_port, custom_mqtt_port.getValue());
  strcpy(blynk_token, custom_blynk_token.getValue());

  //save the custom parameters to FS
  if (shouldSaveConfig) {
    Serial.println("saving config");
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
    json["mqtt_server"] = mqtt_server;
    json["mqtt_port"] = mqtt_port;
    json["blynk_token"] = blynk_token;

    json["ip"] = WiFi.localIP().toString();
    json["gateway"] = WiFi.gatewayIP().toString();
    json["subnet"] = WiFi.subnetMask().toString();

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    json.prettyPrintTo(Serial);
    json.printTo(configFile);
    configFile.close();
    //end save
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());
  Serial.println(WiFi.gatewayIP());
  Serial.println(WiFi.subnetMask());
}

void loop() {
  // put your main code here, to run repeatedly:

}
"

Debug when putting the setting:
*WM: Handle root
*WM: Sent config page
*WM: WiFi save
*WM: Parameter
*WM: server
*WM: 192.168.1.246
*WM: Parameter
*WM: port
*WM: 1883
*WM: Parameter
*WM: blynk
*WM: olamundo
*WM: static ip
*WM: 192.168.1.181
*WM: static gateway
*WM: 192.168.1.254
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.181
*WM: Already connected. Bailing out.
Should save config
connected...yeey :)
saving config
{
  "mqtt_server": "192.168.1.246",
  "mqtt_port": "1883",
  "blynk_token": "olamundo",
  "ip": "192.168.1.181",
  "gateway": "192.168.1.254",
  "subnet": "255.255.255.0"
}local ip
192.168.1.181
192.168.1.254
255.255.255.0
*WM: freeing allocated params!

but when I do a reset this is the debug:
mounting FS...
mounted file system
reading config file
opened config file
{"mqtt_server":"192.168.1.246","mqtt_port":"1883","blynk_token":"olamundo","ip":"192.168.1.181","gateway":"192.168.1.254","subnet":"255.255.255.0"}
parsed json
setting custom ip from config
192.168.1.181
192.168.1.181
olamundo
192.168.1.246
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: blynk
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.181
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: password
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
......

If I comment **_wifiManager.setSTAStaticIPConfig(_ip, _gw, _sn);_** the module sets dhcp and everithing works fine.
Thanks in advance.

sischi, 2019-05-12:
I'm facing the same issue. After first boot, the device will boot its config portal, as expected. After setting up wifi it will give me a
`*WM: Already connected. Bailing out.`
but I can't reach the device (with a ping). After a reboot the device will boot its config portal again (and again, and again, ...).

If I use DHCP (by commenting the static configuration) all works as expected.

tablatronix, 2019-05-12:
try hotfixes branch