tzapu/WiFiManager #832: Changing connection at runtime, second call of autoconnect() portal webpage issue

Bjack795, 2019-02-21:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [X] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [X] Other

### Description
The first time I call autoconnect() function at startup() it's working fine.
Then if I want to change the connection at runtime and I call it again the portal webpage is not working, even though the events in the Serial monitor are the same and I'm able to create the hotspot, I simply can't get access to the portal webpage.

tablatronix, 2019-02-21:
You are calling autoconnect more than once? Not sure I unserstand

Bjack795, 2019-02-22:
I explain it better.
In my project your WiFimanager is called at startup with autoconnect() but I noticed that if I were outside my house the code was blocking there because no known wifi is found and it starts the portal loop never ending.
So I modified it slightly, in practise I trigger the timeout exit after 5 "NUM CLIENTS".
returning true for configPortalHasTimeout(). If I'm outside the code goes on after 5 apClientCheck.
If I want I can open the portal before the timeout and register a wifi station, all goes ok.

Then I wanted to call this configuration on demand and not automatically only at the beginning, because maybe when you turn on the device you're not near to a wifi to be connected but then you are.

Calling again autoconnect() in the middle of the operations it starts the portal if no wifi is known, it gives me the IP.
It also shows up the same [EVENTS] in the Serial Monitor (so no error is showing up) but when I connect to the created temporary wifi and I go to the IP address, the portal webpage is no more showing up.

In my opinion, the first time autoconnect() is used some variable is filled up and differs by default.
So on the second call something is going wrong and alla works fine till portal webpage which is not showing up.

I have studied your code but there is something I'm missing, maybe you could tell me why on the second call I have these problems.
Lastly I say that the second call it's happening in the same situation of the startup so I'm not already connected to any wifi even though I'm in the middle of runtime.

Thank you

Bjack795, 2019-02-22:
I put the two Serial outputs
FIRST CALL

> *WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [2] [EVENT] 2
*WM: [2] Connecting as wifi client... 
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] No saved credentials, skipping wifi 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] password 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA*WM: [2] [EVENT] 3
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  BjackWifi
*WM: [2] [EVENT]
*WM: [2] [EVENT] 3
*WM: [2] [EVENT] 13
*WM: [2] [EVENT] 14
*WM: [2] [EVENT] 13
*WM: [2] [EVENT] 13
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients...*WM: [2] [EVENT] 2
*WM: [2] [EVENT] 1
*WM: [2] WiFi Scan ASYNC completed in 3015 ms
*WM: [2] WiFi Scan ASYNC found: 3
*WM: [2] NUM CLIENTS: 0 
E (44745) event: mismatch or invalid event, id=63
E (44746) event: default event handler failed!
*WM: [2] [EVENT] 63
*WM: [2] [EVENT] 15
*WM: [2] [EVENT] 17
*WM: [2] [EVENT] 17
*WM: [3] -> connectivitycheck.android.com 
> 

SECOND CALL

> *WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] No saved credentials, skipping wifi 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] password 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA*WM: [2] [EVENT] 3
*WM: [2] [EVENT] 3 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  BjackWifi
*WM: [2] [EVENT]
*WM: [2] [EVENT]
*WM: [2] [EVENT] 3
*WM: [2] [EVENT] 3
*WM: [2] [EVENT] 13
*WM: [2] [EVENT] 13
*WM: [2] [EVENT] 14
*WM: [2] [EVENT] 14
*WM: [2] [EVENT] 13
*WM: [2] [EVENT] 13
*WM: [2] [EVENT] 13
*WM: [2] [EVENT] 13
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients...*WM: [2] [EVENT] 2
*WM: [2] [EVENT] 2
*WM: [2] NUM CLIENTS: 0 
*WM: [2] [EVENT] 1
*WM: [2] WiFi Scan ASYNC completed in 3016 ms
*WM: [2] WiFi Scan ASYNC found: 5
*WM: [2] [EVENT] 1
*WM: [2] WiFi Scan ASYNC completed in 3017 ms
*WM: [2] WiFi Scan ASYNC found: 5
E (43441) event: mismatch or invalid event, id=63
E (43442) event: default event handler failed!
*WM: [2] [EVENT] 63
*WM: [2] [EVENT] 63
*WM: [2] [EVENT] 15
*WM: [2] [EVENT] 15
*WM: [2] [EVENT] 17
*WM: [2] [EVENT] 17
*WM: [2] [EVENT] 17
*WM: [2] [EVENT] 17
*WM: [2] NUM CLIENTS: 1 
*WM: [2] NUM CLIENTS: 1 
*WM: [2] NUM CLIENTS: 1 
*WM: [2] NUM CLIENTS: 1 

> 

In the first call I was so fast that NUM CLIENTS is not shown for the first time.
In the second call I never reach  *WM: [3] -> connectivitycheck.android.com"

tablatronix, 2019-02-22:
`event: mismatch or invalid event`

#776

tablatronix, 2019-02-22:
esp said that is a harmless error, but it is odd...

you never get to `*WM: [3] -> connectivitycheck.android.com"`

because your device is not connecting, or failing dns

No idea what would be causing it, I will try to test this, maybe try with startconfigportal, I have used that after autoconnect with no issues

Bjack795, 2019-02-22:
Ok I'll try thank you!

tablatronix, 2019-02-22:
can you try a different device also just to see

Bjack795, 2019-02-22:
I've only my Wemos D32 pro V2

Bjack795, 2019-02-22:
I've tried only the startconfigportal, but the problem is the same.
I can see the hotspot, connect to it but then I can't load the options page to connect to a WiFi.

tablatronix, 2019-02-22:
I mean with another device, like a computer

Bjack795, 2019-02-22:
Oh excuse me I didn't see "client" word.
I'll try from the pc.

tablatronix, 2019-02-23:
I am curious to see if this is some kind of dns issue, have you noticed if the ip is changing the second time?

Bjack795, 2019-02-23:
So
1) I tried with my PC and the result is the same of the Android device.
2) The portal IP is never changing.

Your curiosity could be well posed since I had to add a line to your library some weeks ago.
Since when I was restarting the board (even the autorestart after the uploading of a new script) the first time it was giving me error 202 that is AUTH_FAIL and the code was blocked there without continuing.

Then restarting again all worked fine.

So I modified the code inserting an ESP.restart() in correspondence of AUTH_FAIL occurrence and I solved that problem.
A friend of mine suggested that it could have been a problem related to the Local IP address (that was always remaining the same) and lease time, since I can't put it as continuos.

In practise my D32 was trying to connect with that IP at first restart-> AUTH_FAIL
Then with another restart -> all fine.

So in my opinion this problem could be related to that, but now it's not exiting with AUTH_FAIL since my ESP is not restarting automatically.

Bjack795, 2019-02-23:
PS.
The problem happens either if at startup a connection is established or not.

tablatronix, 2019-02-23:
I was going to say maybe it is your router, sometimes routers do not let a reconnection so quickly because they think the ip is still connected, but it is probably not that. The test would be to have some delay and call a disconnect between the 2 autoconnects

Bjack795, 2019-02-23:
I can't be the router since if I disconnect the wifi and I call again autoconnect() it connects to the wifi if it has the credentials.
The problem of the double call is on the portal side (that is independent on the router since I can choose my neighbour wifi if I want).

I tried either the WiFi.disconnect(true/false) and the WiFi.mode(WIFI_OFF).
Trying autoconnect() or portalconfig() and none of them changed anything.

The portal is working only at startup.
The connection to a memorized wifi is always working.

tablatronix, 2019-02-23:
ok I will see if I can reproduce, and you have the latest pull form development?
And you have a current revision esp32?

Bjack795, 2019-02-23:
I've downloaded it in December maybe I give it a try with the newest one before making you work.
Yup I have the current revision esp32.

Bjack795, 2019-02-23:
Ok same problem with the latest pull from development.

tablatronix, 2019-02-23:
Where is your wm obj declared, new everytime, or a global ?

tablatronix, 2019-02-23:
confirmed, using global wm instance.

```php
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 3 - STA_STOP
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  OnDemandAP
*WM: [2] AP has anonymous access! 
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 14 - AP_STOP
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients...[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
 
*WM: [2] Portal Timeout In 39 seconds
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 1 - SCAN_DONE
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
```

tablatronix, 2019-02-23:
now its working fine, hmm

```
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 1 - SCAN_DONE
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[V][WebServer.cpp:284] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /hotspot-detect.html search: 
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
```

tablatronix, 2019-02-23:
It only happened once, Cannot do it again

```C++
/**
 * OnDemandConfigPortal.ino
 * example of running the configPortal AP manually, independantly from the captiveportal
 * trigger pin will start a configPortal AP for 120 seconds then turn it off.
 * 
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

// select which pin will trigger the configuration portal when set to LOW
#define TRIGGER_PIN 0

int timeout = 40; // seconds to run for
WiFiManager wm; // GLOBAL

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\n Starting");
  pinMode(TRIGGER_PIN, INPUT_PULLUP);
}

void loop() {
  // is configuration portal requested?
  if ( digitalRead(TRIGGER_PIN) == LOW) {
    // WiFiManager wm; // LOCAL

    //reset settings - for testing
    //wifiManager.resetSettings();
  
    // set configportal timeout
    wm.setConfigPortalTimeout(timeout);

    if (!wm.autoConnect("OnDemandAP")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      // ESP.restart();
    }

    //if you get here you have connected to the WiFi
    // Serial.println("connected...yeey :)");

  }

  // put your main code here, to run repeatedly:
}

```

Bjack795, 2019-02-23:
My wm is always new but I tried also to define it globally in the past.
So your last code is working?
I test it and I tell you.
Thank you.

tablatronix, 2019-02-23:
I think they were both working, I had once time where it just sat there. Not sure what to make of this.

Bjack795, 2019-02-23:
I tried your code with a different "trigger" for the autoconnect()
```
/**
 * OnDemandConfigPortal.ino
 * example of running the configPortal AP manually, independantly from the captiveportal
 * trigger pin will start a configPortal AP for 120 seconds then turn it off.
 * 
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

// select which pin will trigger the configuration portal when set to LOW

int timeout = 200; // seconds to run for
WiFiManager wm; // GLOBAL

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\n Starting");
}

void loop() {
  // is configuration portal requested?
  
  if ( Serial.available()>0 && Serial.read() == 99) //if I insert "c" in the Serial Monitor I reload the autoconnect()
  {
    Serial.println();
    // WiFiManager wm; // LOCAL

    //reset settings - for testing
    //wifiManager.resetSettings();
  
    // set configportal timeout
    wm.setConfigPortalTimeout(timeout);
    wm.resetSettings();
    if (!wm.autoConnect("OnDemandAP")) {
      
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      // ESP.restart();
    }

    //if you get here you have connected to the WiFi
    // Serial.println("connected...yeey :)");

  }

  // put your main code here, to run repeatedly:
}
```

The first time I called it, it worked, the second time it didn't:
(I've erased the settings to start again the portal, otherwise it would have connected to the saved wifi)

> *WM: [1] SETTINGS ERASED 
> *WM: [3] WiFi station enable 
> *WM: [1] AutoConnect 
> *WM: [2] ESP32 event handler enabled 
> *WM: [2] Connecting as wifi client... 
> *WM: [1] STA static IP:
> *WM: [2] setSTAConfig static ip not set 
> *WM: [3] WIFI station disconnect 
> *WM: [1] No saved credentials, skipping wifi 
> *WM: [2] Connection result: WL_NO_SSID_AVAIL
> *WM: [3] lastconxresult: WL_NO_SSID_AVAIL
> *WM: [1] AutoConnect: FAILED 
> *WM: [3] WIFI station disconnect 
> *WM: [3] WiFi station enable 
> *WM: [2] Disabling STA 
> *WM: [2] Enabling AP 
> *WM: [1] StartAP with SSID:  OnDemandAP
> *WM: [2] AP has anonymous access! 
> *WM: [1] AP IP address: 192.168.4.1
> *WM: [3] setupConfigPortal 
> *WM: [1] Starting Web Portal 
> *WM: [3] dns server started with ip:  192.168.4.1
> *WM: [2] HTTP server started 
> *WM: [2] WiFi Scan ASYNC started 
> *WM: [2] Config Portal Running, blocking, waiting for clients... 
> *WM: [2] Portal Timeout In 199 seconds
> E (124306) event: mismatch or invalid event, id=63
> E (124306) event: default event handler failed!
> *WM: [2] Portal Timeout In 169 seconds
> E (164123) event: mismatch or invalid event, id=63
> E (164123) event: default event handler failed!
> dhcps: send_offer>>udp_sendto result 0
> *WM: [2] Portal Timeout In 139 seconds
> *WM: [2] Portal Timeout In 109 seconds

tablatronix, 2019-02-23:
Can you connect manually? 192.16.4.1?

Bjack795, 2019-02-24:
To the portal IP?
What do you mean with "manually"?

tablatronix, 2019-02-24:
goto web browser and goto it manually, to see if its just the captive portal that is not working

Bjack795, 2019-02-24:
I've always done like this, I used the captive portal only twice.

tablatronix, 2019-02-24:
interesting, can you ping it ?

Bjack795, 2019-02-24:
On the second call of autoconnect():
4 packages sent
4 received
min: 1 ms
max: 15 ms
average: 6 ms

tablatronix, 2019-02-24:
right so its there, and when to goto http://192.168.4.1, you get a timeout ?

Bjack795, 2019-02-24:
I can connect to the portal but the access page is not available and of course then it results in a timeout since I simply can't reach the page.

tablatronix, 2019-02-24:
and how are you closing the first autoconnect? letting it timeout, pressing exit, saving?

tablatronix, 2019-02-24:
It sounds like the webserver is not setting up correctly the second time.. maybe there is some debugging we can enable, let me check.

Bjack795, 2019-02-24:
It depends, I tried both connecting and saving or leaving it to timeout.

tablatronix, 2019-02-24:
And you are using the example plain, not adding any libraries atm. 

have you tried a full flash erase ( desepration )

Bjack795, 2019-02-24:
I'm using a plain example (the one posted before).
How can I do a full flash erase?

tablatronix, 2019-02-24:
you can use esptool , or there is a board menu option, to erase all or wifi

Bjack795, 2019-02-24:
I can't find this option here [https://user-images.githubusercontent.com/11875156/45855951-20165780-bd18-11e8-9ffb-4f7c739a8dfd.png](url) in my IDE, since with esptool I'm scared to damage the board.
Where is that arduino IDE option?

tablatronix, 2019-02-24:
ohh right you use pio, I do also, in arduino IDE you can erase from the board menu I think

Bjack795, 2019-02-25:
Nope I use Arduino IDE, the link was to show you the option I can't find.
Could you please tell me where this option is in the Arduino IDE?

tablatronix, 2019-02-25:
I don't use arduino, But it looks like the device menu

Bjack795, 2019-02-25:
In my opinion that option is visible only for some kind of devices, not for the esp32.
I will give a try with esptool if I try the correct firmware to be flashed.
Anyways the problem is certainly in flash memory since I also had that problem on "even" restarts as I said in a previous comment.
So it is something staying there even after a restart and even with a plain sketch.

tablatronix, 2019-02-25:
ohh right esp32 uses nvs erase, only know how to do it in code.

tablatronix, 2019-02-25:
esptool erse_flash should work for esp32

Bjack795, 2019-02-25:
But is the erase_flash command enough or then I should re-upload the firmware before loading again the sketches?
Since erase_flash command is totally wiping out the memory, bootloader included, or not?

tablatronix, 2019-02-25:
no bootloader is never erased

Bjack795, 2019-02-25:
I flashed the board with esptool.py --port COM5 erase_flash, successful procedure.

I reloaded the sketch and again the same problem with the same outputs:

> E (78179) event: mismatch or invalid event, id=63
> E (78179) event: default event handler failed!
> E (87352) event: mismatch or invalid event, id=63
> E (87353) event: default event handler failed!
> dhcps: send_offer>>udp_sendto result 0

Erasing flash memory is not working.

tablatronix, 2019-02-25:
man, I have another esp32, Ill check again on a different one. Do you know if your esp32 is NOT rev 0 ?

tablatronix, 2019-02-25:
I wonder if `dhcps: send_offer>>udp_sendto result 0` has anything to do with it, I am not thinking it does, since you can ping it, but I am wondering what that means.

Bjack795, 2019-02-25:
I bought this board two months ago, it's d32 pro v.2 that is also a recent project.
I've not checked the revision but it's highly unlikely that it mounts a revision of more than 2 years ago.
If you want I can check it with the procedure but I think that at 99% it's current rev.

Bjack795, 2019-02-25:
If the problem is only mine I would do an extensive debugging of your library to know what variables are changing within the first and second call.

tablatronix, 2019-02-25:
well if you are using a local scoped wifimanager obj and it still happens, then its not wifimanager, since it gets destroyed and recreated each time. it could be something like a race condition, where it is very hard to reproduce, maybe add some delays or wait then try again, and see if it even responds. I would definitely enable all esp debug levels you can, or switch esp32 versions, are you using staging?

Bjack795, 2019-02-26:
I'm using this sketch https://github.com/tzapu/WiFiManager/issues/832#issuecomment-466691692

Posted before.
As you can see wm is globally defined.

I will try to add some delays and to see what are the variables not properly reinitializing on the second call.

Bjack795, 2019-02-26:
I've enabled a core debug level and I found some strange differences, maybe they're normal but who knows:

First call (first part of log)

> *WM: [1] SETTINGS ERASED 
> *WM: [3] WiFi station enable 
> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
> *WM: [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 14 - AP_STOP
> [1] AutoConnect 
> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
> *WM: [2] ESP32 event handler enabled[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 3 - STA_STOP
> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
> *WM: [2] Connecting as wifi client... 
> *WM: [1] STA static IP:

Second call (first part of log)

> *WM: [1] SETTINGS ERASED 
> *WM: [3] WiFi station enable 
> *WM: [1] AutoConnect 
> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 3 - STA_STOP
> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 3 - STA_STOP
> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
> *WM: [2] ESP32 event handler enabled[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
> *WM: [2] Connecting as wifi client... 
> *WM: [1] STA static IP:

So even though *WM debug is the same, core events seem to differ, what do you think?

Bjack795, 2019-02-26:
Another important observations:
This is the registered behaviour if I timeout the first call

> First call -> I let it go to timeout (captive portal available)
> Second call -> I let it go to timeout (captive portal non working)
> Third call -> I successfully connected (captive portal available)

Otherwise if I connect on first call:

> First call ->  I successfully connected (captive portal available)
> Second call -> I let it go to timeout (captive portal non working)
> Third call -> I let it go to timeout (captive portal non working)
> fourth call -> I successfully connected (captive portal available)

tablatronix, 2019-02-26:
I meant you can use local scope and see if its the same, I think you already did though.

tablatronix, 2019-02-26:
Where is this coming from?

*WM: [1] SETTINGS ERASED

[54 more replies]