tzapu/WiFiManager #834: WifiManager disable autoconnect to wifi

smackflad, 2019-02-24:
Hello, what i'm trying to do is to use wifi manager until the part that it connects to the wifi, for example i want it to enable a AP, the user connect to it via wifi, give wifi credentials and then stop there.
Why i need this: im trying to connect my esp8266 to the blynk server but its unstable while im using WiFiManager.. if I connect to the server with out WiFiManager it works correctly so I figured out that this could be a possible solution to get the credentials and let Blynk configure itself like normal.
I know that this isn't a Blynk forum but the problem is more WiFiManager than Blynk(by my opinion).
Thanks in advance!

Code that connection is unstable:
#define BLYNK_PRINT Serial

#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>

#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>

char auth[] = "xxx";
void setup()
{
  // Debug console
  Serial.begin(9600);
  WiFiManager wifiManager;

  wifiManager.autoConnect("AutoConnectAP");
  Serial.println("connected...yeey :)");
  delay(1000);
  Blynk.config(auth);
}

void loop()
{
  Blynk.run();
}

tablatronix, 2019-02-24:
make sure you are setting mode back to sta incase it is stuck on ap

smackflad, 2019-03-01:
Well, it worked! thank you!

closed 2019-03-01 by tablatronix