tzapu/WiFiManager #836: How do you redirect outside the captives with a link?  What I’m trying to do is after the Wi-Fi manager connects to the network instead of closing the setting page I wanted to open up a web browser to the IP address of the device that I just connected with the Wi-Fi manager which is where my configuration page rest how do I get to that automatically without having to type in the IP address in a browser

athleticmistud, 2019-02-26:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

tablatronix, 2019-02-26:
Use mdns hostname.local

athleticmistud, 2019-02-27:
cant use the .local    need to use the ip address that gets assigned to the
board by the router.  how do you get out of the captive and auto launch a
browser with the ip??

tablatronix, 2019-02-27:
you cannot click links in a captive portal.. so I have no idea what you mean.

athleticmistud, 2019-02-27:
I'm trying to figure out a way to force out captive portal of iOS and open
safari instead, after I save my WiFi settings. I've managed to alter the
standard lib a little so it gets the IP of the AP and server.send before
closing the STA. Problem is, if captive portal does not detect a success
response from internet it does not allow me to open a link or move
automatically to Safari browser.

the idea behind this is to get automatically redirected to the boards IP
address after the AP connects

tablatronix, 2019-02-28:
Afaik the captive portal is sandboxed, you cannot do anything but open local pages in captive window

athleticmistud, 2019-02-28:
ok so how can i do the connection with the wifimanager, then i have a
confirmation page that shows the ip address for them is there a way to then
open a browser from there?  i can long press and it opens the window is
there a way to do that automatically?

the .local is not reliable enough so i need to use the ip to acces the
board.   how else could i do this ?

athleticmistud, 2019-02-28:
what is Afaik?

tablatronix, 2019-02-28:
“As far as I know”

tablatronix, 2019-02-28:
You can show the url and the user has to open it in a browser by copying it etc

jmgk77, 2021-07-31:
https://github.com/tzapu/WiFiManager/issues/1178#issuecomment-890284213