tzapu/WiFiManager #839: BearSSL and WifiManager

quarterturn, 2019-03-01:
Any chance of supporting BearSSL in the future? It would be nice to be able to use like BearSSL::WiFiClientSecure connection; to facilitate things like MQTT with SSL CA cert.

tablatronix, 2019-03-01:
I admit I am ignorant on the subject, it is possible,  can you provide an example ?

I mean wifimanager manager your wifi connection, what does this really have to do with that?

quarterturn, 2019-03-01:
I realize now WifiManager also doesn't work with PubSubClient (for MQTT) so I've had to use the workaround here: https://github.com/CurlyWurly-1/ESP8266-WIFIMANAGER-MQTT/blob/master/MQTT_with_WiFiManager.ino

(I dropped using SSL too for the time being until I get the example above working. FYI use the 5.x ArduinoJson library or you'll have to rewrite the whole thing for 6.x.

I'm not really a C++ expert, but I think BearSSL needs to be able to scope to WifiManager in the same was as WifiClientSecure to be able to work with the connection parameters.

tablatronix, 2019-03-01:
Yeah no idea what is up with pubsubclient, it should work, I think people are just not including stuff properly. I might take a look at that one day

dontsovcmc, 2019-03-27:
@quarterturn I think the issuu is incorrect. WifiManager is WebServer and you ask for HTTPS client support.
Maybe there is any library compiling conflicts? 

I use BearSSL with WifiManager:
http & https: https://github.com/dontsovcmc/waterius/blob/master/ESP8266/src/WateriusHttps.cpp
wifimanager: https://github.com/dontsovcmc/waterius/blob/master/ESP8266/src/setup_ap.cpp

dontsovcmc, 2019-03-27:
@quarterturn I think the issuu is incorrect. WifiManager is WebServer and you ask for HTTPS client support.
Maybe there are any library compiling conflicts? 

I use BearSSL with WifiManager:
http & https: https://github.com/dontsovcmc/waterius/blob/master/ESP8266/src/WateriusHttps.cpp
wifimanager: https://github.com/dontsovcmc/waterius/blob/master/ESP8266/src/setup_ap.cpp