tzapu/WiFiManager #840: What if cannot connect to router?

panosss, 2019-03-04:
Hi everybody.
I was studying the code in the AutoConnectWithTimeOut example.

<pre>#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  
  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;
  //reset settings - for testing
  //wifiManager.resetSettings();

  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds
  wifiManager.setTimeout(180);
  
  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if(!wifiManager.autoConnect("AutoConnectAP")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  } 

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");
 
}

void loop() {
  // put your main code here, to run repeatedly:

}</pre>

Let's say that my router has no power.
So my ESP82266 cannot connect to internet and the code gets stuck forever at:
<pre> if(!wifiManager.autoConnect</pre>

Right?
So how can I make it go to the code in the main loop in this case?

tablatronix, 2019-03-04:
setConfigPortalTimeout

panosss, 2019-03-05:
I suppose you mean [this](https://github.com/tzapu/WiFiManager/blob/master/examples/OnDemandConfigPortal/OnDemandConfigPortal.ino)?

From what I see, the user has to press a button for the ESP to try to connect to the WiFi, right?
This means that it cannot be done automatically but only by the intervention of the user?

tablatronix, 2019-03-05:
No , its in the docs.
https://github.com/tzapu/WiFiManager#configuration-portal-timeout

panosss, 2019-03-05:
How about this code? I tries for 180
<pre>
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFiManager wifiManager;   
   
  wifiManager.setConfigPortalTimeout(180);
 
  if(!wifiManager.autoConnect("AutoConnectAP")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  } 

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");
}

void loop() {
  // put your main code here, to run repeatedly:

}
</pre>

tablatronix, 2019-03-05:
That will start the configportal and timeout after 120 seconds
if you want to try to connect for 3 seconds use setConnectTimeout

```C++
   //sets timeout before AP,webserver loop ends and exits even if there has been no setup.
    //useful for devices that failed to connect at some point and got stuck in a webserver loop
    //in seconds setConfigPortalTimeout is a new name for setTimeout, ! not used if setConfigPortalBlocking
    void          setConfigPortalTimeout(unsigned long seconds);

    //sets timeout for which to attempt connecting, useful if you get a lot of failed connects
    void          setConnectTimeout(unsigned long seconds);
```

panosss, 2019-03-05:
No, I don't want to try to connect for 3 minutes.
I just descried what I thought it does. I was wrong.

So, my program now is completely ok?
If it does not connect at setup, it will, automatically, try to connect sometime at the main loop?

tablatronix, 2019-03-05:
well no because you have 
   ESP.reset();

dontsovcmc, 2019-03-27:
@tablatronix my way is right or I can use setConnectTimeout?
```
#define ESP_CONNECT_TIMEOUT 5000  //5 sec
WiFi.mode(WIFI_STA);
WiFi.begin(); 

uint32_t start = millis();
while (WiFi.status() != WL_CONNECTED && millis() - start < ESP_CONNECT_TIMEOUT) {
    delay(200);
}

if (WiFi.status() == WL_CONNECTED) {
	//do smth
}

```

tablatronix, 2019-03-27:
Both should work