tzapu/WiFiManager #841: Captive Portal does't work on Android devices...

e-music, 2019-03-04:
Hi,

I tried on many different devices, but Captive Portal doesn't work on any of my Android devices. Only on my iPhone, the config portal immediately comes up as soon as the phone is connected to the ESP8266 AP. 

Any advice?

Thanks

h4waii, 2019-03-05:
I noticed this on a stock Samsung Galaxy S9+ running Android 8.1, the "Sign On" page isn't brought up automatically, but I can't reproduce it on my LineageOS 15.1 (Oreo) or LineageOS 16.0 (Pie).

Will update with more details once I can narrow it down a bit.

tablatronix, 2019-03-05:
The latest on this is that, android either has a sandboxed dns system, or use hardcoded google dns 8.8.8.8 and ignores our captive portal there is an issue open in esp arduino if you want to find it and read the comments.

h4waii, 2019-03-05:
Seems like https://github.com/espressif/arduino-esp32/issues/1037 is what you're referring to?

e-music, 2019-03-05:
@h4waii 
Yes, exactly the same issue! 
I tried on Samsung J5 Prime, S8, and Android Tablet. All with the same result!

h4waii, 2019-03-06:
@e-music Setting `8.8.8.8` as AP IP seems to work around this for me. `WiFiManager.setAPStaticIPConfig(IPAddress(8,8,8,8), IPAddress(8,8,8,8), IPAddress(255,255,255,0));`
 before calling `autoConnect()` and see how you fare.

tablatronix, 2019-03-06:
Thanks all, I do not have android to test on

e-music, 2019-03-06:
@h4waii 
Thank you. I will try it asap!

e-music, 2019-03-06:
@h4waii 
WOW!! Changing the IP has solved the issue and now the Captive Portal works on all Android devices! 
Thank you very much for your help!