tzapu/WiFiManager #842: Adding logo to the WifiManager config portal page...

e-music, 2019-03-05:
Hi,

I want to add a banner logo on top of the config portal of WifiManager. I know how to upload the files to SPIFFS and where to insert HTML tags in the code, however, the image doesn't show up in browser. 

I've seen certain examples doing it by serving HTML requests within the code, but not sure how to do the same with WifiManager.

e-music, 2019-03-07:
I guess in order for this to work out, the file and the HTML page that serves that file both have to be on SPIFFS. The problem is the way WiFiManager handles the HTML and HTTP requests. Not so sure as I'm not a web programmer, but there must be a work around to do it. 

Any advice?

tablatronix, 2019-03-07:
You need an end point that serves the image

e-music, 2019-03-07:
Thank you. It would be nice if you can elaborate a little bit about it. I’m coming from pure embedded C background and not well founded in IOT and web programming.

tablatronix, 2019-03-07:
I would give you examples but you didnt fill out the form, you will want to use development branch so you can add event listeners to the webserver, you cannot on stable without modifying the lib code, which is a little easier.

Also you should start with implementing it before trying to integrate into wm

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/FSBrowser/FSBrowser.ino

e-music, 2019-03-07:
Thank you. Examples would help a lot. But, what “form” are you talking about, I didn’t get it.

tablatronix, 2019-03-07:
When you opened the issue there was not a form in the field that you removed?

tablatronix, 2019-03-07:
hHere is an example on route handling in this example in development

https://github.com/tzapu/WiFiManager/blob/development/examples/DEV/OnDemandConfigPortal/OnDemandConfigPortal.ino

e-music, 2019-03-07:
Yes, I get it. I will use it for new issues if that helps. 
Thank you for the examples.

e-music, 2019-03-08:
This sketch "OnDemandConfigPortal.ino" didn't compile. It has tons of issues!

e-music, 2019-03-08:
I've been trying for the past 8 hours but couldn't figure it out. I know that I have to understand how HTTP server works, but for the time being, it would be very helpful if you can provide an example for WM with logo on the config portal page (Captive portal).

e-music, 2019-03-08:
By the way, what makes things more complicated for me is the way the HTML is served. Why not make the HTML pages in SPIFFS instead of using PROGMEM so one can easily customize the page by applying CSS stylesheets and adding customs images.

tablatronix, 2019-03-08:
because then users need to upload the filesystem and it adds complexity and bloat, wm is not a web server, so we do not want to head in that direction. Maybe in a future version

rafalfigura, 2019-03-21:
e-music, did you tried to convert image to base64 and serve that in html?
<img src="data:image/png;base64, iVBORw0KGgoAAAANSUhEUgAAAAUA
AAAFCAYAAACNbyblAAAAHElEQVQI12P4//8/w38GIAXDIBKE0DHxgljNBAAO
    9TXL0Y4OHwAAAABJRU5ErkJggg==" alt="Red dot" />

xtech007, 2020-04-04:
Hi!
Did any of the above worked?
Where you able to add the logo?
If so can you show us, and where in the code!

georgevbsantiago, 2021-05-14:
In the file "strings_en.h" there is a commented PROGMEM with an example of how to insert a logo. It is to generate the representation of the image in base64 (there are free generators on the internet) See below the code snippet:
```
// const char HTTP_ROOT_MAIN[]        PROGMEM = "<img title=' alt=' src='data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAADAAAAAwCAYAAABXAvmHAAADQElEQVRoQ+2YjW0VQQyE7Q6gAkgFkAogFUAqgFQAVACpAKiAUAFQAaECQgWECggVGH1PPrRvn3dv9/YkFOksoUhhfzwz9ngvKrc89JbnLxuA/63gpsCmwCADWwkNEji8fVNgotDM7osI/x777x5l9F6JyB8R4eeVql4P0y8yNsjM7KGIPBORp558T04A+CwiH1UVUItiUQmZ2XMReSEiAFgjAPBeVS96D+sCYGaUx4cFbLfmhSpnqnrZuqEJgJnd8cQplVLciAgX//Cf0ToIeOB9wpmloLQAwpnVmAXgdf6pwjpJIz+XNoeZQQZlODV9vhc1Tuf6owrAk/8qIhFbJH7eI3eEzsvydQEICqBEkZwiALfF70HyHPpqScPV5HFjeFu476SkRA0AzOfy4hYwstj2ZkDgaphE7m6XqnoS7Q0BOPs/sw0kDROzjdXcCMFCNwzIy0EcRcOvBACfh4k0wgOmBX4xjfmk4DKTS31hgNWIKBCI8gdzogTgjYjQWFMw+o9LzJoZ63GUmjWm2wGDc7EvDDOj/1IVMIyD9SUAL0WEhpriRlXv5je5S+U1i2N88zdPuoVkeB+ls4SyxCoP3kVm9jsjpEsBLoOBNC5U9SwpGdakFkviuFP1keblATkTENTYcxkzgxTKOI3jyDxqLkQT87pMA++H3XvJBYtsNbBN6vuXq5S737WqHkW1VgMQNXJ0RshMqbbT33sJ5kpHWymzcJjNTeJIymJZtSQd9NHQHS1vodoFoTMkfbJzpRnLzB2vi6BZAJxWaCr+62BC+jzAxVJb3dmmiLzLwZhZNPE5e880Suo2AZgB8e8idxherqUPnT3brBDTlPxO3Z66rVwIwySXugdNd+5ejhqp/+NmgIwGX3Py3QBmlEi54KlwmjkOytQ+iJrLJj23S4GkOeecg8G091no737qvRRdzE+HLALQoMTBbJgBsCj5RSWUlUVJiZ4SOljb05eLFWgoJ5oY6yTyJp62D39jDANoKKcSocPJD5dQYzlFAFZJflUArgTPZKZwLXAnHmerfJquUkKZEgyzqOb5TuDt1P3nwxobqwPocZA11m4A1mBx5IxNgRH21ti7KbAGiyNn3HoF/gJ0w05A8xclpwAAAABJRU5ErkJggg==' /><h1>{v}</h1><h3>WiFiManager</h3>";
const char HTTP_ROOT_MAIN[]        PROGMEM = "<h1>{v}</h1><h3>WiFiManager</h3>";
```