tzapu/WiFiManager #843: Hotspot not shown

JohannesSchwegler, 2019-03-07:
I tried the following code, but there is no hotspot or access point shown when i upload it to my esp8266.
Does somebody know why?

Code:

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();

    
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:

shamasis, 2019-04-08:
I am pretty new to this. Uploaded the precompiled bin to my Amica NodeMCU v2 https://blog.christophermullins.com/2017/02/11/milight-wifi-gateway-emulator-on-an-esp8266/

LED is blinking every second, but not seeing the ESPXXXX SSID on any device. Thoughts?

tablatronix, 2019-04-08:
You did not fill out form or provide any information