tzapu/WiFiManager #844: I am working on Esp8266 wherein I am trying to configure multiple usernames and password for connecting to Wifi using wifi Manager. I will be using my device at home or at office, So i want to develop the feature where multi wifi and Wifi manger will be integrated so that user can input multiple credentials which will be saved and hence when the device is moved we don't have to manually enter the credentials. Any suggestions are welcome. Thank you

shreyhere8, 2019-03-09:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

acidobinario, 2019-03-23:
@shreyhere8 I'm not sure what the problem is... but I managed to use multiple wifi credentials storing them in SPIFFS with a JSON struct (I'm still working on trying to connect to the best wifi option), tell me what is your approach.

shreyhere8, 2019-03-24:
I am working on a IOT project for smart agriculture. I need to create something like a wifi manager using esp8266 where I would be able to input mulitple wifi credentials and store them on EEPROM. So the Purpose is when I move my device between places, I would not have to re configure the wifi credentials each time. The wifi manager would automatically scan and try to connect to the best available signal.

My Approach was to integrate the Wifi manager and Wifi Multi libraries but I was unable to do so.
The change will also be required in the HTML page for the wifi manager to accept multiple SSIDs and Passwords.

acidobinario, 2019-04-06:
@shreyhere8 check out my blog post of this: https://acidobinario.me/blog/esp8266-wifimanager-multiple-wifi/

an example of that is on my repo https://github.com/acidobinario/espMultiWifiExample