tzapu/WiFiManager #846: WiFiManagerParameter and array is not compiling/working

maikheinrich, 2019-03-09:
Hello All,

I will use the WiFiManagerParameter in combination with an array. This seems not to work.

working example (no array):

WiFiManagerParameter custom_relais_1_off("relais_1_off", "relais_1_off", relais_1_off, 3);
WiFiManagerParameter custom_relais_2_off("relais_2_off", "relais_2_off", relais_2_off, 3);

NOT working example (array):
for (int i = 0; i < 2; i++) {
  WiFiManagerParameter custom_relais_off[i]("relais_off", "relais_off", relais_off[i], 3);
 }

I become an compiler error: variable-sized object 'custom_relais_auto_off' may not be initialized

How must I init this custom_relais_auto_off array ?
 
Issues without basic info will be ignored or closed!

dontsovcmc, 2019-03-27:
This is not C++.

You should use pointers:

    const char *unique_id[2] = {"blah", "hmm"};
    const char *relais_off[2] = {"id1", "id2"};
    WiFiManagerParameter *custom_relais_off[2];
    for (int i = 0; i < 2; i++) {
        custom_relais_off[i] = new WiFiManagerParameter(unique_id[i], "relais_off", relais_off[i], 3);
        wm.addParameter( custom_relais_off[i] );
    }

   // after using WM you should free memory:

    for (int i = 0; i < 2; i++) {
        delete custom_relais_off[i];
    }

Gamelauncher, 2020-10-19:
Hello,

I am struggling with this as well and although I get close, the result is not good .any suggestion?  I've tried following your example with this code:

I need to be able to dynamically allocate a number of rows based on serial numbers of detectors and map them to a zone number. 

```
    int arrArraySize = 5;
    
    WiFiManagerParameter *custom_serial[arrArraySize];

    char displaySerialVal[arrArraySize][11];
    for (int i = 0; i < arrArraySize; i++ ) {
      if (i < sizeStruct) {
        sprintf(displaySerialVal[i], "%lu", rfDetector[i].uuid);
      } else {
        sprintf(displaySerialVal[i], "%i", 0);
      }
    }

    char displaySerialId[arrArraySize][8];
    for (int i = 0; i < arrArraySize; i++ )
      sprintf(displaySerialId[i], "serial%i", i);

    char displaySerialName[arrArraySize][8];
    for (int i = 0; i < arrArraySize; i++ )
      sprintf(displaySerialName[i], "Serial %i", i);

    const char *constDisplaySerialVal[arrArraySize];
    const char *constDisplaySerialId[arrArraySize];
    const char *constDisplaySerialName[arrArraySize];

    for (int i = 0; i < arrArraySize; i++ ) {
      constDisplaySerialVal[i] = new (displaySerialVal) char[11];
      constDisplaySerialId[i] = new (displaySerialId) char[8];
      constDisplaySerialName[i] = new (displaySerialName) char[8];
    }

    for (int i = 0; i < arrArraySize; i++) {
      custom_serial[i] = new WiFiManagerParameter( constDisplaySerialId[i], constDisplaySerialName[i], constDisplaySerialVal[i], 10);
      wm.addParameter( custom_serial[i] );
    }
```

```
void * operator new[](size_t n, const char *s) {
    void *p = malloc(n);
    strcpy((char *)p, s);
    return p;
}
```

Results in:
```
            <label for='serial0'>Serial 0Serial 1Serial 2Serial 3Serial 4</label>
            <br/>
            <input id='serial0' name='serial0' maxlength='10' value='21762832'>
            <label for='serial0'>Serial 0Serial 1Serial 2Serial 3Serial 4</label>
            <br/>
            <input id='serial0' name='serial0' maxlength='10' value='21762832'>
            <label for='serial0'>Serial 0Serial 1Serial 2Serial 3Serial 4</label>
            <br/>
            <input id='serial0' name='serial0' maxlength='10' value='21762832'>
            <label for='serial0'>Serial 0Serial 1Serial 2Serial 3Serial 4</label>
            <br/>
            <input id='serial0' name='serial0' maxlength='10' value='21762832'>
            <label for='serial0'>Serial 0Serial 1Serial 2Serial 3Serial 4</label>
            <br/>
            <input id='serial0' name='serial0' maxlength='10' value='21762832'>
            <br/>
```

Gamelauncher, 2020-10-20:
I've managed to solve this for now, wasn't paying attention to the sizes and overlooked an array indicator. 

Increasing the size of the arraySize and duplicating the code to allow for zone number and zone location values as well, I hit another obstacle ..

CORRUPT HEAP: multi_heap.c:431 detected at 0x3ffedaf8
abort() was called at PC 0x4008c9e4 on core 1

This happens either directly on loading wifimanager or when requesting the param page. Ideally I would like an array size of 64, anyone have an idea on howto optimise the code a bit more for this? Heap seems just happy as after loading successfully with 32, it reports still over 90000 available. When checking the exception decoder it refers to "to many (new) operations" ?

```
    int arrArraySize = 5;
    
    WiFiManagerParameter *custom_serial[arrArraySize];

    char displaySerialVal[arrArraySize][11] = {};
    for (int i = 0; i < arrArraySize; i++ ) {
      if (i < sizeStruct) {
        sprintf(displaySerialVal[i], "%lu", rfDetector[i].uuid);
      } else {
        sprintf(displaySerialVal[i], "%i", 0);
      }
    }

    char displaySerialId[arrArraySize][9] = {};
    for (int i = 0; i < arrArraySize; i++ )
      sprintf(displaySerialId[i], "serial%i", i);

    char displaySerialName[arrArraySize][10] = {};
    for (int i = 0; i < arrArraySize; i++ )
      sprintf(displaySerialName[i], "Serial %i", i);

    const char *constDisplaySerialVal[arrArraySize];
    const char *constDisplaySerialId[arrArraySize];
    const char *constDisplaySerialName[arrArraySize];

    for (int i = 0; i < arrArraySize; i++ ) {
      constDisplaySerialVal[i] = new (displaySerialVa[i]l) char[11];
      constDisplaySerialId[i] = new (displaySerialId[i]) char[9];
      constDisplaySerialName[i] = new (displaySerialName[I]) char[10];
    }

    for (int i = 0; i < arrArraySize; i++) {
      custom_serial[i] = new WiFiManagerParameter( constDisplaySerialId[i], constDisplaySerialName[i], constDisplaySerialVal[i], 10);
      wm.addParameter( custom_serial[i] );
    }
```

```
void * operator new[](size_t n, const char *s) {
    void *p = malloc(n);
    strcpy((char *)p, s);
    return p;
}
```

tablatronix, 2020-10-20:
There is some optimization to be done on the heap that might get you more memory, all the web output uses a single $page string, and no pre allocation is done. There is an open issue discussing this #518

tablatronix, 2020-10-20:
Ill try to find the relevant issue and link it