tzapu/WiFiManager #848: On Android the Portal landing screen is hanged for few seconds.

rohitjust24idpl, 2019-03-13:
@tzapu @tablatronix 
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x] Master
- [x] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] ESP12 F (nodemcu)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)
- [x] 2.4.2

### Description
Specifically on Android ( i have tested on 8.1.0 and 9 ), following is my observation:

1.  Android phone connects to AP quickly.
2. It also open ups the captive portal immediately.
3. But the portal is hanged on landing screen for few seconds and then page appears.

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch

```cpp

#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);
    
    // WiFi.mode(WiFi_STA); // it is a good practice to make sure your code sets wifi mode how you want it.

    //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wm;

    //reset settings - wipe credentials for testing
    //wm.resetSettings();

    // Automatically connect using saved credentials,
    // if connection fails, it starts an access point with the specified name ( "AutoConnectAP"),
    // if empty will auto generate SSID, if password is blank it will be anonymous AP (wm.autoConnect())
    // then goes into a blocking loop awaiting configuration and will return success result

    bool res;
    // res = wm.autoConnect(); // auto generated AP name from chipid
     res = wm.autoConnect("AutoConnectAP"); // anonymous ap
    //res = wm.autoConnect("AutoConnectAP","password"); // password protected ap

    if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }

}

void loop() {
    // put your main code here, to run repeatedly:
    
}
```

### Debug Messages

```
11:17:10.563 -> *WM: [2] Connecting as wifi client... 
11:17:10.563 -> *WM: [1] STA static IP:
11:17:10.563 -> *WM: [2] setSTAConfig static ip not set 
11:17:10.563 -> *WM: [3] WIFI station disconnect 
11:17:10.563 -> *WM: [1] No saved credentials, skipping wifi 
11:17:10.563 -> *WM: [2] Connection result: WL_NO_SSID_AVAIL
11:17:10.563 -> *WM: [3] lastconxresult: WL_NO_SSID_AVAIL
11:17:10.563 -> *WM: [1] AutoConnect: FAILED 
11:17:10.563 -> *WM: [3] WIFI station disconnect 
11:17:10.598 -> *WM: [3] WiFi station enable 
11:17:10.598 -> *WM: [2] Disabling STA 
11:17:10.598 -> *WM: [2] Enabling AP 
11:17:10.598 -> *WM: [1] StartAP with SSID:  AutoConnectAP
11:17:11.088 -> *WM: [2] AP has anonymous access! 
11:17:11.574 -> *WM: [1] AP IP address: 192.168.4.1
11:17:11.574 -> *WM: [3] setupConfigPortal 
11:17:11.574 -> *WM: [1] Starting Web Portal 
11:17:11.611 -> *WM: [3] dns server started with ip:  192.168.4.1
11:17:11.611 -> *WM: [2] HTTP server started 
11:17:11.611 -> *WM: [2] WiFi Scan ASYNC started 
11:17:11.611 -> *WM: [2] Config Portal Running, blocking, waiting for clients... 
11:17:13.759 -> *WM: [2] WiFi Scan ASYNC completed in 2185 ms
11:17:13.759 -> *WM: [2] WiFi Scan ASYNC found: 9
11:17:20.155 -> *WM: [3] -> connectivitycheck.gstatic.com 
11:17:20.155 -> *WM: [2] <- Request redirected to captive portal 
11:17:20.155 -> *WM: [2] <- HTTP Root 
11:17:20.155 -> *WM: [3] -> 192.168.4.1 
11:17:20.155 -> *WM: [3] lastconxresult: WL_IDLE_STATUS
11:17:20.190 -> *WM: [2] Scan is cached 6413 ms ago
11:17:30.352 -> *WM: [3] -> connect.rom.miui.com 
11:17:30.352 -> *WM: [2] <- Request redirected to captive portal 
11:17:40.363 -> *WM: [2] <- HTTP Root 
11:17:40.363 -> *WM: [3] -> 192.168.4.1 
11:17:40.363 -> *WM: [3] lastconxresult: WL_IDLE_STATUS
11:17:40.397 -> *WM: [2] WiFi Scan ASYNC started 
11:17:40.508 -> *WM: [2] NUM CLIENTS: 1 
11:17:40.544 -> *WM: [3] -> 192.168.4.1 
11:17:42.561 -> *WM: [2] WiFi Scan ASYNC completed in 2182 ms
11:17:42.598 -> *WM: [2] WiFi Scan ASYNC found: 8
```

**++++++++++++++++++++++++++++++++++++++++++++++++++**
11:17:20.190 -> *WM: [2] Scan is cached 6413 ms ago
11:17:30.352 -> *WM: [3] -> connect.rom.miui.com 
11:17:30.352 -> *WM: [2] <- Request redirected to captive portal 
11:17:40.363 -> *WM: [2] <- HTTP Root 
**++++++++++++++++++++++++++++++++++++++++++++++++++**
**Notice above 4 lines, it took around 20 seconds to show up the portal**

Please suggest.

tablatronix, 2019-03-13:
Have you pulled recently?

rohitjust24idpl, 2019-03-14:
@tablatronix 
Yes i have latest release v0.14.
Plus i have tried on the latest development version as well.

tablatronix, 2019-03-14:
I mean the latest development there was a bug just fixed for that.

As far as .14 no idea there is often a delay but not usually that long

rohitjust24idpl, 2019-03-15:
@tablatronix 
yes i have tried on both master and development ( both latest ).
issue does not seems to be fixed. i will see if i can shoot a video and send you a link.