tzapu/WiFiManager #849: ESP8266 WifiManager Reconnection and Intermittent connections failure

LeoCharpentier, 2019-03-15:
Hello,

I have been using esp8266 for several months now, and I still have some problems that persist. It is exclusively related to wifi communication, and I can not find real solutions.

This problem is important because my code is used exclusively for the management of actuators (water pump, ventilation, heating, etc.), to control the cultivation above ground of my strawberries.
Sensor measurements sent and stored on server, and management commands retrieved from the server at regular intervals.

 - **FIRST PROBLEM (Reconnection)** :

I am using the "WifiManager" library to automate my connections to the internet router, so that my esp8266 can easily manage a router if it were to be modified. However, I can not make reconnections that work when I lose communication ...

My WifiManager.autoConnect function (XXX, XXX) works fine in the setup (), but when the connection disappears in the loop (), I can not hang it correctly.

For that I tested different things ...

The re-connection with the function "WifiManager.autoconnect ()" directly in the loop () after having recreated and reset my object "WifiManager wifiManager" as seen on this forum:

[Reconnect after setup #738](https://github.com/tzapu/WiFiManager/issues/738)

I have voluntarily added detection conditions to identify a sure way a disconnection with the wifi router, thanks to the response of this forum:

[ESP8266 - WiFi status not changing](https://forum.arduino.cc/index.php?topic=469094.msg3873673#msg3873673)

My test code is then:

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

void setup() {

  // First initialisation of WifiManager in setup() ---> OK
  Serial.begin(115200);
  WiFiManager wifiManager;
  //wifiManager.resetSettings();
  wifiManager.setConfigPortalTimeout(120); 
  wifiManager.setTimeout(120);
  wifiManager.autoConnect("TEST");
  Serial.println("Connected to Wifi");
}

void loop() {

  
 // Detection of router disconnection -> WiFi.status() retrun 6 (WL_DISCONNECTED)
 // OR WiFi.status() retrun 3 (WL_CONNECTED) but with a none IP (0.0.0.0)
 
 if ((WiFi.status() != WL_CONNECTED) || (WiFi.localIP().toString() == "0.0.0.0")) {

  Serial.println("ROUTER DISCONNECTED !!!!");

  // disconnecting the esp to reset the new WifiManager connection ?
  // WHAT IS THE SOLUTION ? 
  //___________________________________________
  WiFi.reconnect(); // ?
  //WiFi.disconnect(); // ?
  //WiFi.mode(WIFI_OFF); // ?
  //WiFi.mode(WIFI_AP_STA); // ?
  //___________________________________________

  WiFiManager wifiManager;
  wifiManager.setConfigPortalTimeout(5); // Access Point not long because just reconnection
  wifiManager.setTimeout(5); // Access Point not long because just reconnection
  wifiManager.autoConnect("TEST");
  Serial.println("Connected to Wifi");
  
 }
 
 else{
  
   /* do the online stuff
    *  ..
    *  ..
    *  ..
    */
    Serial.println("I'M ONLINE ");

 }

 
 Serial.println("");
 delay(1000);
}
```

But no concrete results, impossible to reconnect. No problem in detecting the disconnection between the ESP8266 and the router, but the WifiManager present in the loop (), can not "hang up" the router when it is turned on and operational.

Here is the result of the serial monitor:

```

________________________________________________ // FIRST CONNEXION

*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
wifi evt: 2
wifi evt: 0
wifi evt: 3
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.100
Connected to Wifi
*WM: freeing allocated params!

________________________________________________ // CONNEXION OK !

I'M ONLINE 

I'M ONLINE 

I'M ONLINE 

I'M ONLINE 

I'M ONLINE 

________________________________________________ // VOLUNTARY DISCONNECTION OF THE ROUTER

bcn_timout,ap_probe_send_start

ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
wifi evt: 1
STA disconnect: 200

________________________________________________ // DETECTION DISCONNECTION OK !

ROUTER DISCONNECTED !!!!
scandone
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
scandone
scandone
no Domino-c8 found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
scandone
no Domino-c8 found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
scandone
no Domino-c8 found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
*WM: Connection result: 
*WM: 1
del if0
usl
mode : softAP(ce:50:e3:08:2e:ab)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: 
*WM: Configuring access point... 
*WM: TEST
[AP] softap config unchanged
wifi evt: 8
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
:ur 1
Connected to Wifi
*WM: freeing allocated params!

________________________________________________ // SECOND RECONNECTION TEST

ROUTER DISCONNECTED !!!!
*WM: 
*WM: AutoConnect
bcn 0
del if1
usl
mode : sta(cc:50:e3:08:2e:ab)
ip:0.0.0.0,mask:255.255.255.0,gw:192.168.1.1
add if0
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 0.0.0.0
Connected to Wifi
*WM: freeing allocated params!

wifi evt: 3
wifi evt: 8

________________________________________________ // THIRD RECONNEXION TEST

ROUTER DISCONNECTED !!!!
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
del if0
usl
mode : softAP(ce:50:e3:08:2e:ab)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: 
*WM: Configuring access point... 
*WM: TEST
[AP] softap config unchanged
wifi evt: 8
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
:ur 1
Connected to Wifi
*WM: freeing allocated params!

________________________________________________ //  VOLUNTARY RECONNECTION OF THE ROUTER —— FOURTH RECONNEXION TEST

ROUTER DISCONNECTED !!!!
*WM: 
*WM: AutoConnect
bcn 0
del if1
usl
mode : sta(cc:50:e3:08:2e:ab)
add if0
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
del if0
uwifi evt: 8
sl
mode : softAP(ce:50:e3:08:2e:ab)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: 
*WM: Configuring access point... 
*WM: TEST
[AP] softap config unchanged
wifi evt: 8
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
:ur 1
Connected to Wifi
*WM: freeing allocated params!

________________________________________________

…
....
…
…

```

Considering the lack of solution concerning the reconnection of the router, I apply a rather aggressive patch to my system so as not to lose the connection forever: ESP.restart ();
This temporary solution is now reaching its limits because my actuators are constantly restarting and do not manage my culture correctly ... ..

Do you have a solution ?
Is there a way to reset the wifi settings, between the detection and the first reconnection of the WifiManager in the loop ()?
I'm desperate …

- **SECOND PROBLEM (Connections failure)** :

When I am connected, I use a "WiFiClientSecure" in HTTPS on my secure server, to make an HTTP GET on port 443.

Everything works fine, but intempessively, I observe several "Connection failed" as below:

```
Connecting to XXXXXXXXXX
[hostByName] Host: XXXXXXXXXXX is a IP!
:ref 1
State:	sending Client Hello (1)
:wr 96 96 0
:wrc 96 96 0
:sent 96
:rn 536
:rd 5, 536, 0
:rdi 536, 5
:rd 74, 536, 5
:rdi 531, 74
State:	receiving Server Hello (2)
:rd 5, 536, 79
:rdi 457, 5
:rd 452, 536, 84
:rdi 452, 452
:c0 452, 53:rch 536, 536
:rch 1072, 34
6
:wr 7 7 0
:wrc 7 7 0
Alert: close notify
Server Connection failed
:sent 7
:rcl
:abort
```

What parameters can have a relationship with these connection failures?
What is the good radio signal range to avoid this?
Is there timeout to adjust at the ESP, server or mysql level?
I do not understand why suddenly connections can fail without too much reason ...

Good day, thank you.

LeoCharpentier, 2019-03-15:
@tzapu ?

tzapu, 2019-03-15:
WiFiManager is not really meant to be used that way. It can't control loosing connections, that s what the esp sdk and firmware are doing.
I would say you configure WiFiManager to connect only when you press a button, or only when there are no credentials, but only at boot. If you are loosing connection when the esp has already started and after it connected, you probably need to check your module or your network/router.

tablatronix, 2019-03-15:
the esp8266 automatically reconnnects wifi if you have autoconnect enabled, wifimanager is not needed for that at all, unless you are using static ips and stuff, then yeah you may have issues.

you may want to make sure you are setting wifi_sta mode in your code to make sure softap is not running when it is not supposed to be.

LeoCharpentier, 2019-03-20:
> the esp8266 automatically reconnnects wifi if you have autoconnect enabled, wifimanager is not needed for that at all, unless you are using static ips and stuff, then yeah you may have issues.
> 
> you may want to make sure you are setting wifi_sta mode in your code to make sure softap is not running when it is not supposed to be.
> 
> Also wifi can start failing if your power is not good enough, try boosting it or adding a large capacitor

OK, thanks !
What capacitor value does it take?

tablatronix, 2019-03-20:
no idea, what is your power supply?

LeoCharpentier, 2019-03-26:
> no idea, what is your power supply?

3,3v with battery supply and regulator

tablatronix, 2019-03-26:
Can it supply 500ma ?