tzapu/WiFiManager #850: WiFiManager and painlessMesh together

kbssa, 2019-03-15:
Anyone here have tried to combine WiFiManager and painlessMesh or at least know that is it feasable to use both in the same code.

My application really needs the WiFiManager captive portal like, but I have no idea on how to implement it with the painlessMesh.

If anyone coulp point me to the right direction, I would appreciate a lot.

pranjal-joshi, 2020-05-30:
I also have similar requirements.
My logic is to attempt a direct connection with wifi station firstly. If wifi is unable to connect then switch to mesh mode.
It would be great to hear about updates about this from you!

pranjal-joshi, 2020-06-02:
@kbssa Have you succeeded in using both of them together? I found that WiFiManager is very buggy to use. So i used **ESPAsyncWebServer** lib to create my own web server and with the help of **ArduinoJson**, I am writing my config on the SPIFFS of ESP8266!
Hope this helps! :)

tablatronix, 2020-06-02:
What is painless mesh ?

pranjal-joshi, 2020-06-02:
> What is painless mesh ?
> 
> 

painlessMesh is a library that creates a mesh network of ESPs so that they can communicate with each other without router.
[Link to the painlessMesh library](https://github.com/gmag11/painlessMesh)

tablatronix, 2020-06-02:
So why would you use this with wm if its not networking based?
Or even how would you ?

pranjal-joshi, 2020-06-03:
> So why would you use this with wm if its not networking based?
> Or even how would you ?

Certainly it can be used along with external WiFi networks as well. As ESP provides `MODE_AP_STA`, My design is considers connecting to a WiFi network to access MQTT broker directly. But if WiFi is weak or unavailable due to range issues, I will enter into mesh mode so that I can send MQTT packets over the mesh to a node which is also connected to internet, thus effectively considering that node as a bridge node of the mesh network.

tablatronix, 2020-06-03:
I still do not get it, its 2 seperate things, what would prevent them from working togather?
Its the same as using 2 libraries.

pranjal-joshi, 2020-06-03:
> I still do not get it, its 2 seperate things, what would prevent them from working togather?
> Its the same as using 2 libraries.
I never said that these 2 libs are not compatible with each other. I think @kbssa was asking for logical approach to implement this, maybe like a pseudo-code!
Bdw, @tablatronix Please fix the **ESP requires hard reset** issue. This made me switch from WM to an another library.

tablatronix, 2020-06-03:
What do you mean ESP requires hard reset? There is no issue for that..

tablatronix, 2020-06-03:
I will see about creating an example for combinging with painlessmesh and espmesh or espnow

pranjal-joshi, 2020-06-03:
> What do you mean ESP requires hard reset? There is no issue for that..

While using WM v2.0.2, I faced the issue that after receiving the config, the ESP still stays in blocking mode (even though it successfully saves the config in the memory). However, with a hard reset/power-cycle, the ESP successfully retrieves config from memory and proceeds as expected.

pranjal-joshi, 2020-06-03:
> I will see about creating an example for combinging with painlessmesh and espmesh or espnow
> 
> * [ ]  example candidate

@tablatronix - This might be a good point to take a look before writing an example.
[https://github.com/gmag11/painlessMesh/blob/master/examples/mqttBridge/mqttBridge.ino](https://github.com/gmag11/painlessMesh/blob/master/examples/mqttBridge/mqttBridge.ino)

tablatronix, 2020-06-03:
are you sacing parameters and not wifi? you have to set shoudBreak true or else it doesnt stop the cp

pranjal-joshi, 2020-06-03:
@tablatronix Good to know about this. In my sketch, I was trying to save both WiFi config (SSID+PASS) along with 4-5 custom params which are to be stored on SPIFFS.
If you didn't have mentioned about `shouldBreak` in the docs, please consider it so.

tablatronix, 2020-06-03:
Ill try to find a better fix for this this week and add better callbacks for param saves.

I thought it was added to the readme but I am not seeing it ...
hmm

tablatronix, 2020-06-03:
doh it was added to hotfixes ill adjust that...

```
Save settings
This gets called when custom parameters have been set AND a connection has been established. Use it to set a flag, so when all the configuration finishes, you can save the extra parameters somewhere.

IF YOU NEED TO SAVE PARAMETERS EVEN ON WIFI FAIL OR EMPTY, you must set setBreakAfterConfig to true, or else saveConfigCallback will not be called.

//if this is set, it will exit after config, even if connection is unsuccessful.
    void          setBreakAfterConfig(boolean shouldBreak);
```

tablatronix, 2020-06-03:
You should test out

```C++
  // wm.setParamsPage(true); // move params to seperate page, not wifi, do not combine with setmenu!, use setSaveParamsCallback
```

kubitz, 2021-09-06:
> > I will see about creating an example for combinging with painlessmesh and espmesh or espnow
> > 
> > * [ ]   example candidate
> 
> @tablatronix - This might be a good point to take a look before writing an example.
> https://github.com/gmag11/painlessMesh/blob/master/examples/mqttBridge/mqttBridge.ino

@pranjal-joshi did you end up implementing this in one of your projects?

markg85, 2022-03-20:
I'm curious about this too.
Contrary to what was said in this post earlier, I do think mesh an wifimanager should be complementary **if** you have a root node that lives on your normal network too. Kind of a gateway node into your mesh network.

In that case, the root node has normal plain wifi and needs to be configured as such. Hence the need for something like WifiManager.

An example with mesh + WifiManager would be an awesome starting point!

hemangjoshi37a, 2022-06-07:
I am curious if anyone already has this implemented in thier project.
Please let me know because I am trying this to do this from very long time.
[Hemang Joshi](https://hjlabs.in/)