tzapu/WiFiManager #853: NodeMCU resets during captive portal WiFi configuration when using mobile phone.

scalesr1, 2019-03-18:
#### Hardware
**WiFimanager Branch/Release:**
- [X ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

This has worked perfectly for a simple project that I am working on, after a couple of months I get to try my 'device' on another network. The AP starts OK and I can connect and get the config portal but as I am setting up the WiFi, the  Node MCU gets a soft reset and restarts.
When I look at the output of the serial monitor I see lots of output that appears to relate to a website (AliExpress) that I know I have visited and I have the app on my phone.
Here is the output:

method: POST url: /amdc/mobileDispatch search: appkey=21371601&deviceId=WdJhcPBYgLIDALr%2B3V98A2yi&platform=android&v=4.0
headerName: Connection
headerValue: close
headerName: Accept-Encoding
headerValue: gzip
headerName: Content-Type
headerValue: application/x-www-form-urlencoded
headerName: User-Agent
headerValue: Dalvik/2.1.0 (Linux; U; Android 6.0.1; A0001 Build/MHC19Q)
headerName: Host
headerValue: amdc.aliexpress.com
headerName: Content-Length
headerValue: 324
args: appkey=21371601&deviceId=WdJhcPBYgLIDALr%2B3V98A2yi&platform=android&v=4.0&bssid=de%3A4f%3A22%3A0b%3Aa7%3A4b&sign=d232481c588d9ce29396d6de4521d7820d3397e2&carrier=wifi&domain=amdc.aliexpress.com%20msg-global.m.taobao.com&appName=Aliexpress_Android&lng=0.0&platformVersion=6.0.1&mnc=wifi&sid=720874442&cv=0&appVersion=6.23.1&signType=sec&t=1552926683110&netType=WIFI&lat=0.0&channel=play_in_msite_top
args count: 20
pos 0=@ 6 &@ 15
arg 0 key: appkey value: 21371601
pos 16=@ 24 &@ 51
arg 1 key: deviceId value: WdJhcPBYgLIDALr+3V98A2yi
pos 52=@ 60 &@ 68
arg 2 key: platform value: android
pos 69=@ 70 &@ 74
arg 3 key: v value: 4.0
pos 75=@ 80 &@ 108
arg 4 key: bssid value: de:4f:22:0b:a7:4b
pos 109=@ 113 &@ 154
arg 5 key: sign value: d232481c588d9ce29396d6de4521d7820d3397e2
pos 155=@ 162 &@ 167
arg 6 key: carrier value: wifi
pos 168=@ 174 &@ 220
arg 7 key: domain value: amdc.aliexpress.com msg-global.m.taobao.com
pos 221=@ 228 &@ 247
arg 8 key: appName value: Aliexpress_Android
pos 248=@ 251 &@ 255
arg 9 key: lng value: 0.0
pos 256=@ 271 &@ 277
arg 10 key: platformVersion value: 6.0.1
pos 278=@ 281 &@ 286
arg 11 key: mnc value: wifi
pos 287=@ 290 &@ 300
arg 12 key: sid value: 720874442
pos 301=@ 303 &@ 305
arg 13 key: cv value: 0
pos 306=@ 316 &@ 323
arg 14 key: appVersion value: 6.23.1
pos 324=@ 332 &@ 336
arg 15 key: signType value: sec
pos 337=@ 338 &@ 352
arg 16 key: t value: 1552926683110
pos 353=@ 360 &@ 365
arg 17 key: netType value: WIFI
pos 366=@ 369 &@ 373
arg 18 key: lat value: 0.0
pos 374=@ 381 &@ -1
arg 19 key: channel value: play_in_msite_top
args count: 20
Plain: bssid=de%3A4f%3A22%3A0b%3Aa7%3A4b&sign=d232481c588d9ce29396d6de4521d7820d3397e2&carrier=wifi&domain=amdc.aliexpress.com%20msg-global.m.taobao.com&appName=Aliexpress_Android&lng=0.0&platformVersion=6.0.1&mnc=wifi&sid=720874442&cv=0&appVe
Soft WDT reset

ctx: cont 
sp: 3ffffaf0 end: 3fffffd0 offset: 01b0

>>>>stack>>>>

In desperation I tried connecting using my laptop - it all works fine, the wifi is configured and my device runs as normal.

Is this something to do with the AliExpress App on my phone causing confusion?
A colleague of mine had a similar issue in as much as the ESP8266 would keep resetting whilst he was attempting to configure it. I have NOT seen the serial output of that yet. Neither have I tried disabling the app on my phone (I will try that next too).
Is this an issue and/or something that should be avoided?
Is there a way I can stop if from happening?

### Settings in IDE

Module:  			NodeMu

Additional libraries:  Arduino OTA - though removing this made no difference.

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: 
*WM: Configuring access point... 
*WM: TwoTubeClockAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
New client
method: GET url: /generate_204 search: 
headerName: User-Agent
headerValue: Dalvik/2.1.0 (Linux; U; Android 6.0.1; A0001 Build/MHC19Q)
headerName: Host
headerValue: connectivitycheck.gstatic.com
headerName: Connection
headerValue: Keep-Alive
headerName: Accept-Encoding
headerValue: gzip
args: 
Request: /generate_204
 Arguments: 
request handler not found
*WM: Request redirected to captive portal
New client
method: POST url: /amdc/mobileDispatch search: appkey=21371601&deviceId=WdJhcPBYgLIDALr%2B3V98A2yi&platform=android&v=4.0
headerName: Connection
headerValue: close
headerName: Accept-Encoding
headerValue: gzip
headerName: Content-Type
headerValue: application/x-www-form-urlencoded
headerName: User-Agent
headerValue: Dalvik/2.1.0 (Linux; U; Android 6.0.1; A0001 Build/MHC19Q)
headerName: Host
headerValue: amdc.aliexpress.com
headerName: Content-Length
headerValue: 324
args: appkey=21371601&deviceId=WdJhcPBYgLIDALr%2B3V98A2yi&platform=android&v=4.0&bssid=de%3A4f%3A22%3A0b%3Aa7%3A4b&sign=d232481c588d9ce29396d6de4521d7820d3397e2&carrier=wifi&domain=amdc.aliexpress.com%20msg-global.m.taobao.com&appName=Aliexpress_Android&lng=0.0&platformVersion=6.0.1&mnc=wifi&sid=720874442&cv=0&appVersion=6.23.1&signType=sec&t=1552926683110&netType=WIFI&lat=0.0&channel=play_in_msite_top
args count: 20
pos 0=@ 6 &@ 15
arg 0 key: appkey value: 21371601
pos 16=@ 24 &@ 51
arg 1 key: deviceId value: WdJhcPBYgLIDALr+3V98A2yi
pos 52=@ 60 &@ 68
arg 2 key: platform value: android
pos 69=@ 70 &@ 74
arg 3 key: v value: 4.0
pos 75=@ 80 &@ 108
arg 4 key: bssid value: de:4f:22:0b:a7:4b
pos 109=@ 113 &@ 154
arg 5 key: sign value: d232481c588d9ce29396d6de4521d7820d3397e2
pos 155=@ 162 &@ 167
arg 6 key: carrier value: wifi
pos 168=@ 174 &@ 220
arg 7 key: domain value: amdc.aliexpress.com msg-global.m.taobao.com
pos 221=@ 228 &@ 247
arg 8 key: appName value: Aliexpress_Android
pos 248=@ 251 &@ 255
arg 9 key: lng value: 0.0
pos 256=@ 271 &@ 277
arg 10 key: platformVersion value: 6.0.1
pos 278=@ 281 &@ 286
arg 11 key: mnc value: wifi
pos 287=@ 290 &@ 300
arg 12 key: sid value: 720874442
pos 301=@ 303 &@ 305
arg 13 key: cv value: 0
pos 306=@ 316 &@ 323
arg 14 key: appVersion value: 6.23.1
pos 324=@ 332 &@ 336
arg 15 key: signType value: sec
pos 337=@ 338 &@ 352
arg 16 key: t value: 1552926683110
pos 353=@ 360 &@ 365
arg 17 key: netType value: WIFI
pos 366=@ 369 &@ 373
arg 18 key: lat value: 0.0
pos 374=@ 381 &@ -1
arg 19 key: channel value: play_in_msite_top
args count: 20
Plain: bssid=de%3A4f%3A22%3A0b%3Aa7%3A4b&sign=d232481c588d9ce29396d6de4521d7820d3397e2&carrier=wifi&domain=amdc.aliexpress.com%20msg-global.m.taobao.com&appName=Aliexpress_Android&lng=0.0&platformVersion=6.0.1&mnc=wifi&sid=720874442&cv=0&appVe
Soft WDT reset

ctx: cont 
sp: 3ffffaf0 end: 3fffffd0 offset: 01b0

>>>stack>>>
3ffffca0:  3ffffd60 00000002 3ffef6dc 4020eec0  
3ffffcb0:  3ffffd60 3fff005c 3ffef6dc 4020f1a1  
3ffffcc0:  3ffe8f04 00000014 3ffef6dc 4020f4c3  
3ffffcd0:  3ffe8fe8 3fff005c 3ffef6dc 4020f354  
3ffffce0:  3ffffd60 3fff005c 3ffef6dc 4020f4fc  
3ffffcf0:  3ffffd60 3fff005c 3ffef6dc 4020846d  
3ffffd00:  00000000 00000000 00000000 3fff141c  
```
Can anyone point me in the right direction?
```

tablatronix, 2019-03-19:
This is probably a buffer overflow in the esp webserver. You would have to decode that stack dump and submit it to esp lib, the request could be the app dialing home or just checking in the background, it seems to send a very large payload. 

We can maybe block these I will look into it. But not sure since it is not in wm

scalesr1, 2019-03-19:
Thank you for that pointer, I shall try again having disabled that particular app though - 'in the field' anyone could have any app running that might do the same kind of thing. Is there any other method that would perhaps stop a device from seeing that this is a gateway - like not giving out a gateway address, would that help?

tablatronix, 2019-03-19:
not really, there is going to be every internet enabled app trying to contact the internet, we can try to reject non captive portal requests, but it might not help the flood of requests.

Another option, when you are connecting to a captive portal you are supposed to disable all other connections, do you have LTE, try turning cell data off and see if it works better