tzapu/WiFiManager #854: timer crash when using it with this library

Adrianotiger, 2019-03-20:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release: Development**

**Esp8266/Esp32: ESP32**

**Hardware:			DevKit 1.0**

**ESP Core Version:	1.0.1 and 1.0.2 RC1**

### Description

If using this library with a timer, the esp will crash with this error:
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)

### Sketch

```cpp

#include "esp32-hal-timer.h"
#include <WiFiManager.h>

hw_timer_t *timer = NULL;
WiFiManager wifiManager;

void IRAM_ATTR _onTimer() {
  dacWrite(25, random(0,255));
}

void setup() {
  Serial.begin(115200);
  
  timer = timerBegin(0, 80, true);          // 800 = 80MHz / 80 = 1000kHz (1us pulse)
  timerAttachInterrupt(timer, &_onTimer, true);
  timerAlarmWrite(timer, 10, true);       // every 1ms
  timerAlarmEnable(timer);
  Serial.println("Timer started");
  
  wifiManager.autoConnect("Test", "test1234"); // <-- Crash!
  Serial.println("WiFi started");
}

void loop() {
  delay(1000);
}

```

### Debug Messages

```
Timer started
*WM: [1] AutoConnect 
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40154aa0: bad00bad bad00bad bad00bad
...
```

tablatronix, 2019-03-20:
This might be a esp lib bug, you would have to debug or get a stack trace to find out

Adrianotiger, 2019-03-20:
Thank you for the fast reply.
Yes, I am trying to find out what it can be. But I don't get any stack trace to find out what it can be.

I saw that if I try to access a preference variable once the timer is started, the sketch will crash too. I just noticed it with this library and posted it here. Sorry.

tablatronix, 2019-03-20:
not sure how you debug esp32, maybe someone else can.

is IRAM_ATTR correct?

tablatronix, 2019-03-20:
https://github.com/espressif/arduino-esp32/issues/1907

tablatronix, 2019-03-20:
See if those provide any answers

Adrianotiger, 2019-03-21:
I got an answer directly on the espressif page:
I am accessing the hardware from an interrupt and this should never be done if the interrupt occurs so fast (every 10us - 100us). 
dacWrite need to access some registers and is not so fast like "digitalRead/digitalWrite". If this happens during another access to parameters or memory-call, this can cause a crash.

Since this WiFiManager-library access the nvs, no interrupt routines with hardware access should be executed in background, as this could cause a crash. 

Shortly: it was my error. Thanks you for the links!

closed 2019-03-21 by Adrianotiger