tzapu/WiFiManager #855: Saving custom parameters to EEPROM

megaESP, 2019-03-21:
First of all big thanks to Tzapu, Tablatronix and everyone else involved in this project. Kudos to you guys for your hard work, it is very much appreciated.

I was wondering if there is an example showing saving of custom parameters in EEPROM. I know it is one of the to do list, but in case you guys have something to share even half finished code, that will be helpful.

Thanks again
:-)

Muhammed-Hayr-Ali, 2019-03-22:
#define BLYNK_PRINT Serial
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <EEPROM.h>
#include <BlynkSimpleEsp8266.h>

WiFiManager wifiManager;

/****************************************************************/

//flag for saving data
bool shouldSaveConfig = false;

const unsigned long CONNECT_TIMEOUT = 10; // Wait 10 Seconds  to connect to the real AP before trying to boot the local AP
const unsigned long AP_TIMEOUT = 20; // Wait 20 Seconds in the config portal before trying again the original WiFi creds

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

/********************** Begin EEPROM Section *****************/
#define EEPROM_SALT 12664
typedef struct
{
  int   salt = EEPROM_SALT;
  char blynk_token[33]  = "";
  char device_name[20]  = "";
}

WMSettings;
WMSettings blynk;
WMSettings device;

void eeprom_read()
{
  EEPROM.begin(512);
  EEPROM.get(0, blynk);
  EEPROM.get(100, device);
  EEPROM.end();
}

void eeprom_saveconfig()
{
  EEPROM.begin(512);
  EEPROM.put(0, blynk);
  EEPROM.put(100, device);
  EEPROM.commit();
  EEPROM.end();
}

/*********************************************************************************/

void setup()
{
  Serial.begin(115200);

  /****************************************************************/

  eeprom_read();
  if (blynk.salt != EEPROM_SALT)
  {
    Serial.println("Invalid settings in EEPROM, trying with defaults");
    WMSettings defaults;
    blynk = defaults;
    device = defaults;
  }

  WiFiManagerParameter custom_blynk_token("blynk-token", "Blynk Token", blynk.blynk_token, 33);
  WiFiManagerParameter custom_device_name("device_name", "Device Name",  device.device_name, 20);

  WiFiManager wifiManager;
  wifiManager.setSaveConfigCallback(saveConfigCallback);
  wifiManager.addParameter(&custom_blynk_token);
  wifiManager.addParameter(&custom_device_name);

  //reset settings - for testing
  //wifiManager.resetSettings();
  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  wifiManager.setMinimumSignalQuality();
  //useful to make it all retry or go to sleep
  //in seconds
  wifiManager.setConnectTimeout(CONNECT_TIMEOUT);
  wifiManager.setTimeout(AP_TIMEOUT);

  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect(device.device_name)) {
    Serial.println("failed to connect and hit timeout");
    Serial.println("Reboot Your Device");
    delay(1000);
    ESP.restart();
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  strcpy(blynk.blynk_token, custom_blynk_token.getValue());
  strcpy(device.device_name, custom_device_name.getValue());

  eeprom_saveconfig();

  //end save

  Serial.println("local ip");
  Serial.println(WiFi.localIP());

  Blynk.config(blynk.blynk_token);
  bool result = Blynk.connect();

  if (result != true) {
    Serial.println("Failed To Connect BLYNK Server");
  } else {
    Serial.println("BLYNK Connected");
  }
}
void loop() {
  Blynk.run();
}

Muhammed-Hayr-Ali, 2019-03-22:
https://github.com/Muhammed-Hayr-Ali/wi-fi-Manager

megaESP, 2019-03-26:
Thank you very much Muhammed! Inspired from your example I put together this following snippet. However, I am not getting into 'shouldSaveConfig' condition  where EEPROM should be written. Can you please shed some light on it? I would really appreciate it. Thanks again

`#include <Arduino.h>
#include <EEPROM.h>
#include <WiFiManager.h>

char getParam[200]="someText";

bool shouldSaveConfig = false;
void configModeCallback (WiFiManager *myWiFiManager) {
    Serial.println("config mode");
    }
void saveConfigCallback () {
    Serial.println("Should save config now");
    shouldSaveConfig = true;
    }

void setup() {
     Serial.begin(115200);
     EEPROM.begin(512);
     Serial.println(EEPROM.readString(0));

    WiFiManager wm;
    bool res;
    wm.setSaveConfigCallback(saveConfigCallback);
    WiFiManagerParameter customParam("idParam", "Enter Param here", getParam, 200);
    wm.addParameter(&customParam);
    wm.setAPCallback(configModeCallback);
    res = wm.autoConnect("AutoConnectAP"); 

    if(!res) {
        Serial.println("Failed to connect");
    } 
    else {
        Serial.println("connected...yeey :)");
        strcpy(getParam, customParam.getValue());
        if (shouldSaveConfig) {
        EEPROM.writeString(0, getParam);delay(100);
        Serial.println(EEPROM.readString(0));
        }
    }

}

void loop() {
}
`

Muhammed-Hayr-Ali, 2019-03-27:
```
#define BLYNK_PRINT Serial
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <EEPROM.h>
#include <BlynkSimpleEsp8266.h>

WiFiManager wifiManager;

/****************************************************************/

//flag for saving data
bool shouldSaveConfig = false;

const unsigned long CONNECT_TIMEOUT = 10; // Wait 10 Seconds  to connect to the real AP before trying to boot the local AP
const unsigned long AP_TIMEOUT = 20; // Wait 20 Seconds in the config portal before trying again the original WiFi creds

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

/********************** Begin EEPROM Section *****************/
#define EEPROM_SALT 12664
typedef struct
{
  int   salt = EEPROM_SALT;
  char blynk_token[33]  = "";
  char device_name[20]  = "";
}

WMSettings;
WMSettings blynk;
WMSettings device;

void eeprom_read()
{
  EEPROM.begin(512);
  EEPROM.get(0, blynk);
  EEPROM.get(100, device);
  EEPROM.end();
}

void eeprom_saveconfig()
{
  EEPROM.begin(512);
  EEPROM.put(0, blynk);
  EEPROM.put(100, device);
  EEPROM.commit();
  EEPROM.end();
}

/*********************************************************************************/

void setup()
{
  Serial.begin(115200);

  /****************************************************************/

  eeprom_read();
  if (blynk.salt != EEPROM_SALT)
  {
    Serial.println("Invalid settings in EEPROM, trying with defaults");
    WMSettings defaults;
    blynk = defaults;
    device = defaults;
  }

  WiFiManagerParameter custom_blynk_token("blynk-token", "Blynk Token", blynk.blynk_token, 33);
  WiFiManagerParameter custom_device_name("device_name", "Device Name",  device.device_name, 20);

  WiFiManager wifiManager;
  wifiManager.setSaveConfigCallback(saveConfigCallback);
  wifiManager.addParameter(&custom_blynk_token);
  wifiManager.addParameter(&custom_device_name);

  //reset settings - for testing
  //wifiManager.resetSettings();
  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  wifiManager.setMinimumSignalQuality();
  //useful to make it all retry or go to sleep
  //in seconds
  wifiManager.setConnectTimeout(CONNECT_TIMEOUT);
  wifiManager.setTimeout(AP_TIMEOUT);

  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect(device.device_name)) {
    Serial.println("failed to connect and hit timeout");
    Serial.println("Reboot Your Device");
    delay(1000);
    ESP.restart();
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  strcpy(blynk.blynk_token, custom_blynk_token.getValue());
  strcpy(device.device_name, custom_device_name.getValue());

  eeprom_saveconfig();

  //end save

  Serial.println("local ip");
  Serial.println(WiFi.localIP());

  Blynk.config(blynk.blynk_token);
  bool result = Blynk.connect();

  if (result != true) {
    Serial.println("Failed To Connect BLYNK Server");
  } else {
    Serial.println("BLYNK Connected");
    Serial.println(blynk.blynk_token);
    Serial.println(device.device_name);
  }
}
void loop() {
  Blynk.run();
}
```

megaESP, 2019-03-29:
Thank you! Greatly appreciated
:-)

closed 2019-03-29 by megaESP

Anrikigai, 2019-10-19:
Sorry, is it possible to configure also a blynk host? 
I use my ow server, so Auth token is not enough...

lequangdau1993, 2020-03-29:
@Muhammed-Hayr-Ali GREAT CONTRIBUTION !!!

SerhiiZhilin, 2020-08-20:
> this is example for blynk auth parameter

In this case   eeprom_saveconfig(); is called each time after entering config no matter on shouldSaveConfig value?