tzapu/WiFiManager #856: Static IP wrong connection fix

dzanis, 2019-03-22:
Hi. I got the problem of static IP .If you enter it incorrectly or if the router is turned on DHCP, then even if you remove the settings of the static address, it still connects to it. Solved the problem before wifi connection enter  `WiFi.config(IPAddress(NULL), IPAddress(NULL), IPAddress(NULL)); `
Then added a condition to the code

https://github.com/tzapu/WiFiManager/blob/740005b21e5f5e5c6092dbd4d3037e4edee19113/WiFiManager.cpp#L280

```
 if (_sta_static_ip) {
    DEBUG_WM(F("Custom STA IP/GW/Subnet"));
    WiFi.config(_sta_static_ip, _sta_static_gw, _sta_static_sn);
    DEBUG_WM(WiFi.localIP());
  }
else
{
WiFi.config(IPAddress(NULL), IPAddress(NULL), IPAddress(NULL))
}
```

sorriso93, 2019-04-01:
Is it possible to configure also DNS with staticip config?

tablatronix, 2019-04-01:
What do you mean “remove the settings”?

dzanis, 2019-04-02:
If you incorrectly enter the address of the router, then do not connect until you delete the static ip settings can not connect.

tablatronix, 2019-04-02:
I still do not understand can you explain in detailed steps to reproduce.

Can you not change or remove the static settings in the portal ?

dzanis, 2019-04-06:
Yes, I can not change the static settings on the portal, if DHTP is enabled on the router. I decided to make it easier to use the scanner ip address to find what address the esp was connected to

tablatronix, 2019-04-06:
I have no idea what you are doing. You should be able to just delete static ip settings if not using them and just use dhcp, where is the problem actually occuring?

dzanis, 2019-04-06:
Of course, I can delete static ip settings in the code. The problem is that without a pre-project this cannot be done.

tablatronix, 2019-04-07:
Right so what exactly happens when you erase in portal?? You say it still uses it, can you add a toggle or something?

dzanis, 2019-04-07:
You can enter the AP MODE only by turning off the router. After you can reconnect the connection. But after you entered the wrong static ip, it will no longer connect to the network again, even with the correct settings. The problem is solved only if you change the ssid in the router.

tablatronix, 2019-04-07:
you can enter the ap mode whenever you want if you add a button or something to do it, or add an erase config method, 

I will have to test this cause I still am not getting the whole picture.

So if you have a short sketch or example.

Do you think the esp is remembering the static ip ?

Or do you think the web form is not saving when it is empty ?

dzanis, 2019-04-07:
If incorrectly entered static IP,then I will go through the access point to change the connection,settings are not reset, the output shows
`Already connected. Bailing out.`
Because the previous settings are reset only if connected, in this place the code
https://github.com/tzapu/WiFiManager/blob/740005b21e5f5e5c6092dbd4d3037e4edee19113/WiFiManager.cpp#L201

To reset the previous settings, you need to erase the settings. But they do not erase until you connect. That's why you need to connect to another network. But not everyone has a few open networks. For example, I have one. And I cannot connect to my own because the static ip address for it was incorrectly configured.
Ansvers:
The sketch is too big, if I separate it, I can show it.
It turns out that he remembers.
web form is fine.

dzanis, 2019-04-07:
I decided that it is better not to use a static address, because many routers are enabled DHCP
It's easier to find the address of the esp program  https://github.com/dzanis/IPScanner

tablatronix, 2019-04-07:
Oh you are still using master branch? Yeah that is a known issue if the wifi is the same it just ignores the save

Berzegeri, 2019-04-20:
> Is it possible to configure also DNS with staticip config?

Hi,
yes, but it's redundant a bit... 
After you set the static IP, only thing you need to do is to set it with WiFi.config().

in setup:
`WiFiManager wifiManager;`
`wifiManager.setSTAStaticIPConfig(IPAddress(192,168,0,110), IPAddress(192,168,0,1), 
      IPAddress(255,255,255,0));`
`WiFi.config(ip,gateway,subnet,dns1,dns2);`