tzapu/WiFiManager #859: AP connection not saved up on power failures

scooty334, 2019-03-27:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x ] Master
- [ ] Development

**Esp8266/Esp32:**
- [x ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x ] 2.4.0
- [ ] staging (master/dev)

### Description

hi guys, i have 2 problems that i looking for solutions. 

1, how can i make sure the AP credentials got saved, lets say if power failures happens ? 
but got deleted up on pressing the reset button on the side of the module ?
right now either way just clear the credentials. 

2, if the password didn't entered correctly the screen stays at (Please connect to 123's WIFI)
what i have to add to show a massage that "wifi password failed plz restart" 
please see debug info 

### Settings in IDE

Module:    LOLIN D1 mini V3.1.0

Additional libraries: 
#include "SPI.h"
#include "TFT_22_ILI9225.h"
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <WiFiManager.h>    

### Sketch

```cpp

#include <Arduino.h>

void setup() {

 
 WiFiManager wifiManager;

 tft.setFont(Terminal6x8);
 tft.drawText(0, 10, "Please connect to 123's WIFI");
 

 //delay(500);
 //}

    if (!wifiManager.startConfigPortal("123")) {
      Serial.println("failed to connect and hit timeout");
      ESP.reset();
      delay(5000);
    }

    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
    tft.setFont(Terminal6x8);
    tft.drawText(10, 25, "SUCCESS!!!!");
    delay(1000);
    tft.clear();

 

}

void loop() {

}
```

### Debug Messages

Problem #2 

*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 4
*WM: Failed to connect.
*WM: Request redirected to captive portal
*WM: Handle root

tablatronix, 2019-03-27:
Simple dont clear the settings and start configportal

scooty334, 2019-03-28:
i didn't clear settings thats the thing.  
also i still have the issue that fail to reconnect when turned off for some reason.

once i use ESP.eraseConfig();  it works again

tablatronix, 2019-03-28:
You said clear credentials by pressing reset on side of device?

If your credentials are not working everytime, then your flash might be corrupt, do a full erase from esptool or arduino

scooty334, 2019-03-28:
it turns out i had issue with the latest release of esp8266 v2.5.0
when i downgraded back to v2.4.0 works fine everything. 

is there any way to configure the wifimanager without a 3rd device? 
i have touchscreen lolin tft 2.4 , was wondering if possible to bring up a keyboard on a screen and enter the credentials that get saved later.

tablatronix, 2019-03-28:
Yeah but then why would you need wm

scooty334, 2019-03-28:
i guess if there is a way for it, i might dont need wm, can you share witch one you think of ?
thanks

tablatronix, 2019-03-28:
Wm is a webportal for entering settings if younare going to ise a touchscreen the what is the point of using wm?

Not sure what you mean by a third device?

scooty334, 2019-03-28:
if you use wm you need a 3rd party device. 
i already told you, im not necessarily want to use wm if i can make it with touchscreen. my question is still up.

tablatronix, 2019-03-29:
You mean a second device to connect to the websserver. 

No idea you would need a popup keyboard generator and save the inouts and call begin(ssid,password) this is all normal esp stuff not wm