tzapu/WiFiManager #863: Problem to Save on FS

leofuscaldi, 2019-04-02:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [x ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12F,**
- [ ] ESP01
- [x ] ESP12 E/F/S (**nodemcu**, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Cannot save anything on ESP12F. It seems that  it fails to mount FS.

```
mounting FS...
failed to mount FS
```

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch is the AutoConnectWithFSParametersAndCustomIP where I`ve changed the IP and gateway.

```
#include <FS.h>                   //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

//define your default values here, if there are different values in config.json, they are overwritten.
char mqtt_server[40];
char mqtt_port[6] = "1883";
char blynk_token[34] = "YOUR_BLYNK_TOKEN";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

  //clean FS, for testing
[... 219 lines cut ...]
*WM: [1] STA IP set: 192.168.8.88
*WM: [3] WIFI station disconnect 
*WM: [1] Connecting to new AP: $ecreta
*WM: [3] Using Password: $3nh4$3gur4
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] Connect to new AP [SUCCESS] 
*WM: [1] Got IP Address: 
*WM: [1] 192.168.8.88 
Should save config
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
connected...yeey :)
saving config
failed to open config file for writing
{"mqtt_server":"Teste.xxx.com","mqtt_port":"1883","blynk_token":"YOUR_BLYNK_TOKEN"}local ip
192.168.8.88
*WM: [3] freeing allocated params! 
*WM: [3] unloading

tablatronix, 2019-04-02:
did you do a full flash erase?

leofuscaldi, 2019-04-02:
I just found the problem... looking for my core version, I found that it was on 2.5.

I've downgraded it to 2.4 and it worked!

tablatronix, 2019-04-02:
The memory might have changed you may need a full erase when switching versions

leofuscaldi, 2019-04-02:
I've tried to full erase it, but made no diference. Only when I swiched back to 2.4 it worked.

Albin76, 2019-04-10:
Hi. Same problem for me and also solved by downgrading esp8266 core from 2.5.0 to 2.4.0 thanks to @leofuscaldi comment! Thanks a lot @leofuscaldi, made my day! Have been troubleshooting for some time now. :)

supriyono01, 2019-05-05:
Hi, I experienced the same thing. I have downgraded version 8266 from 2.5 to 2.4 but have not been able to solve the problem.

tablatronix, 2019-05-06:
have you guys tried a basic spiffs example, NOT wm?

tablatronix, 2019-05-07:
2.5 release notes say that build flags changed and is a breaking change, are your arduino IDEs up to date, maybe spiffs is now disabled

supriyono01, 2019-05-07:
Ok...thankyou,  I will try to update my arduino ide.

DanwithArduino, 2019-07-30:
Having issues with this as well. Downgrading to 2.4.x (all versions) breaks my MQTT/pubsub client as well. So, I can either have functioning FS or functioning MQTT... But I can't figure out what the changes are. 
ESP12F, core versions 2.4.x and 2.5.1. WifiManager development branch.

tablatronix, 2019-07-30:
Hmm which default arduino examples have you tried ? i will test them

DanwithArduino, 2019-07-30:
Might not be WiFiManager's problem. I just tried https://github.com/bblanchon/ArduinoJson/blob/6.x/examples/JsonConfigFile/JsonConfigFile.ino and am still unable to mount the file system.

tablatronix, 2019-07-30:
That is for SD cards is it not?

DanwithArduino, 2019-07-30:
Found the solution: In the Arduino IDE (or presumably the board manager in VScode but have not verified - should be same functionality) go to tools\Flash Size\choose one.... So far, I am getting positive results for 4M (1M SPIFFS).

tablatronix, 2019-07-31:
Yeah that is typically the setting you use for spiffs, unless we are having a memory issue here?

DanwithArduino, 2019-07-31:
I had `4M (No Spiffs)` before. Still learnding. Thanks for the replies. I think I'm all good now after a few days of head banging.

carlesgutierrez, 2019-09-10:
Hello, 
 I'm not able to write or read using SPIFFS. 
 I've succesfully checked before that SPIFFS works on a regular example without the wifiManager. 
 How I can check if It FAILS to mount FS? 

My configuration: 
 - nodemcu v2 / v3
 - ESP Core Version: 2.4 or 2.5

tablatronix, 2019-09-10:
do you have spiffs enabled, I think it is now optional not default.

carlesgutierrez, 2019-09-10:
Hey,
 If do you mean by enable:  **mode to Flash Size** to "4M (1M SPIFFS)", yes I did

tablatronix, 2019-09-10:
` if (SPIFFS.begin()) `
should do that.

hmm

carlesgutierrez, 2019-09-10:
You are absolutelly right. I had  forgotten to setup SPIFFS with **SPIFFS.begin()**
Thanks!!

tablatronix, 2019-09-10:
might also want to close spiffs after using it

carlesgutierrez, 2019-09-10:
Thanks for the advise @tablatronix , I will.

cunhaeduardo, 2019-12-19:
Hi guys! Today I have the same problem as you and I solved it changing configuration in tool -> flash size -> 4M(1M SPIFFS). Someone posted it but I'm doing it again to be easier to find!

Zadkiel2, 2020-02-07:
Hi guys! Please help me with the project I am building now. The project needs codes like this but I had problems dealing with ArduinoJson 6 and I am really new to these kind of things  so I really have no idea about it. Hope you guys help me out. Salamat.