tzapu/WiFiManager #867: No way to remove existing AP's

adamsstephen, 2019-04-17:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [X ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [X ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.5.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [X ] staging (master/dev)

### Description

Problem description
There is no way to delete existing connected AP's, such as Home AP's used for testing on ESP-01, so autoconnect sees this and does not look for the AP I want to logon to as it is already connected.
How can you delete existing AP's or delete ALL AP's as part of the sketch, so only the required AP is searched for?.
This is also a security issue on commercial products as it leaves a test ssid and password available in the device.
### Settings in IDE
[image]
Module:  			Generic ESP8266 module

Additional libraries: 
#include <ESP8266WiFi.h>
//autoconnect needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include "WiFiManager.h"   

### Sketch
// Not relevent

void setup() {

}

void loop() {

}
```

### Debug Messages
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.26
*WM: Already connected. Bailing out. 
*WM: IP Address:
*WM: 192.168.1.26
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.26
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.1.26
connected...yeey :)
*WM: freeing allocated params!

messages here
```

agrath, 2019-04-17:
@adamsstephen you can call `resetSettings` on the `WiFiManager wifiManager` to clear saved credentials and access points. 
or, if the device is booted and the existing access points are not available, you'll get a captive portal requesting the credentials for available to join.

adamsstephen, 2019-04-17:
Thanks, that works brilliantly.
Please can you add it to the documentation or How t?, I can only find it in the examples.

closed 2019-04-17 by adamsstephen

tablatronix, 2019-04-17:
The esp only stores 1 credential at a time.. you can either erase flash from arduino IDE or esptool.