tzapu/WiFiManager #871: ESP8266 - WiFiManager, cannot connect to router

panosss, 2019-04-22:
I 've made an ESP8266 device(a thermometer).
Uses WiFiManager library.
It works fine with my router.

BUT, at the client, I do everything as expected (turn to Access Point, Enter credentials, save, reboot ESP back to normal) but **does not send data to thingspeak**.
Like if it cannot connect to the internet.
(the password is 100% correct)

My client's router is a 'Thomson' brand name router.
Does ESP face any problems connecting to some specific routers?

Let me tell you what my client told me:
-his router accepts IPs from 192.﻿﻿168.1.2 up to 192.168.1.253﻿ (254 is the default which is used to enter in the web interface)
-the ESP tries to connect with IP 192.168.4.1, which is out of the router 's range, so it rejects it.

Does this make any sense? 

What can I do?
Any help is welcome.

scropion86, 2019-04-22:
i think you are confused 

- the ESP in AP mode will have 192.168.4.1 by default and you explained that you already provided the correct credentials.

- after that ESP should restart and connect to your router mainly in DHCP will provide a valid IP address to your ESP and not 192.168.4.1 unless you have specified that in your code then router will not join your ESP and you have to correct that in your code.

- make sure your ESP joined and got an IP.

- if the above is not the issue you may check your code or thingspeak credentials maybe the issue in that part of the code.

panosss, 2019-04-22:
You are right, I was confused. I didn't know that DCHP provides the IP to the ESP.
And things my client told me made me think that it's the ESP that chooses the IP.

I have not specified in my code any IP address.

(in order to make sure the ESP joined the router I 'll have to experiment at my client's place, this is something I 'd like to avoid. And will do only if I can't do otherwise)

The credentials for thingspeak are the same as the ones I used at my router, where it works.

(I think my only option is to experiment at my clients router :(, if you have any other idea it 'll be welcome)

panosss, 2019-04-23:
In case it helps somebody with a similar problem:
After reading [this thread](https://github.com/esp8266/Arduino/issues/2795) I told my client to

change:
1. Authentication method to WPA2-Auto
2. WPA encryption to: (AES+TKIP)
3. SSID in something that has 6 characters at least.(his SSID had only 4)

He did all these 3 changes and now it works!
I don't know which one was the problem.

chmpe, 2019-04-27:
The problem is with DHCP.
The only way to fix the problem was with static settings

IPAddress ip(192, 168, 29, 15);
IPAddress gateway(192, 168, 29, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress dns(8,8,8,8);
WiFi.config(ip, dns, gateway, subnet)
WiFi.begin(ssid, password);

tablatronix, 2022-07-21:
Sounds like dhcp is not working properly , not a wm issue