tzapu/WiFiManager #872: What is the best way to display the Ip address when connected to wifi?

TechMaxed, 2019-04-25:
I have tried many ways to do this such as js to change the value of textbox to ip and automatically redirecting the webpage to the ip but then both failed. What is the best way to show the ip without using Serial Monitor and connect it to an android device? Thanks.

tablatronix, 2019-04-25:
Not sure I understand, display where?

If you do not know the ip, then a webpage is useless, you could try using mdns and hostname

TechMaxed, 2019-04-27:
@tablatronix Thanks for replying. What I am trying to say is how do I display the ip of the esp8266 when it connects to the internet. What is the best way to display it in the webpage? Right now, I am tried to use js to update it but then it doesn't work. I will share to you my WiFiManager so that you can see as well. Thank you :)

TechMaxed, 2019-04-27:
I invited you to see the edited version. Thanks

tablatronix, 2019-04-27:
you cannot, this is already discussed at length elsewhere, the ap is shutdown when connecting to save

TechMaxed, 2019-04-27:
How would I be able to connect this to an android phone if I cannot show the ip address? Is there a way to tell the android phone like this is the address? I want the esp8266 to connect like how ewelink does it but then I don't know how I could give the ip to the android phone. Please help me Thanks. I will try to help you in other programming languages in return or like help you make your website better :)

TechMaxed, 2019-04-30:
@tablatronix I have been stuck for a while and I don't know why esp8266.local doesn't respond but then the ip responds. Here is the code I got from the example: #include <ESP8266WiFi.h>
```
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>;
 
// change these values to match your network
char ssid[] = "SSID";       //  your network SSID (name)
char pass[] = "password";    // your network password
 
WiFiServer server(80);
 
String header = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n";
String html_1 = "<!DOCTYPE html><html><head><meta name='viewport' content='width=device-width, initial-scale=1.0'/><meta charset='utf-8'><style>body {font-size:140%;} #main {display: table; margin: auto;  padding: 0 10px 0 10px; } h2,{text-align:center; } .button { padding:10px 10px 10px 10px; width:100%;  background-color: #4CAF50; font-size: 120%;}</style><title>LED Control</title></head><body><div id='main'><h2>LED Control</h2>";
String html_2 = "";
String html_4 = "</div></body></html>";
 
String request = "";
int LED_Pin = D1;
 
void setup() 
{
    pinMode(LED_Pin, OUTPUT); 
 
      Serial.begin(9600);
      delay(500);
      Serial.println(F("Serial started at 9600"));
      Serial.println();
 
      // We start by connecting to a WiFi network
      Serial.print(F("Connecting to "));  Serial.println(ssid);
 
 
      WiFi.begin(ssid, pass);
 
      while (WiFi.status() != WL_CONNECTED) 
      {
          Serial.print(".");    delay(500);
      }
      Serial.println("");
      Serial.println(F("[CONNECTED]"));
      Serial.print("[IP ");              
      Serial.print(WiFi.localIP()); 
      Serial.println("]");
      if (!MDNS.begin("esp8266"))   {  Serial.println("Error setting up MDNS responder!");  }
      else                          {  Serial.println("mDNS responder started");  }
 
      // start a server
      server.begin();
      Serial.println("Server started");
 
} // void setup()
 
 
 
void loop() 
{
 
    // Check if a client has connected
    WiFiClient client = server.available();
    if (!client)  {  return;  }
 
    // Read the first line of the request
    request = client.readStringUntil('\r');
 
    if       ( request.indexOf("LEDON") > 0 )  { digitalWrite(LED_Pin, HIGH);  }
    else if  ( request.indexOf("LEDOFF") > 0 ) { digitalWrite(LED_Pin, LOW);   }
 
 
    // Get the LED pin status and create the LED status message
    if (digitalRead(LED_Pin) == HIGH) 
    {
        // the LED is on so the button needs to say turn it off
       html_2 = "<form id='F1' action='LEDOFF'><input class='button' type='submit' value='Turn off the LED' ></form><br>";
    }
    else                              
    {
        // the LED is off so the button needs to say turn it on
        html_2 = "<form id='F1' action='LEDON'><input class='button' type='submit' value='Turn on the LED' ></form><br>";
    }
 
 
    client.flush();
 
    client.print( header );
    client.print( html_1 );    
    client.print( html_2 );
    client.print( html_4);
 
    delay(5);
  // The client will actually be disconnected when the function returns and 'client' object is detroyed
 
} 
```

tablatronix, 2019-12-04:
I think you are now required to run MDNS.update in loop to service it, not sure

chandra358566, 2023-03-17:
this program ok but
how connect two led control?