tzapu/WiFiManager #874: How to disable *WM message on serial ?

anujmattoo, 2019-04-29:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

tablatronix, 2019-04-29:
Disable debugging it should be on the readme

TechMaxed, 2019-04-30:
edit the cpp file. It is written there.

tablatronix, 2019-05-01:
Do not edit cpp

setDebugOutput()