tzapu/WiFiManager #877: Connect to an open WiFi that uses a web portal login

Elessar-lad, 2019-05-01:
#### Hardware
**WiFimanager Branch/Release:**
- [X] Master
- [ ] Development

**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [X] 2.4.0
- [ ] staging (master/dev)

### Description

Hi

I'd like to know if it is possible to connect to a WiFi network that requires authentication via the browser? The network itself has no password but after you connect to the network and open the internet browser, you're welcomed by a connect screen. 

It works great in other network. 

### Settings in IDE

Module: Wemos D1

tablatronix, 2019-05-01:
not at this time

Elessar-lad, 2019-05-02:
Got it! I'll find ways around it, but thank you so much for confirming I was going crazy trying to figure it out.

tablatronix, 2019-05-02:
it should be possible , but you have to write it yourself, you might have to intercept the page and get any tokens or nonces, if it uses https who knows, afaik no one has made something for that yet, easiest would be to just log and replay a login and see what needs to change each time