tzapu/WiFiManager #879: ESP32 seems that wifimanager does not work

mikekgr, 2019-05-02:
**WiFimanager Branch/Release:**
- [ ] Master
- [*] Development latest version of git

**Esp8266/Esp32:**
- [ ] ESP8266
- [*] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [*] Other ESP32 DEV BOARD, default Arduino settings

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [*] ESP 32 Arduino Core 1.0.2 release

Problem description:
WiFi Manager, although is compiled and run without obvious problems, it doesn't show the captive screen on my Android phones, tablet as well as, my Windows 10 laptop. I tested also the direct IP address in my browser ( 192.168.4.1 ) it shows nothing. All the above even using the simplest example Sketch: basic.ino
The power supply of the ESP32 DEV BOARD is a trusted PSU. I tested many times...
If anyone has the same simple I think setup, kindly requesting to test it and let me know the result.
Please tell me what to do or check?
Sorry, I don't have serial printings.

Thanks and Best Regards,
Mike Kranidis

mikekgr, 2019-05-05:
Anyone to help me regarding the reported issue?

tablatronix, 2019-05-06:
Try the basic captiveportal arduino example and see if you have a problem with it also

mikekgr, 2019-05-06:
Dear @tablatronix ,
what sketch do you mean?
As I wrote above, I have tried unsuccessfully the:
https://github.com/tzapu/WiFiManager/blob/development/examples/Basic/Basic.ino

If you indicate another one, give me the link please.
Thanks !

tablatronix, 2019-05-06:
there should be on in the esp examples

bfaliszek, 2019-05-24:
I checked with captiveportal from examples and it also does not work. iPhone sees the network, it connects for a while, but nothing appears and disconnects the network. I have identical in WiFiManager.

bfaliszek, 2019-05-24:
The same .ino (https://github.com/tzapu/WiFiManager/blob/development/examples/NonBlocking/AutoConnectNonBlocking/AutoConnectNonBlocking.ino) loaded on ESP8266 also does not work. It behaves the same as on ESP32.

tablatronix, 2019-05-24:
#iphone

closed 2019-05-24 by tablatronix

kbickham, 2019-08-20:
I've got it working but having problems with something unrelated. about the captive portal...on iOS you need to change settings. Below network list there is an option to "ask to join session" or something. A lot of Android devices will work if use Google ..and setb 8.8.8.8 or 8.8.4.4 for up address and edit your server libs to use port 53. I