tzapu/WiFiManager #880: crash when trying to connect to MQTT server

TinajaLabs, 2019-05-05:
Getting a crash on a wemos d1 mini (and also on the Feather Huzzah ESP8266 ) after a few minutes of the device trying to reconnect to the wifi router. The router's running openwrt and is about 8 feet away, so I doubt it's a problem with OTA connectivity. 

I'm trying to make a room sensor with a collection of sensors (temp, humidity, pir, light level) and associated libraries - `WifiManager` ("version": "1.0.0" - from library.json), `MQTT` (pupsub), `DHT22` (temp/humidity), `TSL2561 (lux)`, and a `PIR` sensor.

Below is the list of libraries I use to accomplish this.  I'm trying to understand any interaction between these libs.  I've read that the FS library needs to always be first one defined.  Are there any others that need to be sequenced in the order?

The code compiles fine and doesn't seem to be related to a memory problem:
```
Sketch uses 383968 bytes (36%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33948 bytes (41%) of dynamic memory, leaving 47972 bytes for local variables. Maximum is 81920 bytes.
```

Thanks for any tips,

Chris.

```
#include <FS.h> //this needs to be first, or it all crashes and burns...

#include <Wire.h>
#include <Adafruit_Sensor.h>  // Adafruit Unified Sensor Driver - https://github.com/adafruit/Adafruit_Sensor

#include <DNSServer.h>  
#include <ESP8266WiFi.h>      // https://github.com/esp8266/Arduino
#include <WiFiManager.h>      // https://github.com/tzapu/WiFiManager
#include <ESP8266WebServer.h> //Local WebServer used to serve the configuration portal
#include <ArduinoJson.h>      //https://github.com/bblanchon/ArduinoJson

#include <PubSubClient.h>
#ifdef ESP32
#include <SPIFFS.h>
#endif

#include <Adafruit_TSL2561_U.h>
#include <DHT.h>
```

The error stack...  in case it's meaningful to anyone. First some normal output, then a few dozen attempts to reconnect then - boom.
```
Attempting connection to MQTT server: 192.168.1.138...  Not connected.

=========================================================

mqtt pub - PIR state:	0
mqtt pub - RSSI level: -50
mqtt pub - temperature:	73.94 *F
mqtt pub - rel humidity:	60.20 %
mqtt pub - lux value:		0
mqtt pub - lux broaband:	24
mqtt pub - lux infrared:	5
mqttClient not connected...

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3fffed40 end: 3fffffb0 offset: 01b0
3fffeef0:  00000000 127f2f22 3ffed770 40222016  
3fffef00:  00000000 3ffee088 00000000 00000000  
3fffef10:  40240ac4 3ffee088 3ffef040 40201ce8  
3fffef20:  4024353d 3ffee088 3ffef040 3b3138ed  
3fffef30:  40243582 3fffdab0 00000000 3fffdcb0  
3fffef40:  3ffef050 3fffdad0 3ffefd84 40216ec7  
3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 40001878 00000002 00000000  
3fffef70:  3fffff10 aa55aa55 000000f2 4010542c  
3fffef80:  40105432 00000002 00000000 e82bcda6  
3fffef90:  4010000d a3ca2c98 dda558a1 83153daf  
3fffefa0:  401015fc 3fffef3c 401015a9 3fffff58  
[... 233 lines cut ...]
3ffffe40:  3ffef96a 3ffef968 3ffef724 4021920c  
3ffffe50:  42100000 00000001 3ffef724 4021500c  
3ffffe60:  00000000 4bc6a7f0 000f2739 3ffefd84  
3ffffe70:  00000000 00000000 4bc6a7f0 00000000  
3ffffe80:  42100000 3ffef724 40100a9e ba1cac08  
3ffffe90:  00000000 3ffffeb0 3ffefc74 40215bfc  
3ffffea0:  3fffdad0 3ffef720 00000001 40208213  
3ffffeb0:  00000035 00000028 3ffffedd 3ffef724  
3ffffec0:  00000010 3fffff60 3fffff60 402162cf  
3ffffed0:  3fffff20 00000005 3fffff60 4021631b  
3ffffee0:  34320002 00000000 3fffff60 40216364  
3ffffef0:  00000000 4bc6a7f0 000f2739 3ffefd84  
3fffff00:  00000000 00000000 4bc6a7f0 00000000  
3fffff10:  00000000 4bc6a7f0 000f2739 3ffefd84  
3fffff20:  00000000 00000000 4bc6a7f0 00000000  
3fffff30:  3fffdad0 3ffef720 40100a9e b8d4fdf3  
3fffff40:  40105d48 127fbd7f 3ffefe5c 00000000  
3fffff50:  3ffef040 3ffefe5c 3ffe87fc 3ffefe5c  
3fffff60:  3fffdad0 3ffefd84 40216f0c 3fffefa0  
3fffff70:  3ffefe5c 3fffdad0 00000064 40201d6f  
3fffff80:  3fffdad0 3ffef720 00000001 4020a1b6  
3fffff90:  00000000 00000000 00000001 40216f2d  
3fffffa0:  3fffdad0 00000000 3ffefd54 40216fb8  
<<<stack<<<
```

TinajaLabs, 2019-05-05:
Just happened again, and this is the lead up to the crash:
```
mqtt pub - PIR state:	1
mqtt pub - RSSI level: -56
mqtt pub - temperature:		71.96 *F
mqtt pub - rel humidity:	59.40 %
mqtt pub - lux value:		0
mqtt pub - lux broaband:	17
mqtt pub - lux infrared:	3
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqtt pub - PIR state:	1
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqtt pub - PIR state:	1
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...
mqttClient not connected...

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3fffed40 end: 3fffffb0 offset: 01b0
.
.
.
```

tablatronix, 2019-05-06:
see if you have time to decode the stack trace, it will tell you exactly where its crashing

TinajaLabs, 2019-05-08:
Just learned how to decode the stack trace.  Please see below.  This is a more recent crash, not from the prior post.

It is revealing.  I added a check with `Serial.println(ESP.getFreeHeap());` and it shows a slow reduction in free memory.

Is there a trick to tell where the memory leak is happening?

Thank you,
Chris.

```
Memory allocation of 116 bytes failed at 0x4022ad8c: mem_malloc at core/mem.c line 210

Decoding stack results
0x40222086: memp_free at core/memp.c line 447
0x402149ef: loop_task(ETSEvent*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 133
0x40213159: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 634
0x402158ec: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 161
0x40213159: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 634
0x402158ec: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 161
0x40213159: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 634
0x40215871: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 127
0x40214a34: esp_yield() at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 91
0x40100119: __wrap_spi_flash_read at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_phy.c line 267
0x40215788: spiffs_hal_read(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 67
0x4021fd36: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x4021fc70: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 180
0x4021b830: _printf_i at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf_i.c line 194
0x4021fd36: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x4021fc70: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 180
0x4021b930: _printf_i at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf_i.c line 244
0x4021fd36: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x40220147: _svfprintf_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 660
0x4021fc70: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 180
0x40101564: free at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/umm_malloc/umm_malloc.c line 1760
0x40202291: uart_do_write_char at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/uart.c line 450
0x40202652: uart_write at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/uart.c line 496
0x40101564: free at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/umm_malloc/umm_malloc.c line 1760
0x4020b2a4: ClientContext::_s_connected(void*, tcp_pcb*, long) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/include/ClientContext.h line 635
0x402217ad: operator delete(void*) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/del_op.cc line 48
0x4020b4ab: ClientContext::unref() at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/include/ClientContext.h line 123
0x40100a9e: millis at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x4020b004: WiFiClient::connect(char const*, unsigned short) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClient.cpp line 133
0x402081e3: publishLuxValues() at /home/chrisj/Documents/code/arduino/tinaja/roomSensor-wfm-lux-temp-hum-01/roomSensor-wfm-lux-temp-hum-01.ino line 844
0x40219150: PubSubClient::connected() at /home/chrisj/Documents/code/arduino/libraries/PubSubClient/src/PubSubClient.cpp line 606
0x4021337c: HardwareSerial::write(unsigned char const*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/HardwareSerial.h line 174
0x40213681: Print::write(char const*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Print.h line 60
0x40100a9e: millis at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x40214a34: esp_yield() at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 91
0x40201d6f: delay at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 51
0x4020a1e2: loop() at /home/chrisj/Documents/code/arduino/tinaja/roomSensor-wfm-lux-temp-hum-01/roomSensor-wfm-lux-temp-hum-01.ino line 289
0x40214ae0: loop_wrapper() at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 125
```

TinajaLabs, 2019-05-08:
Happened again when It was `Attempting connection to MQTT server: 192.168.1.138...`

Heapfree starts at `39344` and is `33344` just before the crash.  It took less than 5 minutes to happen and 

```
Decoding stack results
0x40214a7f: loop_task(ETSEvent*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 133
0x40213211: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 634
0x4021597c: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 161
0x40213211: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 634
0x4021597c: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 161
0x40213211: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 634
0x40215901: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 127
0x40214ac4: esp_yield() at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 91
0x40100119: __wrap_spi_flash_read at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_phy.c line 267
0x40213235: EspClass::flashRead(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 639
0x40215818: spiffs_hal_read(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 67
0x4020699d: spiffs_phys_rd at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs/spiffs_cache.c line 153
0x402219e9: glue2esp_linkoutput at glue-esp/lwip-esp.c line 299
0x402219e9: glue2esp_linkoutput at glue-esp/lwip-esp.c line 299
0x4021fdc6: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x40221c7a: new_linkoutput at glue-lwip/lwip-git.c line 238
0x4022205c: ethernet_output at netif/ethernet.c line 312
0x40229734: etharp_output_to_arp_index at core/ipv4/etharp.c line 770
0x40229938: etharp_output_LWIP2 at core/ipv4/etharp.c line 885
0x402201d7: _svfprintf_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 660
0x4022a350: ip4_output_if_opt_src at core/ipv4/ip4.c line 1007
0x4021fdc6: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x4021fd00: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 180
0x4022a398: ip4_output_if_opt at core/ipv4/ip4.c line 820
0x4022ae1c: mem_malloc at core/mem.c line 210
0x4022ae1c: mem_malloc at core/mem.c line 210
0x4022a3be: ip4_output_if at core/ipv4/ip4.c line 793
0x4022af2f: ip_chksum_pseudo at core/inet_chksum.c line 395
0x4022731e: tcp_output at core/tcp_out.c line 1361
0x40226609: tcp_create_segment at core/tcp_out.c line 190
0x40214ac4: esp_yield() at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 91
0x40201d6f: delay at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 51
0x4020bb6d: WiFiClient::connect(IPAddress const&, unsigned short) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/include/ClientContext.h line 136
0x40100a9e: millis at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x4020b0bc: WiFiClient::connect(char const*, unsigned short) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClient.cpp line 133
0x402191e0: PubSubClient::connected() at /home/chrisj/Documents/code/arduino/libraries/PubSubClient/src/PubSubClient.cpp line 606
0x4020c02b: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, unsigned char, char const*, unsigned char) at /home/chrisj/Documents/code/arduino/libraries/PubSubClient/src/PubSubClient.cpp line 129
0x40100a9e: millis at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x40213434: HardwareSerial::write(unsigned char const*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/HardwareSerial.h line 174
0x4020c214: PubSubClient::connect(char const*) at /home/chrisj/Documents/code/arduino/libraries/PubSubClient/src/PubSubClient.cpp line 106
0x4021374c: Print::print(char const*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Print.cpp line 122
0x40207d6f: mqttReconnected() at /home/chrisj/Documents/code/arduino/tinaja/roomSensor-wfm-lux-temp-hum-01/roomSensor-wfm-lux-temp-hum-01.ino line 447
0x40100a9e: millis at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x40100b31: __digitalWrite at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring_digital.c line 82
0x4020a262: loop() at /home/chrisj/Documents/code/arduino/tinaja/roomSensor-wfm-lux-temp-hum-01/roomSensor-wfm-lux-temp-hum-01.ino line 263
0x40214b70: loop_wrapper() at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 125
```

TinajaLabs, 2019-05-08:
And another after about an hour and 10 minutes of running OK:

Memory went from 40668 to 35408. It happened right after it's attempt to connect:
`TL: Attempting connection to MQTT server: 192.168.1.138...  Not connected, error code: -2`
... then
```
TL: Attempting connection to MQTT server: 192.168.1.138...
Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3fffed40 end: 3fffffb0 offset: 01b0
3fffeef0:  3ffe9be8 35e33b83 60000600 40222106  
etc.
```

```
Decoding stack results
0x40222106: memp_free at core/memp.c line 447
0x402131fd: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 634
0x40215968: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 161
0x402131fd: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 634
0x40215968: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 161
0x402131fd: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 634
0x402158ed: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 127
0x40213221: EspClass::flashRead(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 639
0x402219d9: glue2esp_linkoutput at glue-esp/lwip-esp.c line 299
0x402219d9: glue2esp_linkoutput at glue-esp/lwip-esp.c line 299
0x40221c6a: new_linkoutput at glue-lwip/lwip-git.c line 238
0x4022204c: ethernet_output at netif/ethernet.c line 312
0x4021fdb6: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x40229724: etharp_output_to_arp_index at core/ipv4/etharp.c line 770
0x4021b9b0: _printf_i at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf_i.c line 244
0x40229928: etharp_output_LWIP2 at core/ipv4/etharp.c line 885
0x4022a340: ip4_output_if_opt_src at core/ipv4/ip4.c line 1007
0x4021fdb6: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x4021fcf0: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 180
0x4022a388: ip4_output_if_opt at core/ipv4/ip4.c line 820
0x4022ae0c: mem_malloc at core/mem.c line 210
0x4022a3ae: ip4_output_if at core/ipv4/ip4.c line 793
0x4022af1f: ip_chksum_pseudo at core/inet_chksum.c line 395
0x4022730e: tcp_output at core/tcp_out.c line 1361
0x402238d9: sys_timeout_abs at core/timeouts.c line 189
0x40214ab0: esp_yield() at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 91
0x40201d6f: delay at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 51
0x4020bb59: WiFiClient::connect(IPAddress const&, unsigned short) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/include/ClientContext.h line 136
0x40100a9e: millis at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x4020b0a8: WiFiClient::connect(char const*, unsigned short) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClient.cpp line 133
0x40208237: publishLuxValues() at /home/chrisj/Documents/code/arduino/tinaja/roomSensor-wfm-lux-temp-hum-01/roomSensor-wfm-lux-temp-hum-01.ino line 855
0x402191cc: PubSubClient::connected() at /home/chrisj/Documents/code/arduino/libraries/PubSubClient/src/PubSubClient.cpp line 606
0x4020c017: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, unsigned char, char const*, unsigned char) at /home/chrisj/Documents/code/arduino/libraries/PubSubClient/src/PubSubClient.cpp line 129
0x40100a9e: millis at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x40213420: HardwareSerial::write(unsigned char const*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/HardwareSerial.h line 174
0x4020c200: PubSubClient::connect(char const*) at /home/chrisj/Documents/code/arduino/libraries/PubSubClient/src/PubSubClient.cpp line 106
0x40213738: Print::print(char const*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Print.cpp line 122
0x40207d6f: mqttReconnected() at /home/chrisj/Documents/code/arduino/tinaja/roomSensor-wfm-lux-temp-hum-01/roomSensor-wfm-lux-temp-hum-01.ino line 447
0x40100a9e: millis at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x40100b31: __digitalWrite at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring_digital.c line 82
0x4020a24e: loop() at /home/chrisj/Documents/code/arduino/tinaja/roomSensor-wfm-lux-temp-hum-01/roomSensor-wfm-lux-temp-hum-01.ino line 263
0x40214b5c: loop_wrapper() at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 125
```

tablatronix, 2019-05-08:
are you writing to spiffs over and over ?
What are you using spiffs for?

is there a loop that updates some variables than then get saved ?

TinajaLabs, 2019-05-08:
I do write to the file system with spiffs library to save 2 parameters (as json), the MQTT server, and the location part of the topic string (e.g., ) for the calls being made when published.  These are accessed in setup and during a reset when running wifimanager.

My code for my room sensor - PIR, TSL2561 (lux/light), DHT22 (temp, humidity), RSSI + MQTT and wifimanager.

It's a tedious bit of code, but I've broken out much of the bits into separate methods.

The key parts:

- void setup()
- void loop() 
- wifi manager methods
  - void checkWifiManagerButton()
  - void runWifiManager(String mode)
  - bool loadConfigFile()
  - bool saveConfigFile()

Variables:
- char topic_loc[30] = "maker/house/livingroom/";
- char mqtt_server[40] = "192.168.0.254";

I don't expect anyone to go through this code in any detail, but I appreciate any insights into where I'm going wrong here.

BTW, I'm hoping to show how this can work at Maker Faire next week.

Thank you,
Chris.

```C++
/*
   TinajaLabs.com, Chris Jefferies, May 2019
  ---------------------------------------------------------------
   roomSensor-wfm-lux-temp-hum-01

   Attributes:
   - WifiManager
   - FS
   - MQTT

   - RSSI: dbm, wifi signal strength
   - TSL2561: light level in lux
   - DHT22: temperature & humidity
   - pir: motion sensor
   - // bme280: temperature, humidity, barometric pressure
   - // tmp36: temperature

   Arduino IDE settings:
    Board: Feather HUZZAH ESP8266
    Flash Size: 1M (64k SPIFFS)
    lwIPVariant: v2 Lower Memory
    VTables: Flash
    CPU Frequency: 80MHz
    Upload Speed: 230400
    Port: dev/ttyUSB0, dev/ttyUSB1

    Board: LOLIN (WEMOS) D1, R2 & Mini
    Flash Size: 4M (64k SPIFFS)
    lwIPVariant: v2 Lower Memory
    VTables: Flash
    CPU Frequency: 80MHz
    Upload Speed: 921600
    Port: dev/ttyUSB0, dev/ttyUSB1

  ---------------------------------------------------------------
*/

#include <FS.h> // this needs to be first, or it all crashes and burns...
#include <Wire.h>
#include <ESP8266WiFi.h> // https://github.com/esp8266/Arduino
#include <ArduinoJson.h> // https://github.com/bblanchon/ArduinoJson
#include <PubSubClient.h>
[... 807 lines cut ...]

// ----------------------------------------------------------
void setTopicsPir()
{
  strcat(pir_topic, topic_loc);
  strcat(pir_topic, topic_device);
  strcat(pir_topic, chipID);
  strcat(pir_topic, "/");
  strcat(pir_topic, "pir/");
  strcat(pir_topic, pir_base);

  Serial.println(pir_topic);
}

// ========================== specilized methods

// ----------------------------------------------------------
void setChipId()
{
  snprintf(chipID, 10, "%ld", ESP.getChipId());
  Serial.print("TL: === Device id: ");
  Serial.println(chipID);
  // return chipID;
}
```

TinajaLabs, 2019-05-08:
Things I've done to try and fix the problem...

Aligned the variables associated with the payload value when publishing to MQTT.
Changed code like this:
```C++
char luxMsg[50];
    snprintf(luxMsg, 75, "%ld", lux);
```
to this:
```C++
char luxMsg[50];
    snprintf(luxMsg, 50, "%ld", lux);
```

Changed timer variables that change within the loop from `long` to `unsigned long`.
```C++
unsigned long now = millis();
unsigned long pubLastPir = 0;
```

tablatronix, 2019-05-09:
Lots of code lol 

Are you closing spiffs when done with it?
Are you passing any of those large char to a async pub func as refs 

Are you creating any objects on loops say wifimanager?

I would reduce the complexity, bug could be any one of those sensor libraries, make your code so you can disable large pieces maybe use includes or defines also add a debugging wrapper.

Put a debug log call at every benchmark function say debug func name then in debuglog you can output memory logging and try to narrow it down.

TinajaLabs, 2019-05-09:
[image]

TinajaLabs, 2019-05-09:
The above is an image of the free heap over the afternoon.  The code reboots after the crash.  Once the leak starts it cascades out over about 15 minutes and eventually chokes. 

The room sensor sends mqtt data (the publish routines) to node-red and node-red forwards to grafana.

Obviously I don't know where the leak is happening, but I'll look at your advice and see where I can inject a peek at the memory status.  The tedious part is that, as you can see in the chart, it takes about an hour to an hour and a half to start the memory leak cascade...  ugh.

Thank you.

tablatronix, 2019-05-09:
Yeah that is hard to debug, at least its not 24 hours.. 
maybe you can log what it was doing when the memory starts to change beyond a nominal value, and catch it that way, maybe you get a different response from something, maybe you hit a rate limit threshold

tablatronix, 2019-05-09:
I would also remove all string operations for testing, those strcat for example

tablatronix, 2019-05-09:
I would also suspect anything that you are doing repeatedly like getting something from a class or library, in case the leak in inside one of those, could even be in side esp lib.

ALSO
never use loops in esp
while (!Serial)
    ; // wait for serial monitor to open

these are all bad for the watch dog, make sure there is a yield in every loop

dontsovcmc, 2019-05-09:
@TinajaLabs a use MQTT with WiFiManager without problems. It can be a bug in char arrays operations...
This is a code in my project: https://github.com/dontsovcmc/waterius/blob/master/ESP8266/src/sender_mqtt.h

I don't use SPIFFS, maybe some heap problems...  
I also had a bugs with arrays and dangerous functions that don't check null terminators. 
And my favourite bug is unsigned/signed arithmetic... Every operations with millis() should be unsigned.. 

Refactor idea:
You needn't call setTopicsDevices 
You can build topic realtime.

```
int32_t rssiValue = WiFi.RSSI();
  char rssiMsg[50];
  // snprintf(rssiMsg, 75, "%ld", WiFi.RSSI());
  snprintf(rssiMsg, 75, "%ld", rssiValue);

  Serial.print("TL: mqtt pub - RSSI level: \t");
  Serial.println(rssiValue);
  mqttClient.publish(rssi_topic, rssiMsg);

Example:
String topic = topic_loc + topic_device + chipID + "/device/" + rssi_base;
String msg = String((uint16_t)WiFi.RSSI());
mqttClient.publish(topic, msg);

Example 2:
String root = topic_loc + topic_device + chipID;

bool mqtt_publish(const String &section, const String &title, const String &msg)  {
   LOG_DEBUG() << section << title << ": " << msg;
   mqttClient.publish(root + section + title, msg);
}

mqtt_publish("/device", "/rssi/state/room", String((uint16_t)WiFi.RSSI());
```

TinajaLabs, 2019-05-11:
Evgeny,

You have advised a refactoring similar to what I have just done.  I refactored all the parts where I used `snprintf` and it has been running for a few hours now without any change to the HeapFree value.  We'll see if it holds.

Instead of using the approach where I use something like this:
```
int16_t rssiValue = WiFi.RSSI();
char rssiMsg[50];
snprintf(rssiMsg, 50, "%ld", rssiValue);

Serial.print("TL: mqtt pub - RSSI level: \t");
Serial.println(rssiMsg);
mqttPublish(rssi_topic, rssiMsg, false);
```
I am using this:
```
int16_t rssiValue = WiFi.RSSI();
mqttPublish(rssi_topic, String(rssiValue).c_str(), true);
```

Also, your approach to create the concatenated topic string, real-time, with String(type) cast looks very efficient.  My thought was to create all the topic strings within the setup and thereafter when I need them, they already exist.

As you can probably see all of the topic string manipulation is so I can use wifimanager to update the location part in the field.  The rest of the topic is embedded and is associated with the set of sensors and values built into the device.

Thank you for your help,
Chris.

TinajaLabs, 2019-05-11:
Ach...

How does one read the stack?  Top to bottom, bottom to top? Is this an exact trace through to the actual lines where the error occurred?

I see references to spiffs which I executed early on when launching the application.  Perhaps that should be removed per Evgeny's suggestion.  What is the alternative to persisting data on esp8266?

Thanks.

```C++
0x40214adb: loop_task(ETSEvent*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 133
0x4021326d: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 634
0x402159d8: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 161
0x4021326d: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 634
0x402159d8: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 161
0x4021326d: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 634
0x4021595d: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 127
0x40214b20: esp_yield() at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 91
0x40100119: __wrap_spi_flash_read at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_phy.c line 267
0x40213291: EspClass::flashRead(unsigned int, unsigned int*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Esp.cpp line 639
0x40215874: spiffs_hal_read(unsigned int, unsigned int, unsigned char*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/spiffs_hal.cpp line 67
0x40221a41: glue2esp_linkoutput at glue-esp/lwip-esp.c line 299
0x40221cd2: new_linkoutput at glue-lwip/lwip-git.c line 238
0x402220b4: ethernet_output at netif/ethernet.c line 312
0x4021b581: _printf_common at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf_i.c line 94
0x40221a41: glue2esp_linkoutput at glue-esp/lwip-esp.c line 299
0x4022978c: etharp_output_to_arp_index at core/ipv4/etharp.c line 770
0x40229990: etharp_output_LWIP2 at core/ipv4/etharp.c line 885
0x4021fe1e: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x4021b918: _printf_i at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf_i.c line 194
0x4021fe1e: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x4021b918: _printf_i at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf_i.c line 194
0x4021fd58: __ssputs_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 180
0x4021ba18: _printf_i at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf_i.c line 244
0x4022022f: _svfprintf_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 660
0x40212c9a: DHT::read(bool) at /home/chrisj/Documents/code/arduino/libraries/DHT_sensor_library/DHT.cpp line 221
0x40202291: uart_do_write_char at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/uart.c line 450
0x4021c264: sprintf at ../../../../../../dl/newlib-xtensa/newlib/libc/stdio/sprintf.c line 646
0x40101401: realloc at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/umm_malloc/umm_malloc.c line 1504
0x40213edf: String::changeBuffer(unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 156
0x40213490: HardwareSerial::write(unsigned char const*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/HardwareSerial.h line 174
0x40213795: Print::write(char const*) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Print.h line 60
0x40213edf: String::changeBuffer(unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 156
0x40213f2b: String::reserve(unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 146
0x40219238: PubSubClient::connected() at /home/chrisj/Documents/code/arduino/libraries/PubSubClient/src/PubSubClient.cpp line 606
0x4020c294: PubSubClient::publish(char const*, unsigned char const*, unsigned int, unsigned char) at /home/chrisj/Documents/code/arduino/libraries/PubSubClient/src/PubSubClient.cpp line 399
0x40213490: HardwareSerial::write(unsigned char const*, unsigned int) at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/HardwareSerial.h line 174
0x40100a9e: millis at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x40208277: publishLuxValues() at /home/chrisj/Documents/code/arduino/tinaja/roomSensor-wfm-lux-temp-hum-01/roomSensor-wfm-lux-temp-hum-01.ino line 869
0x40214b20: esp_yield() at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 91
0x40201d6f: delay at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 51
0x4020a2f9: loop() at /home/chrisj/Documents/code/arduino/tinaja/roomSensor-wfm-lux-temp-hum-01/roomSensor-wfm-lux-temp-hum-01.ino line 295
0x40214bcc: loop_wrapper() at /home/chrisj/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 125
```

bkrajendra, 2019-05-12:
Hi @TinajaLabs , The best MQTT client i used till date is https://github.com/256dpi/arduino-mqtt
It does not have much features, but its light weight and has TLS support. Im using it with my mosquito broker in cloud..!

I also get these issues with Latest release of Arduino Esp8266 core and wifimanager not working together  but everytime there was some or other issue i resolved by using some best practices. 

Knowing the flow of code and laying things as per it.!

For example once I was having crashing issue with it, I figured it out was due to mDns. If i do mdns before mqtt begin, it used to crash. I resolved it by putting latter.

Next i did not understood but it was unable to connect if i use string for client id and topic. So made sure to convert those string with .c_str(), which resolved issue. I dont undesrstood the exact reson behind this or might be it was related to something else but it worked.

Recently everything was ok. But my server was refusing to connect. Now my code is similar to yours. Complex and long, im using for my IoT hardware... nothing was changed from previous code, but still it was refusing to subscribe or sometime wont connect.

After lot of tries i decide to put a scheduler delay to start mqtt connection after everything finishes on boot. using library [https://github.com/Toshik/TickerScheduler](https://github.com/Toshik/TickerScheduler)

And everything started working fine. 

Hope this might help you!

SteveRMann, 2019-08-14:
Don't know if you fixed this, but I just had the same issue.

In these lines:
```
WiFiClient myWifiClient;
PubSubClient mqttClient(myWifiClient);

```myWifiClient must be unique  within your LAN.  If you have another device connecting to PubSubClient, it'sclient name has to also be unique.

Ostrich-revolution-slide, 2025-02-04:
> Yeah that is hard to debug, at least its not 24 hours.. maybe you can log what it was doing when the memory starts to change beyond a nominal value, and catch it that way, maybe you get a different response from something, maybe you hit a rate limit threshold

https://stackdev.space/questions/converting-a-constant-character-hexadecimal-string-to-an-unsigned-character-in-c