tzapu/WiFiManager #881: How to re-connect Wifi in Loop function? if wifi is in sleep mode then how to connect saved network???

Urenthummar, 2019-05-05:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

tablatronix, 2019-05-06:
WiFi.begin()

Urenthummar, 2019-05-06:
Wifi.begin(); 
not working, i'm using it in function to wake nodemcu & connect to Wifi network.  
void wifiConnect() {
  WiFi.forceSleepWake();
  WiFi.mode(WIFI_STA);
  delay(10);
  WiFi.begin();
  Serial.print("Connecting to ");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

Daemach, 2019-05-06:
How can WiFi Manager turn into a complete wifi management solution?  In a
perfect world it would combine the best of wifimulti and wifi manager, and
hit the following bullet points:

   - Attempt to connect to (potentially multiple) networks with provided
   credentials
   - If that fails, attempt to connect to available open APs
   - If that fails, launch wifiManager
   - Or launch wifiManager on demand if a button is pressed.
   - In sleep mode, or if wifi is lost for some reason, the system should
   try to reconnect in this order every time.

tablatronix, 2019-05-06:
well the esp auto connects on its own already, so if thats not working something else is wrong with your board, try erasing flash and reload

I imagine very few people need wifi multi or auto connect to open, but there is no reason it needs to be in wm , it is easy to code those functions.

Daemach, 2019-05-06:
Would wifimulti conflict with the connection attempt that wifimanager is
trying to do?

tablatronix, 2019-05-06:
if you call autoconnect after already connected then it just exits

Daemach, 2019-05-06:
Will it unload the webserver completely?  I'm using the ESP32 with
asyncwebserver - a far more advanced and capable webserver that can support
mulltiple clients, websockets, etc.  They both want to bind to port 80 so I
need to make sure wifimanager doesn't conflict.

tablatronix, 2019-05-06:
you are probably going to have issues, either with webserver or dns server, I would not suggest it