tzapu/WiFiManager #883: get only Request redirected to captive portal

GvLente, 2019-05-10:
17:09:17.123 -> *WM: 
17:09:17.157 -> *WM: AutoConnect
17:09:17.157 -> *WM: Connecting as wifi client...
17:09:17.157 -> *WM: Using last saved values, should be faster
17:09:17.157 -> *WM: Connection result: 
17:09:17.157 -> *WM: 0
17:09:17.157 -> *WM: 
17:09:17.157 -> *WM: Configuring access point... 
17:09:17.157 -> *WM: ESP8266
17:09:17.157 -> *WM: password
17:09:17.638 -> *WM: AP IP address: 
17:09:17.638 -> *WM: 192.168.4.1
17:09:17.638 -> *WM: HTTP server started
17:09:23.432 -> *WM: Request redirected to captive portal
17:09:23.500 -> *WM: Request redirected to captive portal
17:09:29.368 -> *WM: Request redirected to captive portal
17:09:30.461 -> *WM: Request redirected to captive portal
etc.
etc. each time when i refresh

using Sketch: ESP8266_LED_Control_06_Station_Mode_with_mDNS_and_wifiManager
 
I dont get the wifimanager main page only get in android and apple a emty page 

File Not Found
URI: /
Method: GET
Arguments : 0

GvLente, 2019-05-10:
which informatie is missing

GvLente, 2019-05-10:
sorry i am not  familiar with github

GvLente, 2019-05-10:
AutoConnectWithTimeout.ino

gives in  serial:
20:08:22.105 -> *WM: AutoConnect
20:08:22.105 -> *WM: Connecting as wifi client...
20:08:22.105 -> *WM: Using last saved values, should be faster
20:08:22.105 -> *WM: Connection result: 
20:08:22.105 -> *WM: 0
20:08:22.105 -> *WM: 
20:08:22.105 -> *WM: Configuring access point... 
20:08:22.105 -> *WM: AutoConnectAP
20:08:22.615 -> *WM: AP IP address: 
20:08:22.615 -> *WM: 192.168.4.1
20:08:22.615 -> *WM: HTTP server started
20:08:34.265 -> *WM: Request redirected to captive portal
20:08:46.222 -> *WM: Request redirected to captive portal
20:08:47.343 -> *WM: Request redirected to captive portal
20:08:47.411 -> *WM: Request redirected to captive portal

tablatronix, 2019-05-10:
might be a bug that was fixed in development branch already, not sure

GvLente, 2019-05-11:
ok, I tryd development 0.14 there was the same result

dontsovcmc, 2019-05-19:
It's better to use Wireshark to sniff requests.. 

I got same infinitive `*WM: Request redirected to captive portal` when connected to ESP by MacOS with running virtual Windows7. When I turn off Windows7 (Parallels), ESP web server works great. It means there are some unknown requests, that DDOS ESP server.

tablatronix, 2019-05-19:
.14 is not development branch

tablatronix, 2019-05-19:
It is more likely that the host is ignoring the dns responses, but yeah you need to wireshark this to figure it out for sure.

GvLente, 2019-05-20:
Which is the development branch?
I will whireshark and see whats going wrong thanks voor the lead and the good work