tzapu/WiFiManager #884: Request redirected to captive portal

GastonMelo, 2019-05-13:
until 3 days ago I was using WiFiManager without problems. I install Arduino 1.8.9 and everything stop working with this message on serial monitor:
xxxxxxxxxxxxxxxxxxx
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 4
*WM: Failed to connect.
So I reinstall arduino 1.8.7 and try again but the error is still there, dont know where and why.
this are my headers:
#include <FS.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <DNSServer.h>
#include <WiFiManager.h>
#include "ArduinoJson.h"

and I using this code:
  WiFiManagerParameter custom_WiFiRelay("WiFiRelay", "WiFiRelayName", WiFiRelay, 21);
  // Connect to WiFi network
  WiFiManager wifiManager;
  //set config
  wifiManager.setSaveConfigCallback(saveConfigCallback);
  //agregar nuevo parametro
  wifiManager.addParameter(&custom_WiFiRelay);
  //***************RESET WIFIMANAGER***********************
  //wifiManager.resetSettings();
  //*******************************************************
  wifiManager.autoConnect("WiFiRelay");//coment serial.begin()
  if (!MDNS.begin("esp01")) {}
  server.begin();
  strcpy(WiFiRelay, custom_WiFiRelay.getValue());
if (shouldSaveConfig) {
    Serial.println("saving config");
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
    json["WiFiRelay"] = WiFiRelay;
    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    json.printTo(Serial);
    json.printTo(configFile);
    configFile.close();
  }

I check the router, disable firewall PC,  re-re-install arduino 1.8.9 and 1.8.7. Check code and compared to working examples. How can something that was working with arduino 1.8.7 until 4 days ago stop working installing arduino 1.8.9?

tablatronix, 2019-05-13:
erase your flash

GastonMelo, 2019-05-13:
thanks, solve the problem.