tzapu/WiFiManager #887: Autoconnect on if credentials are saved

malaki86, 2019-05-15:
I have a project that can use WiFi, but it's not required. What I'm looking for is a way to only attempt to autoconnect if there is already a WiFi connection saved. If not, skip it and the user can hit a button later on to start the AP config page.

I know that I can use the config page timeout, but I don't want the system to 'hang'.

tablatronix, 2019-05-15:
You should be able to just getssid and check but its a little different in esp32

SomeGuyNamedJay, 2019-05-15:
Does this work for you?
  if (WiFi.SSID()=="")
  {
    Serial.println("Blank");
  }
  else
  {
    Serial.println(WiFi.SSID());
    WiFi.begin(WiFi.SSID(), WiFi.psk());
  }

malaki86, 2019-05-15:
I saw the getssid in the keywords file, but when I try to use it, I get a 'function not defined' error 

WiFi.SSID returns the said for the current, active WiFi connection. I need to read the saved one, if it exists.

tablatronix, 2019-05-15:
Yeah that is the behavior on esp32 also, if you want to read the saved ssid you might have to read directly from the config struct.

```C++
    struct station_config conf;
    wifi_station_get_config_default(&conf);
    // conf.ssid;
```

tablatronix, 2019-05-15:
on esp32 it is 

```C++
String WiFiManager::WiFi_SSID(){
  #ifdef ESP8266
    return WiFi.SSID();
  #elif defined(ESP32)
    //@todo , workaround only when not connected
    wifi_config_t conf;
    esp_wifi_get_config(WIFI_IF_STA, &conf);
    return String(reinterpret_cast<const char*>(conf.sta.ssid));
  #endif
}
```

tablatronix, 2019-05-15:
Might be nice to make a getter for this

malaki86, 2019-05-21:
It wouldn't even have to return the actual SSID or password that WiFiManager has saved. It could be a simple boolean return value.

Example:
  if WiFiManager.Saved == TRUE {
    WiFiManager.autoconnect;
  }