tzapu/WiFiManager #889: why restart-up while attachInterrupt used;

wsdxyz, 2019-05-22:
### Basic Infos
why esp8266 always restart-up while attachInterrupt used;

#### Hardware
**Esp8266
- [ ] ESP8266

**Hardware:			ESP-12e, 

- [ ] ESP12 E(nodemcu)

### Description

why esp8266 always restart-up while attachInterrupt used;

### Settings in IDE

Module:  			NodeMcu, 

Additional libraries: 
#include    "WIFIManager.h"

### Sketch

void setup() {
  attachInterrupt(5,interrupt_1,3);
}

void loop() {
  // put your main code here, to run repeatedly:
}

void interrupt_1    ()
{

}

#include <Arduino.h>

void setup() {

}

void loop() {

}

### Debug Messages

[Info] Opened the serial port - COM29
r$
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vac02aff5
~ld
[Done] Closed the serial port

tablatronix, 2019-05-22:
and it works if you remove wifimanager?

tablatronix, 2019-05-22:
Should interrupt_1  be in IRAM?
'ICACHE_RAM_ATTR`

wsdxyz, 2019-05-23:
hi, tablatronix, thanks for your reply.
it's worked while i only use attachInterrupt without #include WiFiManager; 
and also worked while #include WiFiManager without attchInterrupt
but not works while only conbined them both;

wsdxyz, 2019-05-23:
hi, tablatronix, Thanks a lot for your solution on ICACHE_RAM_ATTR!
this problem of restart-up is perfectly solved by your solution: ICACHE_RAM_ATTR.
below is my code:
void ICACHE_RAM_ATTR interrupt_1    ()                      ;

closed 2019-05-23 by wsdxyz