tzapu/WiFiManager #892: storeSTAmode and persistent(true) / (false)

ortegafernando, 2019-05-29:
Hi, I am using since last week the ota branch. (I have check development source code and for this question is the same)

After configuring my wemos with wifimanager, it we reset it I get this serial output with "wm.debugPlatformInfo();"

```
21:38:24.418 -> *WM: [1] getCoreVersion():          2_5_1
21:38:24.418 -> *WM: [1] system_get_sdk_version():  2.2.1(cfd48f3)
21:38:24.418 -> *WM: [1] system_get_boot_version(): 31
21:38:24.418 -> *WM: [1] getFreeHeap():             48248
21:38:24.452 -> Try to wm.autoConnect
21:38:24.452 -> *WM: [1] AutoConnect 
21:38:24.452 -> *WM: [2] Connecting as wifi client... 
21:38:24.452 -> *WM: [2] Custom STA IP/GW/Subnet 
21:38:24.452 -> ip:192.168.1.XXX,mask:255.255.255.0,gw:192.168.1.1
21:38:24.452 -> *WM: [1] STA IP set: 192.168.1.XXX
21:38:24.452 -> *WM: [3] WIFI station disconnect 
21:38:24.452 -> scandone
21:38:24.452 -> *WM: [1] Connecting to saved AP: XXXXXXXX
21:38:24.452 -> *WM: [3] WiFi station enable 
21:38:24.452 -> *WM: [3] enableSTA PERSISTENT ON 
21:38:24.452 -> *WM: [2] 30000 ms connectTimeout set
21:38:24.452 -> *WM: [2] 30000 ms timeout, waiting for connect...
21:38:24.487 -> *WM: [2] . 
21:38:27.272 -> *WM: [2] . 
21:38:27.307 -> scandone
21:38:27.307 -> state: 0 -> 2 (b0)
21:38:27.307 -> state: 2 -> 3 (0)
21:38:27.307 -> state: 3 -> 5 (10)
21:38:27.307 -> add 0
21:38:27.307 -> aid 5
21:38:27.307 -> cnt 
21:38:27.375 -> *WM: [2] . 
21:38:27.477 -> *WM: [2] . 
21:38:27.578 -> *WM: [2] . 
21:38:27.611 -> 
21:38:27.611 -> connected with XXXXXXXXX, channel 13
21:38:27.611 -> ip:192.168.1.XXX,mask:255.255.255.0,gw:192.168.1.1
21:38:27.677 -> *WM: [2] Connection result: WL_CONNECTED
21:38:27.677 -> *WM: [3] lastconxresult: WL_CONNECTED
21:38:27.677 -> *WM: [1] AutoConnect: SUCCESS 
21:38:27.677 -> *WM: [1] STA IP Address: 192.168.1.XXX
21:38:27.677 -> connected...yeey :)
21:38:27.677 -> bcn 0
21:38:27.677 -> del if1
21:38:27.677 -> pm open,type:2 0
21:38:27.677 -> mode : sta(XXXXXXXXX)
21:38:27.712 -> 
21:38:27.712 -> Wifi mode: STA
21:38:27.712 -> After waiting 0.00 secs in setup() connection result is 3
21:38:27.712 -> local ip: 192.168.1.XXX
```

After checking source code, I can see how wifimanager uses persistent(false) and (true) to get managed not to corrupt flash memory.

But I can see here these two lines:

```
21:38:24.452 -> *WM: [3] WiFi station enable 
21:38:24.452 -> *WM: [3] enableSTA PERSISTENT ON
```

It is because storeSTAmode variable is set to true in wifimanager.h.

Then, I understand that everytime the ESP8266 begins/resets it will write once to flash.

1. Is this correct?
2. What is the problem if we set storeSTAmode = false ? (I dont exactly know what will be the disadvantages or advantages). Is it really necessary ?

It is not an issue but I want to understand the risk of brick my device because of corrupted flash vs setting storeSTAmode = false

Thanks a lot.

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development
- [X] Ota

**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)
- [x] 2.5.1

tablatronix, 2019-05-29:
Esp only saves flash if it differs, so no it should not matter, I think I added that to prevent saving the wifi if it fails or for testing the autoconnect autocp failover repeatedly.

Not sure what it is exactly you have a problem with or are trying to fix

tablatronix, 2019-05-29:
by brick your esp do you mean wear out the flash?

ortegafernando, 2019-05-30:
> Esp only saves flash if it differs, so no it should not matter, I think I added that to prevent saving the wifi if it fails or for testing the autoconnect autocp failover repeatedly.
> 
> Not sure what it is exactly you have a problem with or are trying to fix

Hi, it is about this two lines:

But I can see here these two lines:
```
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON
```

It is because storeSTAmode variable is set to true in wifimanager.h, so we get "enableSTA PERSISTENT ON" so our ESP8266 chip will save in flash the STA mode.

As you say, you added code to prevent saving the wifi credentials every time if they aren't changed, but you can see that every time the esp8266 boots or reset it self (after sleep for example), we are saving to flash the STA mode.

Yesterday I checked your code (development branch and ota branch) and I finally finished to this function:
```
// set mode ignores WiFi.persistent 
bool WiFiManager::WiFi_Mode(WiFiMode_t m,bool persistent) {
    bool ret;
    #ifdef ESP8266
      if((wifi_get_opmode() == (uint8) m ) && !persistent) {
          return true;
      }
      ETS_UART_INTR_DISABLE();
      if(persistent) ret = wifi_set_opmode(m);
      else ret = wifi_set_opmode_current(m);
      ETS_UART_INTR_ENABLE();
    return ret;
    #elif defined(ESP32)
      if(persistent && esp32persistent) WiFi.persistent(true);
      ret = WiFi.mode(m); // @todo persistent check persistant mode , NI
      if(persistent && esp32persistent) WiFi.persistent(false);
      return ret;
    #endif
}
```

As you can see, with storeSTAmode == true, we finally call to "wifi_set_opmode". I have read that "wifi_set_opmode" saves mode to flash memory.

So could it be bad/good to avoid that and put storeSTAmode = false? Why we need to save mode ?

I hope your understand me now, sorry for my bad explanations and english

tablatronix, 2019-05-31:
The esp lib handles flash saves and it only writes flash if config changed.

ortegafernando, 2019-05-31:
> Hmm let me check, I see what you mean now. Good point

Thanks a lot to check it.

tablatronix, 2019-05-31:
I mean the espressif docs say it will "write to flash if changed" But since we cannot see their code, I might just add a extra check to not save if already set to the same mode. Just in case.

Now I am curious, I might break out my scope and see if I can log flash writes on an esp board

ortegafernando, 2019-06-03:
> I mean the espressif docs say it will "write to flash if changed" , to me this means it handles all flash writes and ignores them if not needed, But since we cannot see their code, I might just add a extra check to not save if already set to the same mode. Just in case.
> 
> Now I am curious, I might break out my scope and see if I can log flash writes on an esp board

I understand what you mean. Yes, it makes us doubt about if it is really flashing or not. 

Any way ... is it neccesary ? Why we need to save the mode ? 

I could change my WifiManager.h file, but I prefer to keepmy libraries as yours to get any update without any problem. (I am now using OTA branch).

If you think that it is not neccesary to save the WIFI MODE to flash, may we have to change WifiManager.h and change storeSTAmode to FALSE.

May be I don't understand correctly how wifimanager works, but I dont think that storeSTAmode has to be TRUE.

Regards,

tablatronix, 2019-06-03:
Well you have to set it at some point so esp auto-connect works