tzapu/WiFiManager #893: MQTT publish using wifi manger+ http server

amrithmmh, 2019-05-31:
is it possible to add mqtt publish to nodemcu with wifimanager +http server ?

amrithmmh, 2019-06-01:
i found an alternate method . we can  use javascript on server side to transmit data using MQTT
but is it possible with the device(arduino or esp8266 as server) serve a html page as well as transmit data to external server using MQTT

tablatronix, 2019-06-01:
I see no reason why you cannot use a mqtt pub library, what exactly is the problem here?

amrithmmh, 2019-06-03:
working esp8266 wbserver+wifimanager
`
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <FS.h>
#include <WiFiManager.h>
#include <DNSServer.h>
#include <ArduinoOTA.h>

#include "SPI.h" // SPI library
#include "MFRC522.h" // RFID library (https://github.com/miguelbalboa/rfid)

const int pinRST = 5;
const int pinSDA = 15;

const int trigPin = 2;//digital 4
const int echoPin = 16;//digital

// defines variables
long duration;
int distance;

MFRC522 mfrc522(pinSDA, pinRST); // Set up mfrc522 on the Arduino
ESP8266WebServer server ( 80 );

void handlelock() {
  int a = digitalRead(D1);
  String adcValue = String(a);
  server.send(200, "text/plane", adcValue); //Send ADC value only to client ajax request
}

void handleLevelPercentage() {

 digitalWrite(trigPin, LOW);
delayMicroseconds(50);
digitalWrite(trigPin, HIGH);
delayMicroseconds(10);
digitalWrite(trigPin, LOW);
duration = pulseIn(echoPin, HIGH);
// Calculating the distance
distance= duration*0.034/2;
// Prints the distance on the Serial Monitor
//Serial.print("Distance: ");
Serial.println(distance);
  String adcValue = String(distance);
  server.send(200, "text/plane", adcValue); //Send ADC value only to client ajax request
}

void setup() {
  SPI.begin(); // open SPI connection
  mfrc522.PCD_Init(); // Initialize Proximity Coupling Device (PCD)
  Serial.begin (115200);
  pinMode(trigPin, OUTPUT); // Sets the trigPin as an Output
  pinMode(echoPin, INPUT); // Sets the echoPin as an Input
  pinMode(D1, OUTPUT);
  WiFiManager wifiManager;
  WiFi.hostname("Novatek_Level_IoT");
  wifiManager.autoConnect("Novatek_IoT", "Novatek");
  //ArduinoOTA.setPassword((const char *)"Novatek");

  ArduinoOTA.onStart([]() {
   // Serial.println("OTA Start");
  });
  ArduinoOTA.onEnd([]() {
   // Serial.println("\nOTA End");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    //Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
   // Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) ;// Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) ;//Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) ;//Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) ;//Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) ;//Serial.println("End Failed");
  });
  ArduinoOTA.begin();

  Serial.println("Ready");

  if (!SPIFFS.begin())
  {
   // Serial.println("SPIFFS Mount failed");
  }
  else
  {
    //Serial.println("SPIFFS Mount succesfull");
  }
  server.begin();
  server.on("/", fileindex);
  server.on("/index.html", fileindex);
  server.on("/bootstrap.min.css", bootstrap);
  server.on("bootstrap.min.css", bootstrap);
  server.on("/bootstrap.min.js", bootstrapmin);
  server.on("bootstrap.min.js", bootstrapmin);
  server.on("/jquery.min.js", jquery);
  server.on("jquery.min.js", jquery);
  server.on("/jQ.css", jquery_circle);//jQuery-plugin-progressbar.css
  server.on("jQ.css", jquery_circle);
  server.on("/jQ.js", jquery_circlejs);
  server.on("jQ.js", jquery_circlejs);
  server.on("/stateL", handlelock);
  server.on("/levelpercent", handleLevelPercentage);

 // Serial.println ( "HTTP server started" );

}

unsigned long previousMillis = 0;

const long interval = 1000;

void loop() {
  // put your main code here, to run repeatedly:
  unsigned long currentMillis = millis();
  ArduinoOTA.handle();
  server.handleClient();

  if (currentMillis - previousMillis >= interval) {
    // save the last time you blinked the LED
    previousMillis = currentMillis;

    if (mfrc522.PICC_IsNewCardPresent()) { // (true, if RFID tag/card is present ) PICC = Proximity Integrated Circuit Card
      if (mfrc522.PICC_ReadCardSerial()) { // true, if RFID tag/card was read
        String content = "";
        byte letter;
        for (byte i = 0; i < mfrc522.uid.size; i++)
        {
          Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
          Serial.print(mfrc522.uid.uidByte[i], HEX);
          content.concat(String(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " "));
          content.concat(String(mfrc522.uid.uidByte[i], HEX));
        }

        // Print out of id is complete.
        content.toUpperCase();
       // Serial.print(content.substring(1));
        if (content.substring(1) == "C2 12 46 1C")
        {
         // Serial.println("UNLOCKED!!!");
          if (digitalRead(D1))
            digitalWrite(D1, LOW);
          else
            digitalWrite(D1, HIGH);

        }
      }
    }
  }
}

void fileindex()
{
  File file = SPIFFS.open("/index.html.gz", "r");
  size_t sent = server.streamFile(file, "text/html");
}
void bootstrap()
{
  File file = SPIFFS.open("/bootstrap.min.css.gz", "r");
  size_t sent = server.streamFile(file, "text/css");
}
void jquery()
{
  File file = SPIFFS.open("/jquery.min.js.gz", "r");
  size_t sent = server.streamFile(file, "application/javascript");
}
void bootstrapmin()
{
  File file = SPIFFS.open("/bootstrap.min.js.gz", "r");
  size_t sent = server.streamFile(file, "application/javascript");
}
void  jquery_circlejs()
{
  File file = SPIFFS.open("/jQ.js.gz", "r");
  size_t sent = server.streamFile(file, "application/javascript");
}

void  jquery_circle()
{
  File file = SPIFFS.open("/jQ.css.gz", "r");
  size_t sent = server.streamFile(file, "text/css");
}`

amrithmmh, 2019-06-03:
`
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <ESP8266WebServer.h>
#include <FS.h>
#include <WiFiManager.h>
#include <DNSServer.h>
#include <ArduinoOTA.h>

#include "SPI.h" // SPI library
#include "MFRC522.h" // RFID library (https://github.com/miguelbalboa/rfid)

const int pinRST = 5;
const int pinSDA = 15;

const int trigPin = 2;//digital 4
const int echoPin = 16;//digital

// defines variables
long duration;
int distance;

MFRC522 mfrc522(pinSDA, pinRST); // Set up mfrc522 on the Arduino
ESP8266WebServer server ( 80 );

const char* mqtt_server = "myadress.com";

WiFiManager wifiManager;
WiFiClient espClient;
PubSubClient client(espClient);

void handlelock() {
  int a = digitalRead(D1);
  String adcValue = String(a);
  server.send(200, "text/plane", adcValue); //Send ADC value only to client ajax request
}

void handleLevelPercentage() {

 digitalWrite(trigPin, LOW);
delayMicroseconds(50);
digitalWrite(trigPin, HIGH);
delayMicroseconds(10);
digitalWrite(trigPin, LOW);
duration = pulseIn(echoPin, HIGH);
// Calculating the distance
distance= duration*0.034/2;
// Prints the distance on the Serial Monitor
//Serial.print("Distance: ");
Serial.println(distance);
  String adcValue = String(distance);
  server.send(200, "text/plane", adcValue); //Send ADC value only to client ajax request
}

void callback(char* topic, byte* payload, unsigned int length) {
Serial.print("Message arrived [");
Serial.print(topic);
Serial.print("] ");
for (int i = 0; i < length; i++) {
Serial.print((char)payload[i]);
}
Serial.println();

// Switch on the LED if an 1 was received as first character
if ((char)payload[0] == '1') {
digitalWrite(BUILTIN_LED, LOW); // Turn the LED on (Note that LOW is the voltage level
// but actually the LED is on; this is because
// it is acive low on the ESP-01)
} else {
digitalWrite(BUILTIN_LED, HIGH); // Turn the LED off by making the voltage HIGH
[... 150 lines cut ...]
{
  File file = SPIFFS.open("/bootstrap.min.css.gz", "r");
  size_t sent = server.streamFile(file, "text/css");
}
void jquery()
{
  File file = SPIFFS.open("/jquery.min.js.gz", "r");
  size_t sent = server.streamFile(file, "application/javascript");
}
void bootstrapmin()
{
  File file = SPIFFS.open("/bootstrap.min.js.gz", "r");
  size_t sent = server.streamFile(file, "application/javascript");
}
void  jquery_circlejs()
{
  File file = SPIFFS.open("/jQ.js.gz", "r");
  size_t sent = server.streamFile(file, "application/javascript");
}

void  jquery_circle()
{
  File file = SPIFFS.open("/jQ.css.gz", "r");
  size_t sent = server.streamFile(file, "text/css");
}`

tablatronix, 2019-06-03:
Can you elaborate on "doesnt work" ?