tzapu/WiFiManager #896: lots of additional Serial info causing watch dog reset

ThatchSS, 2019-06-11:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X] Development

**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [X] 2.3.0
- [X] 2.4.0
- [ ] staging (master/dev)

### Description

I am getting a lot of new info flooding over serial line causing watch dog timer to restart.
This is new and the setDebugOutput(false) dose not suppress it.
I have tried a few different boards and different core versions.
I have also tried some older sketches that I know did not produce this.
it happens on windows and apple when the captive portal connects but
don't know where it is originating from.
any help would be great.
Thanks

here is a sample of the info.

mounting FS...
mounted file system
New client
method: GET url: /hotspot-detect.html search: 
headerName: Host
headerValue: captive.apple.com
headerName: Connection
headerValue: close
headerName: User-Agent
headerValue: CaptiveNetworkSupport-355.260.2 wispr
args: 
Request: /hotspot-detect.html
 Arguments: 
request handler not found
New client
method: GET url: / search: 
headerName: Host
headerValue: 192.168.4.1
headerName: Connection
headerValue: close
headerName: User-Agent
headerValue: CaptiveNetworkSupport-355.260.2 wispr
args: 
Request: /
 Arguments: 
New client
method: GET url: /hotspot-detect.html search: 
headerName: Host
headerValue: captive.apple.com
headerName: Upgrade-Insecure-Requests
headerValue: 1
headerName: Accept
headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
headerName: User-Agent
headerValue: Mozilla/5.0 (iPhone; CPU iPhone OS 12_3_1 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Mobile/15E148
headerName: Accept-Language
headerValue: en-us
headerName: Accept-Encoding
headerValue: gzip, deflate
headerName: Connection
headerValue: keep-alive
args: 
Request: /hotspot-detect.html
 Arguments: 
request handler not found
New client
method: GET url: / search: 
headerName: Host
headerValue: 192.168.4.1
headerName: Upgrade-Insecure-Requests
headerValue: 1
headerName: Accept
headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
headerName: User-Agent
headerValue: Mozilla/5.0 (iPhone; CPU iPhone OS 12_3_1 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Mobile/15E148
headerName: Accept-Language
headerValue: en-us
headerName: Accept-Encoding
headerValue: gzip, deflate
headerName: Connection
headerValue: keep-alive
args: 
Request: /
 Arguments: 
New client
method: GET url: /hotspot-detect.html search: 
headerName: Host
headerValue: captive.apple.com
headerName: Connection
headerValue: close
headerName: User-Agent
headerValue: CaptiveNetworkSupport-355.260.2 wispr
args: 
Request: /hotspot-detect.html
 Arguments: 
request handler not found
New client
method: GET url: / search: 
headerName: Host
headerValue: 192.168.4.1
headerName: Connection
headerValue: close
headerName: User-Agent
headerValue: CaptiveNetworkSupport-355.260.2 wispr
args: 
Request: /
 Arguments:

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 
I seems to happen on all sketches this is from the auto connect FSP parameters and custom IP
I've tried many and can't find one that doesn't produce this

`#include <FS.h>          // this needs to be first, or it all crashes and burns...
#include "WiFiManager.h" // https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h> // https://github.com/bblanchon/ArduinoJson

#ifdef ESP32
  #include <SPIFFS.h>
#endif

//default custom static IP
char static_ip[16] = "10.0.1.56";
char static_gw[16] = "10.0.1.1";
char static_sn[16] = "255.255.255.0";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setupSpiffs(){
  //clean FS, for testing
  // SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");

          if(json["ip"]) {
            Serial.println("setting custom ip from config");
            strcpy(static_ip, json["ip"]);
            strcpy(static_gw, json["gateway"]);
            strcpy(static_sn, json["subnet"]);
            Serial.println(static_ip);
          } else {
            Serial.println("no custom ip in config");
          }

        } else {
          Serial.println("failed to load json config");
        }
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
  //end read
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.println();

  setupSpiffs();

  // WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wm;
  wm.setDebugOutput(false);
  //set config save notify callback
  wm.setSaveConfigCallback(saveConfigCallback);

  // set static ip
  IPAddress _ip,_gw,_sn;
  _ip.fromString(static_ip);
  _gw.fromString(static_gw);
  _sn.fromString(static_sn);
  wm.setSTAStaticIPConfig(_ip, _gw, _sn);

  //reset settings - wipe credentials for testing
  //wm.resetSettings();

  //automatically connect using saved credentials if they exist
  //If connection fails it starts an access point with the specified name
  //here  "AutoConnectAP" if empty will auto generate basedcon chipid, if password is blank it will be anonymous
  //and goes into a blocking loop awaiting configuration
  if (!wm.autoConnect("AutoConnectAP", "password")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    // if we still have not connected restart and try all over again
    ESP.restart();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  //save the custom parameters to FS
  if (shouldSaveConfig) {
    Serial.println("saving config");
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();

    json["ip"]          = WiFi.localIP().toString();
    json["gateway"]     = WiFi.gatewayIP().toString();
    json["subnet"]      = WiFi.subnetMask().toString();

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    json.prettyPrintTo(Serial);
    json.printTo(configFile);
    configFile.close();
    //end save
    shouldSaveConfig = false;
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());
  Serial.println(WiFi.gatewayIP());
  Serial.println(WiFi.subnetMask());
}

void loop() {
  // put your main code here, to run repeatedly:

}`

### Debug Messages

```
messages here
```

tablatronix, 2019-06-11:
do you have some kind of esp debugging enabled? Cause that looks like your webserver library

tablatronix, 2019-06-11:
```
/Users/user/projects/microcontrollers/dev/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/Parsing.cpp:
  153  
  154        #ifdef DEBUG_ESP_HTTP_SERVER
  155:       DEBUG_OUTPUT.print("headerName: ");
  156        DEBUG_OUTPUT.println(headerName);
  157        DEBUG_OUTPUT.print("headerValue: ");
  ...
```

ThatchSS, 2019-06-11:
Not intentionally? 
Do you know how I would disable it? 
Is it in the Arduino IDE? Or ...?

Daemach, 2019-06-11:
Tools menu in arduino IDE - make sure core debug is set to none.

ThatchSS, 2019-06-11:
UGH!
Yah it was but I cycled it to serial and then back to none and it took care if it I think.
The little things that hang you up.
Thanks all!

closed 2019-06-11 by ThatchSS