tzapu/WiFiManager #897: How can i add my own pages to the library captive portal?

noise5joker, 2019-06-12:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

tablatronix, 2019-06-12:
See DEV example for adding endpoints

noise5joker, 2019-06-12:
Where can i find it?
 @tablatronix

tablatronix, 2019-06-12:
Seriously?

https://github.com/tzapu/WiFiManager/blob/development/examples/DEV/OnDemandConfigPortal/OnDemandConfigPortal.ino#L83

noise5joker, 2019-06-12:
Thanks, I'm going to check that...I didn't find it because the last library
I downloaded didn't have this folder...

tablatronix, 2019-06-12:
Sorry I thought I saw you checked development oops.

Yes that is development branch on github, you have to use that to be able to share the webserver, or you can edit the .14 and change it to a shared or public object

noise5joker, 2019-06-12:
Perfect! thanks for the attention I'm grateful 🙌🏼

noise5joker, 2019-06-15:
Hey! i was looking the article #461 because i didnt solve my problem...i dont know where im going to put the HTML file or how can i edit the template to include my code and the esp8266 open it after the connection.

Bighoneypot, 2023-03-17:
> Seriously?
> 
> https://github.com/tzapu/WiFiManager/blob/development/examples/DEV/OnDemandConfigPortal/OnDemandConfigPortal.ino#L83

Dear @tablatronix check link, is wrong.