tzapu/WiFiManager #898: WMM: Error sending message to register the device?

TinajaLabs, 2019-06-14:
I'm working with the basic.ino version and somewhere in the bootup it says:
`WMM: Error sending message to register the device...`

I set up an mqtt subscriber in my node-red application and subscribed to `deviceLog/ESP_a020a6194f75` but I never see it arrive in node-red.

Looking for clues as to where this code lives.

Thanks,
Chris.

Here's the basic startup output as seen in the serial screen:
```
WiFiMQTTManager Night Light
WMM: clientId: ESP_a020a6194f75
WMM: mounting FS...
WMM: mounted file system...
WMM: reading config file...
WMM: opened config file...
{"friendly_name":"Tinaja","mqtt_server":"Tinaja-co","mqtt_port":"1883"}
WMM: parsed json...
WMM: lets try to connect to WiFi...
WMM: connected to WiFi!!...
WMM: 
WMM: local IP:192.168.1.113
WMM: _mqtt_server: Tinaja-co
WMM: _mqtt_port: 1883
WMM: clientId: ESP_a020a6194f75
WMM: attempting MQTT connection...mqtt connected...via setup...
subscribing to some topics:
   subscribed to: home/bedroom/night-light/wemosd1-1/ESP_a020a6194f75/output
Sending message to MQTT topic: deviceLog/ESP_a020a6194f75
{
  "deviceType": "ESP8266",
  "deviceId": "ESP_a020a6194f75",
  "name": "Tinaja",
  "chipId": "194f75",
  "sketchName": "tinaja-wemosd1mini-neopixel-nightlight-1.ino"
}
messageBuffer: {"deviceType":"ESP8266","deviceId":"ESP_a020a6194f75","name":"Tinaja","chipId":"194f75","sketchName":"tinaja-wemosd1mini-neopixel-nightlight-1.ino"}
WMM: Error sending message to register the device...
tinaja starting...
Message arrived on topic: home/bedroom/night-light/wemosd1-1/ESP_a020a6194f75/output. Message: 1
```

liebman, 2019-06-14:
I don't think those messages are from WiFiManager.  I think WiFiManager messages print with "WM*" not "WMM".

TinajaLabs, 2019-06-14:
Interesting clue.  Thanks.

tablatronix, 2019-06-14:
This is not this library

closed 2019-06-14 by tablatronix