tzapu/WiFiManager #901: Spiffs Not Saving Data  ESP-07

parthbhat13, 2019-06-21:
Hi,
i have used the wifimanager for many of my projects so far and it works like a charm on the esp8266-12E. 
Recently, due to some signal issues i was forced to use an ESP with the external antenna, i selected the ESP07. As always i ran the wifiAutoConnectwithFSParameter sketch and it did work for the first time. i started debugging and testing my new code and as well as the codes which were actually working on some of my nodes in my house. 
all of the sudden now the esp07 is not ready to save the SPIFFS data at all, it does save it for the first boot and startup after the WifiManager is configured, but after that it won't keep the data saved and on the restart everthing is lost. 
to go deeper and in depth, i even did some tests with the spiffs and even ran it on other fresh ESP07 (Thinking i may have damaged the flash on the Test ESP07) .

Now, even on the fresh esp07 everything is happening the same, its been 3 weeks i have been trying to fix the issue, and now it isnt ready to work on the new Chips also.

I have added a SPIFFS Info Code in the AutoConnect Example sketch and here is the serial output

### Serial Output For First Run

mounting FS...
mounted file system
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: blynk
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: password
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Scan done
*WM: DUP AP: Temkar's House
*WM: Temkar's House
*WM: -51
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: WiFi save
*WM: Parameter
*WM: server
*WM: m24.cloudmqtt.com
*WM: Parameter
*WM: port
*WM: 8080
*WM: Parameter
*WM: blynk
*WM: YOUR_BLYNK_TOKEN
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
Should save config
connected...yeey :)
saving config
{"mqtt_server":"m24.cloudmqtt.com","mqtt_port":"8080","blynk_token":"YOUR_BLYNK_TOKEN"}local ip
192.168.1.18
*WM: freeing allocated params!
m24.cloudmqtt.com

#####################
__________________________

Firmware: 
    Chip Id: 0025842C
    Core version: 2_5_2
    SDK version: 2.2.1(cfd48f3)
    Boot version: 31
    Boot mode: 1
__________________________

Flash chip information: 
    Flash chip Id: 00146085 (for example: Id=001640E0  Manuf=E0, Device=4016 (swap bytes))
    Sketch thinks Flash RAM is size: 1.00 MB
    Actual size based on chip Id: 1.00 MB ... given by (2^( "Device" - 1) / 8 / 1024
    Flash frequency: 40.00 MHz
    Flash write mode: DIO
__________________________

File system (SPIFFS): 
    Total KB: 139.96 KB
    Used KB: 0.49 KB
    Block size: 4096
    Page size: 256
    Maximum open files: 5
    Maximum path length: 32

SPIFFS directory {/} :
  /config.json
__________________________

CPU frequency: 80 MHz

#####################m24.cloudmqtt.com

---

I have been printing that data in the loop every second, that's the first print where it did show the "/config.json" and the used memory as 0.49KB. But on the second time data print, it shows nothing inside the spiffs.

---

### Serial Output First Run, Re-Printing The Data

File system (SPIFFS): 
    Total KB: 139.96 KB
    Used KB: 0.49 KB
    Block size: 4096
    Page size: 256
    Maximum open files: 5
    Maximum path length: 32

SPIFFS directory {/} :
__________________________

#####################m24.cloudmqtt.com

---

### Finally after the reset

mounting FS...
mounted file system
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: blynk
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.18
connected...yeey :)
local ip
192.168.1.18
*WM: freeing allocated params!

#####################
__________________________

Firmware: 
    Chip Id: 0025842C
    Core version: 2_5_2
    SDK version: 2.2.1(cfd48f3)
    Boot version: 31
    Boot mode: 1
__________________________

Flash chip information: 
    Flash chip Id: 00146085 (for example: Id=001640E0  Manuf=E0, Device=4016 (swap bytes))
    Sketch thinks Flash RAM is size: 1.00 MB
    Actual size based on chip Id: 1.00 MB ... given by (2^( "Device" - 1) / 8 / 1024
    Flash frequency: 40.00 MHz
    Flash write mode: DIO
__________________________

File system (SPIFFS): 
    Total KB: 139.96 KB
    Used KB: 0.00 KB
    Block size: 4096
    Page size: 256
    Maximum open files: 5
    Maximum path length: 32

SPIFFS directory {/} :
__________________________

CPU frequency: 80 MHz

#####################

Above again you can see that the SPIFFS directory is show empty as well as the used KB is also 0.00KB

i hope i was able to make my question understandable. 
Thanks in advance

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [] Master
- [ ] Development

**Esp8266/Esp32:**
- [] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [] Other (ESP07)

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)
- [] 2.5.2

simplyhippy, 2019-07-31:
Is your Issue solve

parthbhat13, 2019-07-31:
> Is your Issue solve

hey, 
Nope i spent some good amount of time to debug the situation but as i was having some time constrain i could not solve the issue properly. 
But i can defiantly update you on the whole scenario ive tried to solve so far.

I felt the library at least needs more than 1MB to function properly and save the data to the SPIFFS, after trying out more, i realized my ESP07 was not able to save data to the SPIFFS. it would write it and then i can see the data once, but then after a split second the data is like gone. on the next reset all i have is error reading the spiffs or there is no data.

Later, i tried to solve the issue with just trying to work on the spiffs alone, but then realized the spiffs is not at all properly functional!!

last option i am left with is to replace the SPIFFS with a 4MB one!. it works like a charm on 12E. 

So, Conclusion, i still cannot figure out what exactly went wrong the WifiManager or the SPIFFS, So far i feel Flash Memory is the issue!

Hope, My whole explanation was of some use!
Cheers!

tablatronix, 2019-07-31:
what does your compilation say about your space when you compile , it should tell you how much space your program uses

Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   4.2% (used 13884 bytes from 327680 bytes)
PROGRAM: [==        ]  19.1% (used 249719 bytes from 1310720 bytes)

parthbhat13, 2019-07-31:
Hi There, 

Here's what you asked for 
`Sketch uses 355128 bytes (46%) of program storage space.
 Maximum is 761840 bytes. 
Global variables use 35096 bytes (42%) of dynamic memory, leaving 46824 bytes for local variables. Maximum is 81920 bytes.`

simplyhippy, 2019-07-31:
Erasing the flash completely (you need to upload the blank_1MB.bin file to 0x000000, 0x100000, 0x200000 and 0x300000) as per the following link, solved it for me
can you paste your code

tablatronix, 2019-07-31:
Just use esptool to erase, or erase from arduino board menu..

parthbhat13, 2019-07-31:
@simplyhippy , well yup i did that but didn't work. , You can just use the basic WifiManager_FS code, even that didn't work for me, after the first try!
@tablatronix , yes, ofcourse i did use the esptool to erase the whole flash and make it work again, but it won't work at all, specially the spiffs. 
Now i hear you asking if i tried another chip, yes, i had a set of freshly packed roll of the ESP07 and i tested them on 3 chips but had same issues with all 3. i even asked my friends to do the same. still same issue!

theres an issue with the Flash memory of the 07.

tablatronix, 2019-08-01:
What is the flash size? Maybe its too small

tablatronix, 2019-08-01:
I have an esp07 I will have to solder it to a board to test

parthbhat13, 2019-08-01:
I selected 256k for spiffs from the total 1MB flash I've got on the chip. 

And yes even I feel the flash size can be the problem, but then I ain't able to figure why the most basic codes of just saving the data to the spiffs also won't work ! And it's happening with all the 3 chips I've got.

tablatronix, 2019-08-01:
Are you getting an error?

```c++
if (!SPIFFS.begin()) 
{
  Serial.println("Failed to mount the file system");
  return;
}
```

parthbhat13, 2019-08-01:
Nope! I'd rather get an error, saying It can't read the file. 

Like, once I save the config to the file, it will show the bytes used! But immediately it will not show any file, but used bytes! And once I reset, nothing. No more used bytes, back to normal! 

I hope you got what I'm trying to say!

wvsbsp, 2019-08-11:
Check the filename. In your log it states
`/config.json`
but the FS.h only supports 8.3 filenames. Try config .jsn or config.txt as filenames. 
I just finised a similar project to read and store configuration in SPIFFS. I also had to change the wifimanager.cpp to get access to the current SSID and password. Details see here
https://github.com/tzapu/WiFiManager/issues/918#issuecomment-520250685

parthbhat13, 2019-08-11:
@wvsbsp hey, 
Well did this error happen with you while working on the esp07 specifically? 

Because for me, it works great on esp12 without the editing's you asked to do! 

Still I'll try it later and get back to you on the update. 

Thanks for referring the post!

wvsbsp, 2019-08-11:
in my current project i'm using a NodeMCU which is basically the ESP12-E. But the only difference between the ESP-07 and the ESP-12E ist the integrated or external antenna. I've used the ESP-07 in another project on the same board as the ESP-12 without any differences in code (and it was in a larger project with SD-card, external multichannel 24Bit ADC using SPIFFS in parallel with SD-filesystem, download files with measurement data from the SD-card over the web-interface of the ESP while saving configuration data and web-stuff in the SPIFFS...)
Try to erase the whole memory while uploading (arduino, tools/erase flash/erase all flash contents) at least once. I've had strange problems with some boards that went away after erasing the whole flash.

parthbhat13, 2019-08-12:
Hey, 

Yup, I already did try to erase the whole memory still no luck, I will try to do what you asked in the previous reply soon. 
And yes ik, that 07 and 12e got the exact same things going underhood, but even the flash plays a role I feel, 07 offers 1MB where as the 12e has 4MB, I have a feeling that the flash is the fault here. But let me check again with your solution! 

Thanks for reply !

parthbhat13, 2019-08-12:
hey, @wvsbsp ,

like you said i did try to do the things accordingly and even did some testing again. but yeah no luck,  i just ran a very basic sample sketch of the SPIFFS to save a static data to config file and then read it back, plus i added some code to read whats the size of the spiffs and see what i have got, Once i save something to spiffs it does reflects and then it disappears, heres a sample of what i mean. 

---
`Here it mounted the FS`
> Mounting FS...
---
`Here I Just Printed the Parameters Of the Spiffs Before Saving Anything`
 > File system (SPIFFS): 
    Total KB: 228.20 KB
    Used KB: 0.00 KB
    Block size: 4096
    Page size: 256
    Maximum open files: 5
    Maximum path length: 32
  SPIFFS directory {/} : 

---
`I get the message as Config Saved, but then it tries to read it and it fail's to read`

>Config saved
Failed to parse config file
Failed to load config
---
`So again i print the Parameters of the spiffs After Saving the data`
>File system (SPIFFS): 
    Total KB: 228.20 KB
    Used KB: 0.49 KB
    Block size: 4096
    Page size: 256
    Maximum open files: 5
    Maximum path length: 32
SPIFFS directory {/} :
  /config.jsn

---
`At this point my void Setup(), Was done, and i print the parameters in the loop every 2 Sec`
>File system (SPIFFS): 
    Total KB: 228.20 KB
    Used KB: 0.49 KB
    Block size: 4096
    Page size: 256
    Maximum open files: 5
    Maximum path length: 32
SPIFFS directory {/} :
---

You can see the changes in the file size and everything here, after the reset everything goes back to normal!, i tried everything to fix this through software, now the last thing i am left to do is change the Flash Memory. And this is not just for one esp07, i got a roll of those, i used 3 from the roll and all are behaving the same!

And This is the code i [used](https://pastebin.com/KcF6zL0N)  

Thanks for your time and reading this.

mindforger, 2019-08-12:
you never flush or close your file from the look or is this not relevant?!

parthbhat13, 2019-08-12:
@mindforger hey, 
Well, i don't really know if that's relevant, as this is the main example taken from the Fs.h 
If it was relevant then it should have been highlighted in the code, which they haven't. 

And almost every other code and example people use this! 

Thanks a lot for replying!

wvsbsp, 2019-08-12:
same as mindforger... add
`configFile.flush(); `
`configFile.close(); `
after line 67. Otherwise the file-buffer that is stores in ram never ends in the spiffs-file.

Another hint: you seeem to be using the older version of ArduinoJson. With the current version, you have to make several changes. see https://arduinojson.org/v6/doc/upgrade/
My code is as follows (only the relevant lines and i didn't translate all the german comments)

void saveConfig(void) {
  configFile = SPIFFS.open("/config.jsn", "w");
  StaticJsonDocument<1024> jsonBuffer; // instead of StaticJsonBuffer
  jsonBuffer["ssid"] = ssid;
  jsonBuffer["pass"] = pass;
  jsonBuffer["mqtt_server"] = mqtt_server;
  jsonBuffer["mqtt_port"] = mqtt_port;
  jsonBuffer["mqtt_topic"] = mqtt_topic;
  serializeJson(jsonBuffer, configFile);   // instead of json.printto
  configFile.flush();         // alles schreiben
  configFile.close();         // Datei schließen
}

The following code ist part of my setup. If no configFile is found, the code generates one file with some default values (could be omitted if a config.jsn is present for shure)
  // *******************************************************
  // SPIFFS öffnen, Konfig-Datei lesen
  // *******************************************************
  if(SPIFFS.begin()) {
    Serial.println("SPIFFS Initialize....ok");
    configFile = SPIFFS.open("/config.jsn", "r");
    // noch keine Config-Datei vorhanden? Dann lege eine neue
    // Datei mit Standard-werten an.
    if (!configFile) {
      Serial.println("ConfigFile nicht gefunden");
      // no file found? Start with default values.
      strcpy(ssid, "MeinNetzwerk");
      strcpy(pass, "StrengGeheim");
      strcpy(mqtt_server, "test.mosquitto.org");
      strcpy(mqtt_port, "1883");
      strcpy(mqtt_topic, "#");  // never try this at home :-)
      saveConfig(); // write to file
    }
    // Wenn die Datei vorhanden ist, auslesen und in die Variablen kopieren
    else {
      Serial.println("ConfigFile gefunden!");
      size_t filesize = configFile.size();
      if (filesize > 1024) { 
        Serial.println("Konfigurationsdatei zu groß!");
        return; // Abbruch, Fataler Fehler!
      }
      // DEBUG -> print current content of config.jsn
      while(configFile.available()) {
        //Lets read line by line from the file
        String line = configFile.readStringUntil('n');
        Serial.println(line);
      }
      configFile.seek(0); // return to beginning of file, otherwise the following part gives faults
      // END OF DEBUG
      
      // Puffer buf anlegen, um die Datei zu lesen.
      std::unique_ptr<char[]> buf(new char[filesize]);
      // Config-Datei in den Puffer einlesen
      configFile.readBytes(buf.get(), filesize);
      
      DynamicJsonDocument jsonBuffer(filesize);
      DeserializationError error = deserializeJson(jsonBuffer, buf.get());
      if (error) {
        Serial.println("JSON Fehler!");
        Serial.println(error.c_str());  // print the JSON error -> see 
        return; // Abbruch, Fataler Fehler!
      }
      // JSON Strings in Variablen kopieren
      strcpy(ssid, jsonBuffer["ssid"]);
      strcpy(pass, jsonBuffer["pass"]);
      strcpy(mqtt_server, jsonBuffer["mqtt_server"]);
      strcpy(mqtt_port, jsonBuffer["mqtt_port"]);
      strcpy(mqtt_topic, jsonBuffer["mqtt_topic"]);
      configFile.close();
    }
  }    

Together with the WifiManager i can now change and save my settings (Wifi and MQTT). Further parameters can be stored (used this to save calibration data of my ADC / weighscale).

wvsbsp, 2019-08-12:
i forgot to mention... my strings are empty char-arrays.
char ssid[50];
char pass[20];
char mqtt_server[50];
char mqtt_port[6];
char mqtt_topic[34];

parthbhat13, 2019-08-12:
hey,
Yup mine too, they are just in sample, and it really doesn't matter if they are filled or empty.

wvsbsp, 2019-08-12:
you could initialize the variables and then only use saveConfig( ) if no config file is present.
Just never use const char ssid[ ] = "something";

parthbhat13, 2019-08-12:
Yup, I guess I've done that, not in this sample code then for sure in the main one which I used to deploy!

parthbhat13, 2019-08-13:
@wvsbsp 
Hi, 
I added the flush and close to my code, and ran it again. but nope no luck with the spiffs, im getting the same issues like i had earlier. i don't know what exactly is wrong with these boards. 

i'll replace the flash Memory and then get back on this thread!
Thanks everyone for helping

rodri16, 2020-01-28:
Hi, I have a similar problem (Dev Branch), my parameters are 
``
char mqtt_server[40] = "monitor.com"; 
char mqtt_port[6] = "80";
char TimeZone[5] = "-3.0";
char NumDisp[5] = "17";
char mult_V[12] = "448128.6";
char mult_I[12] = "16591.2";
char mult_P[13] = "12136999.39";
``
After full erase flash, everything is ok, but when booting again and finding the file config.jsn it crashes after the 5th parameter. Do you know why?

Log:

parsed json
Fatal exception 28(LoadProhibitedCause):

IronBamBam1990, 2021-09-22:
Fatal exception 28(LoadProhibitedCause): <--- I Think Some Librarys are corupted and Esp got problem on boot (28).

alirezaimi, 2022-01-06:
same problem here, i can't even open config page any more :
`
**************************************
00:06:39.978 -> *wm:[2] <- HTTP Wifi 
00:06:39.978 -> *wm:[2] WiFi Scan SYNC started 
00:06:42.166 -> *wm:[2] WiFi Scan completed in 2186 ms
00:06:42.199 -> *wm:[1] 2 networks found
00:06:42.199 -> *wm:[2] AP:  -57 A@Z
00:06:42.199 -> *wm:[2] AP:  -84 ab
00:06:42.199 -> 
00:06:42.199 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
00:06:42.199 -> 
00:06:42.199 -> Exception (28):
00:06:42.199 -> epc1=0x4021ff68 epc2=0x00000000 epc3=0x00000000 excvaddr=0x007a1200 depc=0x00000000
00:06:42.199 -> 
00:06:42.199 -> >>>stack>>>
00:06:42.199 -> 
00:06:42.199 -> ctx: cont
00:06:42.199 -> sp: 3ffffb80 end: 3fffffc0 offset: 0190
00:06:42.199 -> 3ffffd10:  00000000 402529cf 3ffffd4c 40211144  
00:06:42.199 -> 3ffffd20:  00000000 007a1200 3ffffd40 40210ef4  
00:06:42.350 -> 3ffffd30:  00000000 007a1200 3ffffd94 40203079  
00:06:42.350 -> 3ffffd40:  00000000 0016001f 00000000 007d747b  
00:06:42.350 -> 3ffffd50:  00000000 03000000 3ffffd94 40211144  
00:06:42.350 -> 3ffffd60:  402529d3 3ffeed6c 00000000 402042f3  
00:06:42.350 -> 3ffffd70:  00000e12 00000000 00000000 00000000  
00:06:42.350 -> 3ffffd80:  3ffe884a 00000000 00000000 000000e1  
00:06:42.350 -> 3ffffd90:  00000ef3 3fff33ac 0066006f 80000000  
00:06:42.350 -> 3ffffda0:  3fff2164 005c005f 80000000 402045af  
00:06:42.350 -> 3ffffdb0:  3fff3954 3fff32b4 3ffffdd0 00000009  
00:06:42.350 -> 3ffffdc0:  0000000c 00000037 ffffffff 00000011  
00:06:42.350 -> 3ffffdd0:  00000047 00000053 00000058 ffffffff  
00:06:42.350 -> 3ffffde0:  3ffffe00 00000000 3ffeed6c 3ffe884a  
00:06:42.350 -> 3ffffdf0:  402529d3 3ffeee6c 3ffeed6c 40209bae  
00:06:42.350 -> 3ffffe00:  00000000 0122012f 00000000 3fff32b4  
00:06:42.350 -> 3ffffe10:  00e100ef 80000000 3fff3954 0efc0eff  
00:06:42.350 -> 3ffffe20:  80000000 6a33d20f 00100000 00000000  
00:06:42.350 -> 3ffffe30:  00000000 3fff12e0 00000000 402115d4  
00:06:42.350 -> 3ffffe40:  00000000 3fff2164 3fff1864 3ffef1b0  
00:06:42.350 -> 3ffffe50:  3ffeed6c 00000001 3fff1284 4021b034  
00:06:42.350 -> 3ffffe60:  3fff0400 00000000 00000000 401000e9  
00:06:42.350 -> 3ffffe70:  00000000 00000000 3fff1824 40205669  
00:06:42.350 -> 3ffffe80:  00000000 3fff1154 3fff110c 402086b0  
00:06:42.350 -> 3ffffe90:  00000000 00000000 00000000 69746365  
00:06:42.350 -> 3ffffea0:  00000000 00000020 401002bd 00000000  
00:06:42.350 -> 3ffffeb0:  00006348 00000000 00000000 00000001  
00:06:42.350 -> 3ffffec0:  00000000 00000000 3fff110c 3ffef1b0  
00:06:42.350 -> 3ffffed0:  3ffeed6c 3fff1130 3fff110c 4020a892  
00:06:42.350 -> 3ffffee0:  4021bf08 00000000 00001388 4020d964  
00:06:42.350 -> 3ffffef0:  00000000 00000000 3fff1f8c 00000000  
00:06:42.350 -> 3fffff00:  3ffeed6c 3ffeee6c 3ffeed6c 4020aad0  
00:06:42.350 -> 3fffff10:  007a1200 7d63c1e8 3ffef000 3ffef1b0  
00:06:42.350 -> 3fffff20:  4020eb20 3ffe8b1c 3ffef03c 4020eb2c  
00:06:42.350 -> 3fffff30:  4020eb20 3ffe8b1c 3ffef03c 4020fb8c  
00:06:42.371 -> 3fffff40:  3fffdad0 00000026 3ffef03c 3ffef1b0  
00:06:42.371 -> 3fffff50:  3ffeed6c 00000000 3ffef03c 4020ac2b  
00:06:42.371 -> 3fffff60:  007a1200 00000000 3ffef03c 40201447  
00:06:42.371 -> 3fffff70:  00000000 32323a30 00000000 feefeffe  
00:06:42.371 -> 3fffff80:  00000000 00000000 00000001 401001a8  
00:06:42.398 -> 3fffff90:  3fffdad0 00000000 3ffef19c 3ffef1b0  
00:06:42.398 -> 3fffffa0:  3fffdad0 00000000 3ffef19c 40212124  
00:06:42.398 -> 3fffffb0:  feefeffe feefeffe 3ffe860c 40100d05  
00:06:42.398 -> <<<stack<<<
00:06:42.398 -> 
00:06:42.398 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
00:06:42.431 -> 
00:06:42.431 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
00:06:42.431 -> 
00:06:42.431 -> load 0x4010f000, len 3460, room 16 
00:06:42.431 -> tail 4
00:06:42.431 -> chksum 0xcc
00:06:42.431 -> load 0x3fff20b8, len 40, room 4 
00:06:42.431 -> tail 4
00:06:42.431 -> chksum 0xc9
00:06:42.431 -> csum 0xc9
00:06:42.431 -> v000606f0
00:06:42.431 -> ~ld
00:06:42.497 -> mounting FS...
00:06:42.530 -> mounted file system
00:06:42.530 -> *wm:[2] Added Parameter: server
00:06:42.530 -> *wm:[2] Added Parameter: port
00:06:42.530 -> *wm:[2] Added Parameter: apikey
00:06:42.530 -> The values in the file are: 
00:06:42.563 -> 	mqtt_server : 
00:06:42.563 -> 	mqtt_port : 8080
00:06:42.563 -> 	api_token : YOUR_API_TOKEN
00:06:42.563 -> *wm:[1] AutoConnect 
00:06:42.563 -> *wm:[1] No Credentials are Saved, skipping connect 
00:06:42.563 -> *wm:[2] Starting Config Portal 
00:06:42.563 -> *wm:[2] AccessPoint set password is VALID 
00:06:42.563 -> *wm:[2] Disabling STA 
00:06:42.563 -> *wm:[2] Enabling AP 
00:06:42.563 -> *wm:[1] StartAP with SSID:  imi-esp-c-portal
00:06:44.455 -> *wm:[1] AP IP address: 192.168.4.1
00:06:44.455 -> *wm:[1] Starting Web Portal 
00:06:44.455 -> *wm:[2] HTTP server started 
00:06:44.455 -> *wm:[2] Config Portal Running, non blocking/processing 
00:06:44.455 -> fialed to connect to wifi.

`
I tested with all possible FS size and esp8266 versions.

emersongcardona, 2022-04-01:
did you solve it? i just got the same issue, my code works well on ESP12 but crash on ESP07

parthbhat13, 2022-04-01:
Nope, I couldn't solve it for Esp-07. It did work with the Esp 12 tho. 
What I've done now is I ditched esp8266 and even if I use it, I've made my own basic web page and make it work that way. I believe it should even work with the esp07 here's the link to the YouTube video 

https://youtu.be/1VW-z4ibMXI

Sorry I am replying from my phone so not sure how much formatted this thing gonna go.

tablatronix, 2022-04-01:
I doubt you will be able to run modern esp libraries on anything with 1MB flash

IronBamBam1990, 2022-05-06:
ok so i do many try on that issue and problem appear in actually two situation:

1. When you saving values to JSON
2. When you dont configure your WiFi config right 
 

so for Json i got little solution so you are not out of heap 
#ifdef ARDUINOJSON_VERSION_MAJOR >= 6
        DynamicJsonDocument json(ESP.getMaxFreeBlockSize() - 4096);

IronBamBam1990, 2022-05-06:
And I found some info can causing problem:

https://www.sagecity.com/us/sage_erp_x3/b/sageerp_x3_product_support_blog/posts/sizing-recommendations-for-sage-x3-web-server-syracuse

IronBamBam1990, 2022-05-06:
Yes i tested out once more and thats solution:
DynamicJsonDocument json(ESP.getMaxFreeBlockSize() - 4096);

now console throwing soft wdt reset when i make ddos of data. Also i had problem with realtime save files in json format (front -end and back-end values). Error 29 was showing up when SPIFFS cant handle such amount of data. So i block real time saving from interface and i do it with SetInterval method and i start to use Task Manager to do job. With task manager you must be caution cause he also can make some troubles.