tzapu/WiFiManager #903: AutoConnect always goes to AP on reboot

devilstower, 2019-06-24:
I'm thinking that either I have a fundamental misunderstanding, or something is fundamentally wrong.

If I reduce my code to the smallest snippet possible:

```
    WiFiManager wifiManager;
    wifiManager.autoConnect("Config");

```
What happens the first time through is that the config screen appears. I select my network, enter the password, the system restarts on the selected network and all is well (well, I need to understand how to get it to display something _other than_ the config screen on the new network, but that's for another day). 

The problem is that, every time I restart the system, it goes right back to AP mode and behaves as if the config was never successful. If I check, it appears that the information is being successfully saved, and the switchover occurs without a hitch. It's just that every restart acts like the first.

I'm on version 0.14, on an Arducam ESP8266 Uno V2. 

What am I missing? Thanks.

tablatronix, 2019-06-24:
So your code stops at autoconnect and waits ?

devilstower, 2019-06-25:
No, it goes to AP mode, even if there are good credentials. I can trap for
it by simply doing a loop with WiFi myself, but otherwise it seems to plow
on past.  Not a show stopper, since I can simply avoid going to autoconnect
if I make a connection with WiFi.  But something I thought autoconnect was
supposed to handle.

Thanks! Terrific library. Now I need to find a good example of someone
modifying the config screen to display more info.

Mark

tablatronix, 2019-06-25:
I am saying what do logs say are you sure the esp is not starting ap mode?

akouz, 2019-06-27:
I have the same issue. Until I saw this topic I thought this is as it supposed to be, and I should store/restore credentials myself.

tablatronix, 2019-06-27:
Well the esp auto starts the ap so just making sure wm is actually running and you are not just assuming it is, set wifi mode to sta in your sketch to make sure.

If your credentials are not being remembered after reboots its usually a flash corruption and erase flash fixes it, always erase flash when switching to new sdks etc

akouz, 2019-06-27:
In my case apparently the reason was in a redundant old WiFi initialisation code I left in the setup(). After cleaning up it works as described. Thank you so much for your effort. Overall it is a great and very handy library, thanks a lot.

devilstower, 2019-06-27:
The mode problem was indeed the issue. The WiFi credentials were being
remembered, but for some reason the system was booting directly into AP.
Forcing it into STA allowed me to proceed.

Thanks.

markg85, 2019-07-19:
I have this very same issue.
My fix too was forcing the mode to be WIFI_STA when there is a connection....

I'm also very sure this doesn't happen on the main branch so you might have a regression here.

tablatronix, 2019-07-20:
This is the default ESP behavior, just set the mode in setup and it will remember it always

markg85, 2019-07-20:
> This is the default ESP behavior, just set the mode in setup and it will remember it always

isn't that why we use a WiFiManager in the first place, to manage these things for us? ...
This isn't the first thing either that i need to put in my setup and loop code that is WiFi specific (besides the WiFiManager initialization code). The WiFiManager _really_ should do this! And if it's not then it's a bug in my opinion.

I think you can _expect_ there to be no* WiFi specific code in the main setup/loop code besides setting up WiFiManager itself. So you should probably set the mode in the constructor (or in connectWifi).

\* well, i do need to check if there is a connection for other services to start once that connection is there. It would be neat to have a callback when there is a connection (and a disconnect too) as that prevents making my loop code heavier.

tablatronix, 2019-07-21:
This is part of ESP default functionality of persistent, it has nothing to do with WM

If a user wants to run a softap, in their own code then they can, why should WM override their code ?

This is already a feature in development branch, if you call autoconnect WM assumes you want WIFI_STA and sets it for you when configportalExits is called _usermode, I can even add a setter for people that do not know how esp works.

markg85, 2019-07-21:
> This is part of ESP default functionality of persistent, it has nothing to do with WM
> 
> If a user wants to run a softap, in their own code then they can, why should WM override their code ?
> 
> Either way
> This is already a feature in development branch, if you call `autoconnect` WM assumes you want `WIFI_STA` and sets it for you when configportal Exits it is called `_usermode`, I can add a setter for people that do not know how esp works also, but this should be working

You might be missing the point here.
I m well aware of `_usermode` and am on the development branch. You say that it sets the mode to WIFI_STA "when configportal exits". It doesn't when it never enters the configportal (aka, when it connects to a user/pass it had stored). It should set the mode to WIFI_STA in _both_ cases. Not _only_ when the configportal exits.

That should - imho - be the default behavior. If a user wants to override it to softap then the user should just do so... i don't see a problem there. WM "might" need an option to allow a user override in that case _or_ the user should be expected to always override _after_ WM has done it's thing.

I hope you can reconsider marking this issue as a wontfix as there definitely is a bug still.

tablatronix, 2019-07-21:
The default behavior is to not modify esp persistent mode.( which is WIFI_STA+WIFI_AP)

The better option here is for a library to do nothing, and require users to write proper user code and not have library change stuff when it does different things.

SteveRMann, 2019-08-14:
I had a similar issue on a recent project (not using WiFiManager), but it may explain what happened.

I had some code working just fine on a Wemos D1 Mini, but after a library upgrade, it stopped connecting to my WiFi.   I determined that if I use ESP8266WiFi library Version 2.5.0 or later, all I get from WiFi.status() is **WL_NO_SSID_AVAIL**.

By reverting to ESP8266WiFi library Version 2.4.2, in my IDE, I _can_ get a connection with the same sketch.

The fix was to put _WiFi.enableInsecureWEP();_ after WiFi.begin().

I have not looked at the WiFiManager.cpp file because modifying libraries is a bit above my pay grade.

tablatronix, 2019-08-14:
interesting, I just noticed these new options a couple days ago when browsing the esp code. This would be router dependent I assume, or an open network

SteveRMann, 2019-08-15:
Yes, my router is ancient; 3-years old.  It does not support WPA but the esp8266WiFi library Version 2.5.0 made WPA the default.

Now, how do I handle bad passwords?

tablatronix, 2019-12-04:
Add this to docs and an example

closed 2022-04-01 by tablatronix